espressif/esp-idf #5319: Linking fails on Ubuntu 20.04

DiegoActionItem, 2020-05-19:
I just upgraded to Ubuntu 20.04, using ESP v4.0. My project compiles under Ubuntu 18.04. After updating I get the attached error.

After update, I also removed `~/.espressif` and re-run `install.sh` (so, the system is more or less a fresh install).
```
[ 98%] Generating ld/esp32.project.ld
cd /home/diego/src/vayyar-smart-home/walabot-home/build/esp-idf/esp32 && python /home/diego/src/esp-idf-4.0/tools/ldgen/ldgen.py --config /home/diego/src/vayyar-smart-home/walabot-home/sdkconfig --fragments /home/diego/src/esp-idf-4.0/components/esp_ringbuf/linker.lf	/home/diego/src/esp-idf-4.0/components/spi_flash/linker.lf	/home/diego/src/esp-idf-4.0/components/esp_wifi/linker.lf	/home/diego/src/esp-idf-4.0/components/lwip/linker.lf	/home/diego/src/esp-idf-4.0/components/esp_event/linker.lf	/home/diego/src/esp-idf-4.0/components/espcoredump/linker.lf	/home/diego/src/esp-idf-4.0/components/esp32/linker.lf	/home/diego/src/esp-idf-4.0/components/esp32/ld/esp32_fragments.lf	/home/diego/src/esp-idf-4.0/components/xtensa/linker.lf	/home/diego/src/esp-idf-4.0/components/soc/linker.lf	/home/diego/src/esp-idf-4.0/components/heap/linker.lf	/home/diego/src/esp-idf-4.0/components/freertos/linker.lf	/home/diego/src/esp-idf-4.0/components/newlib/newlib.lf	/home/diego/src/esp-idf-4.0/components/app_trace/linker.lf	/home/diego/src/esp-idf-4.0/components/esp_gdbstub/linker.lf --input /home/diego/src/esp-idf-4.0/components/esp32/ld/esp32.project.ld.in --output /home/diego/src/vayyar-smart-home/walabot-home/build/esp-idf/esp32/ld/esp32.project.ld --kconfig /home/diego/src/esp-idf-4.0/Kconfig --env-file /home/diego/src/vayyar-smart-home/walabot-home/build/config.env --libraries-file /home/diego/src/vayyar-smart-home/walabot-home/build/ldgen_libraries --objdump /usr/bin/objdump
Traceback (most recent call last):
  File "/home/diego/src/esp-idf-4.0/tools/ldgen/ldgen.py", line 150, in <module>
    main()
  File "/home/diego/src/esp-idf-4.0/tools/ldgen/ldgen.py", line 126, in main
    mapping_rules = generation_model.generate_rules(sections_infos)
  File "/home/diego/src/esp-idf-4.0/tools/ldgen/generation.py", line 354, in generate_rules
    self._create_exclusions(mapping_rules, default_rules, sections_infos)
  File "/home/diego/src/esp-idf-4.0/tools/ldgen/generation.py", line 448, in _create_exclusions
    general_rule.add_exclusion(specific_rule, sections_info)
  File "/home/diego/src/esp-idf-4.0/tools/ldgen/generation.py", line 100, in add_exclusion
    expansions = do_section_expansion(self, section)
  File "/home/diego/src/esp-idf-4.0/tools/ldgen/generation.py", line 84, in do_section_expansion
    expansions = fnmatch.filter(sections_in_obj, section)
  File "/home/diego/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/fnmatch.py", line 60, in filter
    for name in names:
TypeError: 'NoneType' object is not iterable

```

DiegoActionItem, 2020-05-19:
Reminder to anyone else who uses `ccache`, please run `ccache -c` when changing distros. Now code compiles, and links for me.

walidamriou, 2020-05-21:
@DiegoActionItem I hope the notes  of my experience with ESP-IDF with its VScode extension Espressif IDF in ubuntu 20.04  can help you: 
https://notes.walidamriou.com/2020/05/my-experience-with-esp-idf-with-its.html

holubcsongor, 2020-05-22:
Note: if you have fresh install, you should have python3 instead of 2

walidamriou, 2020-05-22:
I hope this notes of  my experience with ESP-IDF with its VScode extension Espressif IDF in ubuntu 20.04 can help you:
https://notes.walidamriou.com/2020/05/my-experience-with-esp-idf-with-its.html

DiegoActionItem, 2020-05-24:
@walidamriou using `chmod a+rw` is an horrible idea. All you need to do is add your development user to the group `dialout` (the devices are regenerated each time you insert them into USB, so permissions will change).

```
sudo adduser diegorocks dialout
# you can logout -> re-login, no need to restart
# or run this, to get a terminal with new group permissions
newgrp dialout 
````

Anyway, the compilation works, so this issue is going to get closed.

closed 2020-05-24 by DiegoActionItem

walidamriou, 2020-05-24:
@DiegoActionItem I'll add your note to the notes 😀 thank you.

chegewara, 2020-05-24:
> Note: if you have fresh install, you should have python3 instead of 2

You have both, at least i got both verssions on fresh install. This solved my issue with python and prerequisities:
https://linuxconfig.org/ubuntu-20-04-python-version-switch-manager