espressif/esp-idf #5320: esp_read_mac derives wrong MAC address for Bluetooth in some configurations

devanlai, 2020-05-19:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.2-dev-1575-ga3520970f
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

When the number of universally administered MAC addresses is set to two, the derived Bluetooth MAC address does not match the [API documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system.html#mac-address).

Per the documentation, the derived BT MAC address when the number of universal MAC addresses is set to two should be formed by adding one to the last octet. However, instead, the BT MAC address is formed by adding two to the last octet.

In the test program below with `CONFIG_ESP32_UNIVERSAL_MAC_ADDRESSES=2`, the base MAC address is set to `00:01:02:03:04:00`.

### Expected Behavior
The bluetooth MAC address should be set to `00:01:02:03:04:01`

### Actual Behavior
The bluetooth MAC address is set to `00:01:02:03:04:02`
```
I (325) BASE MAC    : 00:01:02:03:04:00
I (325) WIFI_STA MAC: 00:01:02:03:04:00
I (335) SoftAP MAC  : 02:01:02:03:04:00
I (335) BT MAC      : 00:01:02:03:04:02
I (345) Ethernet MAC: 02:01:02:03:04:01
```

For reference, here is the output when the number of universal MAC addresses is set to 4, which is correct:
```
I (324) BASE MAC    : 00:01:02:03:04:00
I (324) WIFI_STA MAC: 00:01:02:03:04:00
I (334) SoftAP MAC  : 00:01:02:03:04:01
I (334) BT MAC      : 00:01:02:03:04:02
I (344) Ethernet MAC: 00:01:02:03:04:03
```

### Code to reproduce this issue

Sample gist based on the base_mac_address example:
https://gist.github.com/devanlai/8d2ce9aacd197ebc3214027a7ffd042b

Wth-Esp, 2020-05-20:
Hi, @devanlai 

Please apply this patch [bt_mac.zip](https://github.com/espressif/esp-idf/files/4654483/bt_mac.zip)
.

Thanks
Weitianhua

devanlai, 2020-05-20:
Hi @Wth-Esp, I can confirm that this patch fixes the issue for me. Looking forward to seeing the patch land in the public repo.

Wth-Esp, 2020-05-21:
Hi @devanlai 

There is already an MR to fix this issue, and will soon sync on Github.
Thanks for your support.

Thanks
Weitianhua

closed 2020-06-12 by igrr