espressif/esp-idf #5321: Broken build

sle118, 2020-05-19:
## Environment

- Development Kit:      N/A
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v4.0-386-gb0f053d82
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):  8.2.0 
- Operating System:     Windows and Linux
- (Windows only) environment type: Linux has the same issue
- Using an IDE?: N/A
- Power Supply:        N/A

## Problem Description
Build fails with the following errors
```
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/wpa_supplicant/libwpa_supplicant.a(crypto_mbedtls.c.obj):(.literal.crypto_bignum_mulmod+0x0): undefined reference to `esp_mpi_mul_mpi_mod'
/home/user/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/wpa_supplicant/libwpa_supplicant.a(crypto_mbedtls.c.obj): in function `crypto_bignum_mulmod':
/var/lib/jenkins/workspace/v4.0/components/wpa_supplicant/src/crypto/crypto_mbedtls.c:177: undefined reference to `esp_mpi_mul_mpi_mod'
```
### Expected Behavior
the project should link without an error

### Actual Behavior
linking errors when building

### Steps to reproduce

1. Clone the following repository
2. Build the project

### Code to reproduce this issue
N/A

## Debug Logs
N/A

## Other items if possible
clone the squeezelite-esp32 project and try to build against the most recent esp-idf v4.0 commit

projectgus, 2020-05-20:
Hi @sle118 ,

Thanks for providing this bug report.

To check I understand correctly: you're reporting this bug when building the `master-cmake` branch of [this project](https://github.com/sle118/squeezelite-esp32) with the described IDF version, yes?

It seems like there is a configuration combination which leads to build failure. The combination is:

* [MbedTLS is configured not to use hardware MPI acceleration](https://github.com/sle118/squeezelite-esp32/blob/master-cmake/sdkconfig#L505)
* [wpa_supplicant is configured to use mbedTLS API](https://github.com/sle118/squeezelite-esp32/blob/master-cmake/sdkconfig#L637)

We will fix this ASAP.

In the meantime, you can work around the bug by switching either of these config options (if possible to do so then [enabling mbedTLS MPI acceleration](https://docs.espressif.com/projects/esp-idf/en/release-v4.0/api-reference/kconfig.html#config-mbedtls-hardware-mpi) is the recommended way - should result in smaller binary size and most MPI operations will be faster.)

sle118, 2020-05-20:
@projectgus ,

Thank you for pointing me to the configuration issue; I have changed the sdkconfig and this has resolved that linking problem.  I chased the issue myself, but could not find a trace of the missing function anywhere. I guess this is because it's a chip function rather than an actual soft coded routine. 

Thank you for being so responsive!

projectgus, 2020-05-20:
Hi @sle118,

Glad you were able to resolve the problem. This is still a bug in IDF (any configuration combination that's allowed should be able to build), so I'm going to keep the issue open for now. It'll be closed automatically when the fix is merged.

> I chased the issue myself, but could not find a trace of the missing function anywhere. I guess this is because it's a chip function rather than an actual soft coded routine.

For reference of anyone who finds this issue looking for the same function, it's here (part of the port-specific layer for mbedTLS hardware acceleration, although in this case the function isn't part of the upstream mbedTLS API):
https://github.com/espressif/esp-idf/blob/release/v4.0/components/mbedtls/port/esp_bignum.c#L290

closed 2020-06-12 by igrr