espressif/esp-idf #5322: FAILED: esp-idf/mbedtls/x509_crt_bundle (Invalid certificate)

pc0808f, 2020-05-20:
INSTRUCTIONS
============

## Environment

- Build System:         [idf.py]
- Operating System:     [Windows]
- (Windows only) environment type: [Plain Command Prompt].
- Using an IDE?: [No]

## Problem Description

//Detailed problem description goes here.
When i build in IDF4.0 do not have this problem.

but when i build in IDF4.2 it happend.

Is there anyone know how to solve this problem?

### Expected Behavior
Just helloworld....

### Actual Behavior
[601/796] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle 
cmd.exe /C "cd /D D:\esp23\esp-idf\workspace\blink\build\esp-idf\mbedtls && D:\esp23\.espressif2\python_env\idf4.0_py3.8_env\Scripts\python.exe D:/esp23/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input D:/esp23/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q"
gen_crt_bundle.py: Invalid certificate in D:/esp23/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate

### Steps to reproduce

1. clone master
2. idf.py it compiler it

### Code to reproduce this issue
the origial helloworld.

## Debug Logs

```
[601/796] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle 
cmd.exe /C "cd /D D:\esp23\esp-idf\workspace\blink\build\esp-idf\mbedtls && D:\esp23\.espressif2\python_env\idf4.0_py3.8_env\Scripts\python.exe D:/esp23/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input D:/esp23/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q"
gen_crt_bundle.py: Invalid certificate in D:/esp23/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
```

ESP-Marius, 2020-05-20:
Hi @pc0808f, thanks for reporting this issue.

How did you setup the IDF environment for 4.2? (using the ESP-IDF shortcut? running export.bat?). I'm asking because I notice your python environment still points to the 4.0 install (see the line about \python_env\idf4.0_py3.8_env).

pc0808f, 2020-05-20:
I try using visual studio code to build.

and i got the same result.

If I disable the "Certificate Bundle" in menuconfig, then it can be build.

is it right to disable the "Certificate Bundle"?

-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "hello-world" version: 1
-- Adding linker script D:/esp-vs/esp-idf-42/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script D:/esp-vs/esp-idf-42/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script D:/esp-vs/esp-idf-42/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script D:/esp-vs/esp-idf-42/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script D:/esp-vs/esp-idf-42/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script D:/esp-vs/esp-idf-42/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script D:/esp-vs/workspace/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script D:/esp-vs/esp-idf-42/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script D:/esp-vs/esp-idf-42/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: D:/esp-vs/esp-idf-42/esp-idf/components/app_trace D:/esp-vs/esp-idf-42/esp-idf/components/app_update D:/esp-vs/esp-idf-42/esp-idf/components/asio D:/esp-vs/esp-idf-42/esp-idf/components/bootloader D:/esp-vs/esp-idf-42/esp-idf/components/bootloader_support D:/esp-vs/esp-idf-42/esp-idf/components/bt D:/esp-vs/esp-idf-42/esp-idf/components/cbor D:/esp-vs/esp-idf-42/esp-idf/components/coap D:/esp-vs/esp-idf-42/esp-idf/components/console D:/esp-vs/esp-idf-42/esp-idf/components/cxx D:/esp-vs/esp-idf-42/esp-idf/components/driver D:/esp-vs/esp-idf-42/esp-idf/components/efuse D:/esp-vs/esp-idf-42/esp-idf/components/esp-tls D:/esp-vs/esp-idf-42/esp-idf/components/esp32 D:/esp-vs/esp-idf-42/esp-idf/components/esp_adc_cal D:/esp-vs/esp-idf-42/esp-idf/components/esp_common D:/esp-vs/esp-idf-42/esp-idf/components/esp_eth D:/esp-vs/esp-idf-42/esp-idf/components/esp_event D:/esp-vs/esp-idf-42/esp-idf/components/esp_gdbstub D:/esp-vs/esp-idf-42/esp-idf/components/esp_hid D:/esp-vs/esp-idf-42/esp-idf/components/esp_http_client D:/esp-vs/esp-idf-42/esp-idf/components/esp_http_server D:/esp-vs/esp-idf-42/esp-idf/components/esp_https_ota D:/esp-vs/esp-idf-42/esp-idf/components/esp_https_server D:/esp-vs/esp-idf-42/esp-idf/components/esp_local_ctrl D:/esp-vs/esp-idf-42/esp-idf/components/esp_netif D:/esp-vs/esp-idf-42/esp-idf/components/esp_ringbuf D:/esp-vs/esp-idf-42/esp-idf/components/esp_rom D:/esp-vs/esp-idf-42/esp-idf/components/esp_serial_slave_link D:/esp-vs/esp-idf-42/esp-idf/components/esp_system D:/esp-vs/esp-idf-42/esp-idf/components/esp_timer D:/esp-vs/esp-idf-42/esp-idf/components/esp_websocket_client D:/esp-vs/esp-idf-42/esp-idf/components/esp_wifi D:/esp-vs/esp-idf-42/esp-idf/components/espcoredump D:/esp-vs/esp-idf-42/esp-idf/components/esptool_py D:/esp-vs/esp-idf-42/esp-idf/components/expat D:/esp-vs/esp-idf-42/esp-idf/components/fatfs D:/esp-vs/esp-idf-42/esp-idf/components/freemodbus D:/esp-vs/esp-idf-42/esp-idf/components/freertos D:/esp-vs/esp-idf-42/esp-idf/components/heap D:/esp-vs/esp-idf-42/esp-idf/components/idf_test D:/esp-vs/esp-idf-42/esp-idf/components/jsmn D:/esp-vs/esp-idf-42/esp-idf/components/json D:/esp-vs/esp-idf-42/esp-idf/components/libsodium D:/esp-vs/esp-idf-42/esp-idf/components/log D:/esp-vs/esp-idf-42/esp-idf/components/lwip D:/esp-vs/workspace/hello_world/main D:/esp-vs/esp-idf-42/esp-idf/components/mbedtls D:/esp-vs/esp-idf-42/esp-idf/components/mdns D:/esp-vs/esp-idf-42/esp-idf/components/mqtt D:/esp-vs/esp-idf-42/esp-idf/components/newlib D:/esp-vs/esp-idf-42/esp-idf/components/nghttp D:/esp-vs/esp-idf-42/esp-idf/components/nvs_flash D:/esp-vs/esp-idf-42/esp-idf/components/openssl D:/esp-vs/esp-idf-42/esp-idf/components/partition_table D:/esp-vs/esp-idf-42/esp-idf/components/perfmon D:/esp-vs/esp-idf-42/esp-idf/components/protobuf-c D:/esp-vs/esp-idf-42/esp-idf/components/protocomm D:/esp-vs/esp-idf-42/esp-idf/components/pthread D:/esp-vs/esp-idf-42/esp-idf/components/sdmmc D:/esp-vs/esp-idf-42/esp-idf/components/soc D:/esp-vs/esp-idf-42/esp-idf/components/spi_flash D:/esp-vs/esp-idf-42/esp-idf/components/spiffs D:/esp-vs/esp-idf-42/esp-idf/components/tcp_transport D:/esp-vs/esp-idf-42/esp-idf/components/tcpip_adapter D:/esp-vs/esp-idf-42/esp-idf/components/tinyusb D:/esp-vs/esp-idf-42/esp-idf/components/ulp D:/esp-vs/esp-idf-42/esp-idf/components/unity D:/esp-vs/esp-idf-42/esp-idf/componen [... cut ...]
FAILED: esp-idf/mbedtls/x509_crt_bundle 
cmd.exe /C "cd /D D:\esp-vs\workspace\hello_world\build\esp-idf\mbedtls && d:\esp-vs\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe D:/esp-vs/esp-idf-42/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input D:/esp-vs/esp-idf-42/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q"
gen_crt_bundle.py: Invalid certificate in D:/esp-vs/esp-idf-42/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
ninja: build stopped: subcommand failed.

ESP-Marius, 2020-05-22:
You shouldnt have to disable it, but havent been able to reproduce the error on my end.

Could do the following to help me debug it:

- Do `pip3 show cryptography` and `pip show cryptography` in your terminal/cmd line before building and show me what it outputs.
- Attach the file you have at `D:/esp-vs/esp-idf-42/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem` to your github comment

Thanks for your help!

xuhongv, 2020-06-10:
Hi @ESP-Marius ，this is above ：

[image]

[cacrt_all.zip](https://github.com/espressif/esp-idf/files/4758258/cacrt_all.zip)

ESP-Marius, 2020-06-11:
@xuhongv Thanks, internally we also found some issues due to encoding. This should be fixed when commit: c39b3730bf05da2cba1e3bbce5e368a6543f5863 is merged and synced to Github

ESP-Marius, 2020-08-07:
Closed by [c343323863c55f8e1](https://github.com/espressif/esp-idf/commit/c343323863c55f8e1de571b65384ffb411666e85)

closed 2020-08-07 by ESP-Marius

imammashur, 2021-09-30:
Hello everyone, I still got the error. After setting the cert file to UNIX, it is still return an error. Does anybody know the solution? Thank you.

Typoception, 2021-09-30:
same. im having the same error 
Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle 
cmd.exe /C "cd /D C:\Users\yahsa\Desktop\NTU\FYP\ESP_Workspace\blink\build\esp-idf\mbedtls && C:\Users\yahsa\esp\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe C:/Users/yahsa/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input C:/Users/yahsa/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q"
gen_crt_bundle.py: Invalid certificate in C:/Users/yahsa/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
ninja: build stopped: subcommand failed.

anyone knows how to solve this?

ingcicala, 2021-09-30:
same now with new IDF 4.3.1
---
[1/307] cmd.exe /C "cd /D C:\Users\ingfe\eclipse-workspace\esp32_eclipsetest\build\bootloader && C:\Espressif\ESP_Tool\.espressif\tools\cmake\3.16.4\bin\cmake.exe --build ."
ninja: no work to do.
[2/305] cmd.exe /C "cd /D C:\Users\ingfe\eclipse-workspace\esp32_eclipsetest\build\esp-idf\mbedtls && python C:/Espressif/ESP_idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input C:/Espressif/ESP_idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q"
FAILED: esp-idf/mbedtls/x509_crt_bundle 
cmd.exe /C "cd /D C:\Users\ingfe\eclipse-workspace\esp32_eclipsetest\build\esp-idf\mbedtls && python C:/Espressif/ESP_idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input C:/Espressif/ESP_idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q"
gen_crt_bundle.py: Invalid certificate in C:/Espressif/ESP_idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
ninja: build stopped: subcommand failed.
---

RobertWCarey, 2021-09-30:
Same issue. ESP-IDF v4.3.1 running ubuntu 20.04.
[cacrt_all.zip](https://github.com/espressif/esp-idf/files/7259295/cacrt_all.zip)

I was able to build by disabling the "certificate bundle" in the menuconfig as mentioned above.

```
[ 70%] Linking CXX static library libmbedtls.a
[ 70%] Built target mbedtls
[ 70%] Generating x509_crt_bundle
gen_crt_bundle.py: Invalid certificate in /home/rob/swdev/esp-idf-v4.3.1/components/mbedtls/esp_crt_bundle/cacrt_all.pem
Invalid certificate
make[2]: *** [esp-idf/mbedtls/CMakeFiles/__idf_mbedtls.dir/build.make:67: esp-idf/mbedtls/x509_crt_bundle] Error 2
make[1]: *** [CMakeFiles/Makefile2:2546: esp-idf/mbedtls/CMakeFiles/__idf_mbedtls.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
make failed with exit code 2
```

```
% pip3 show cryptography                                                                                                                                                                                                                                                
Name: cryptography
Version: 2.8
Summary: cryptography is a package which provides cryptographic recipes and primitives to Python developers.
Home-page: https://github.com/pyca/cryptography
Author: The cryptography developers
Author-email: cryptography-dev@python.org
License: BSD or Apache License, Version 2.0
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
 ~
% pip show cryptography                                                                                                                                                                                                                                                 
Name: cryptography
Version: 2.8
Summary: cryptography is a package which provides cryptographic recipes and primitives to Python developers.
Home-page: https://github.com/pyca/cryptography
Author: The cryptography developers
Author-email: cryptography-dev@python.org
License: BSD or Apache License, Version 2.0
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by: 
```

jipi, 2021-09-30:
i tried converting the cacrt_all.pem file to utf-8 encodingbut it says cannot convert. there seems to be a line with this (line 1149)
NetLock Arany (Class Gold) Főtanúsítvány

$ iconv -f US-ASCII -t UTF-8 cacrt_all.pem > cacrt_all.pem.utf8
iconv: cacrt_all.pem:1149:28: cannot convert

tomasmcguinness, 2021-09-30:
I found this - https://github.com/espressif/esp-idf/issues/7621

The certificates have expired today. This explains how to disable the certificates.

italocjs, 2021-09-30:
This error appeared again, i cant build even exemples (like blink),

electrocucaracha, 2021-10-01:
To be more precisely, the [EC-ACC certificate](https://github.com/espressif/esp-idf/blob/v4.3/components/mbedtls/esp_crt_bundle/cacrt_all.pem#L1648-L1676) is invalid. It works after I removed it.

andy31415, 2021-10-01:
A possible hackish workaround for us was to downgrade the cryptography package (35.0.0 was just released, the previous  version  seems to  'work'):

```
/opt/espressif/tools/python_env/idf4.3_py3.9_env/bin/pip uninstall -y cryptography 
/opt/espressif/tools/python_env/idf4.3_py3.9_env/bin/pip install cryptography==3.4.8 
```

substituting the correct path (it seems esp tools at  least for us do not use the host pip packages directly)

Fl4v, 2021-10-01:
Thanks @andy31415, it worked for me.

electrocucaracha, 2021-10-01:
> To be more precisely, the [EC-ACC certificate](https://github.com/espressif/esp-idf/blob/v4.3/components/mbedtls/esp_crt_bundle/cacrt_all.pem#L1648-L1676) is invalid. It works after I removed it.

The EC-ACC certificate contains a negative serial number:
```
$ openssl x509 -in ec-acc.pem -text                                                                                               
Certificate:                                                    
    Data:                                                       
        Version: 3 (0x2)                                        
        Serial Number:                                          
             (Negative)11:d4:c2:14:2b:de:21:eb:57:9d:53:fb:0c:22:3b:ff
        Signature Algorithm: sha1WithRSAEncryption
```
According the the [cryptography release doc](https://github.com/pyca/cryptography/blob/35.0.0/CHANGELOG.rst) those values are prohibited.

dvillalva21, 2021-10-02:
Hello guys, I am also experiencing the same issue. I attached my compile code. 
I am on a Windows 10 machine, and reinstalled Eclipse twice (fully). Nothing works. Hope I could receive advice on how to fix the bug.

[663 cert error.txt](https://github.com/espressif/esp-idf/files/7271113/663.cert.error.txt)

italocjs, 2021-10-02:
> Hello guys, I am also experiencing the same issue. I attached my compile code. I am on a Windows 10 machine, and reinstalled Eclipse twice (fully). Nothing works. Hope I could receive advice on how [...]
 
This can help.
https://user-images.githubusercontent.com/9244531/135537815-125dc605-1789-4d8d-84ea-1423b9b31b76.png

Similar issues:
#7621 ,  #7624  , #7631

electrocucaracha, 2021-10-02:
I have submitted a fix [here](https://github.com/espressif/esp-idf/pull/7632)

volkanunal, 2021-10-03:
> I have submitted a fix [here](https://github.com/espressif/esp-idf/pull/7632)

You saved my day!

dvillalva21, 2021-10-04:
> This can help. https://user-images.githubusercontent.com/9244531/135537815-125dc605-1789-4d8d-84ea-1423b9b31b76.png
>
> Similar issues: #7621 , #7624 , #7631

Is there a way to do this on Eclipse IDE? I am not yet familiar with VS code and mid project.

italocjs, 2021-10-04:
> Is there a way to do this on Eclipse IDE? I am not yet familiar with VS code and mid project.

IFAIK the sdkconfig file is the same, you can edit it with any text editor, or run a command on esp-idf cmd that opens a configuration GUI, not sure which command as i never used it.

dvillalva21, 2021-10-05:
> > > > Hello guys, I am also experiencing the same issue. I attached my compile code. I am on a Windows 10 machine, and reinstalled Eclipse twice (fully). Nothing works. Hope I could receive advice on how to fix the bug.
> > > > [663 cert error.txt](https://github.com/espressif/esp-idf/files/7271113/663.cert.error.txt)
> > > 
> > > 
> > > This can help. https://user-images.githubusercontent.com/9244531/135537815-125dc605-1789-4d8d-84ea-1423b9b31b76.png
> > > Similar issues: #7621 , #7624 , #7631
> > 
> > 
> > Is there a way to do this on Eclipse IDE? I am not yet familiar with VS code and mid project.
> 
> IFAIK the sdkconfig file is the same, you can edit it with any text editor, or run a command on esp-idf cmd that opens a configuration GUI, not sure which command as i never used it.
> 
> you can also download the changed file from this commit and paste it into you pc. it will also fix this

So I ended up finding the sdkconfig file and made changes to only check the basic certificates and it compiled! So happy. Thank you for the help! 

[image]

Ro8ch, 2021-10-05:
Hi Guys,
This was not obvious to me at first but this is what I did to fix this error on Ubuntu:
1. nano into sdkconfig file <your project>$ nano sdkconfig
2. Find "Certificate Bundle" section
3. Comment out "CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL" (add a "#" in front)
4. Uncomment "CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN" (remove the "#")
5. Make sure the line displays "CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y"
6. Save
7. Build again (idf.py build)

That should solve the issue without having to check all pip cryptography stuff.

Hope this helps someone.

farisikhmal, 2021-10-06:
> Hi Guys, This was not obvious to me at first but this is what I did to fix this error on Ubuntu:
> 
>     1. nano into sdkconfig file $ nano sdkconfig
> 
>     2. Find "Certificate Bundle" section
> 
>     3. Comment out "CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL" (add a "#" in front)
> 
>     4. Uncomment "CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN" (remove the "#")
> 
>     5. Make sure the line displays "CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y"
> 
>     6. Save
> 
>     7. Build again (idf.py build)
> 
> 
> That should solve the issue with

> Hi Guys, This was not obvious to me at first but this is what I did to fix this error on Ubuntu:
> 
>     1. nano into sdkconfig file $ nano sdkconfig
> 
>     2. Find "Certificate Bundle" section
> 
>     3. Comment out "CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL" (add a "#" in front)
> 
>     4. Uncomment "CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN" (remove the "#")
> 
>     5. Make sure the line displays "CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y"
> 
>     6. Save
> 
>     7. Build again (idf.py build)
> 
> 
> That should solve the issue without having to check all pip cryptography stuff.
> 
> Hope this helps someone.

THANKS DUDE, IT WORKS !!!

Ro8ch, 2021-10-09:
Hi @ZeRico ,
Did you make sure to save the file within Nano.
It's not mentioned at the bottom of the screen when in nano but ctrl+S still works.

Try that before closing up and right after reopen the file and confirm if the file changed.

Let me know if this does not work then I can help look for some other way.

williamsokol, 2021-10-10:
@ZeRico I had same issue until if fixed the spacing at the beginning to look more like the other lines

I was doing:
```
#CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=y
 CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y
```
I changed it to:
```
# CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_FULL=y
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y
```

Sreedharbot, 2021-10-10:
@williamsokol 

Thanks for point it out.. It works !!

ZeRico, 2021-10-11:
hi ther!
finally I managed to make I work!

probably that space before the sentence was the issue.
thanks everybody for you help

andershedberg, 2021-10-16:
For anyone using Visual Code on Win10 and following the Getting Started guide, and ends up on this page :-)

Select: View->Command Palette
Enter: ESP-IDF SDK Configuration Editor (menuconfig)
In the search box on the menuconfig page, enter: "Certificate Bundle"
Change setting for "Default certificate bundle options" from "Use the full default certificate bundle" to "Use only the most common certificates from the default bundles"

Voila! Thanks all for the above suggestions!

moefear85, 2021-10-31:
this problem suddenly appeared for me today

cvnli, 2021-11-03:
> same. im having the same error Generating x509_crt_bundle FAILED: esp-idf/mbedtls/x509_crt_bundle cmd.exe /C "cd /D C:\Users\yahsa\Desktop\NTU\FYP\ESP_Workspace\blink\build\esp-idf\mbedtls && [...]

miLORD1337, 2021-11-05:
You can do it with `idf.py menuconfig` as follows:
Navigate to `Component config --> mbedTLS --> Certificate Bundle --> Default certificate bundle options` and choose `use only most common certificates`, then build again. Good luck!

LucasEstevo, 2021-11-09:
> 
> 
> Para qualquer pessoa que use o Visual Code no Win10 e siga o guia de primeiros passos e acabe nesta página :-)
> 
> Selecione: Exibir-> Paleta de Comandos Digite: ESP-IDF SDK Configuration Editor (menuconfig) Na caixa de pesquisa da página menuconfig, digite: "Pacote de certificado" Altere a configuração de "Opções de pacote de certificado padrão" de "Usar o pacote de certificado padrão completo" para "Usar apenas os certificados mais comuns dos pacotes padrão"
> 
> Voila! Obrigado a todos pelas sugestões acima!

Segui esta recomendação e funcionou perfeitamente, muito obrigado!

yupanqui, 2021-11-12:
> You can do it with `idf.py menuconfig` as follows: Navigate to `Component config --> mbedTLS --> Certificate Bundle --> Default certificate bundle options` and choose `use only most common certificates`, then build again. Good luck!

yupanqui, 2021-11-12:
@miLORD1337  Well done man, your hint is the only that works out of Eclipse or VSCode. Of course the editing of the file "sdkconfig" is no sense as it is clearly stated the file is automatically created each time by the framework, so the patch is overridden.
Thanks a lot !!!

connorfuhrman, 2021-11-16:
Experienced this issue using ESP-MDF (https://github.com/espressif/esp-mdf/tree/cf502740f5a6c82a0dc3059e7591c262795f70b2) with ESP-IDF 4.3.1. Described fix by @miLORD1337 worked for me as well. 

What are the side effects of “use only most recent certificates”?