espressif/esp-idf #5323: Smartconfig does not work if device reboot from running AP mode

AxelLin, 2020-05-20:
## Environment

- Module or chip used: ESP32-WROOM-32 
- IDF version: v4.2-dev-1575-ga3520970f633
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: NO
- Power Supply: USB

## Problem Description

The problem:
I modify the smart_config example code to reproduce the issue:

The application will randomly run either softap or start smartconfig.

1. run softap, wait 1 sec then call esp_restart() to reboot
2. if device boot to smartconfig this time, it won't enter sniffer mode.
   The latest log stop at:
   I (843) smartconfig: SC version: V2.9.0

Note, this only happen when calling esp_restart() to reboot.
If run softap and press reset button, then it works if boot to smartconfg.

Below is the boot log -- running AP first, call esp_restart() then reboot to run smartconfig.

entry 0x40080688
I (31) boot: ESP-IDF v4.2-dev-1575-ga3520970f633-dir 2nd stage bootloader
I (31) boot: compile time 15:59:46
I (32) boot: chip revision: 1
I (36) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (43) boot.esp32: SPI Speed      : 40MHz
I (48) boot.esp32: SPI Mode       : DIO
I (52) boot.esp32: SPI Flash Size : 2MB
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (107) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x17e8c ( 97932) map
I (153) esp_image: segment 1: paddr=0x00027eb4 vaddr=0x3ffb0000 size=0x039a0 ( 14752) load
I (159) esp_image: segment 2: paddr=0x0002b85c vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/axel/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (160) esp_image: segment 3: paddr=0x0002bc68 vaddr=0x40080404 size=0x043b0 ( 17328) load
I (176) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x78914 (493844) map
0x400d0020: _stext at ??:?

I (365) esp_image: segment 5: paddr=0x000a893c vaddr=0x400847b4 size=0x1144c ( 70732) load
0x400847b4: esp_mbedtls_mem_calloc at /home/axel/esp/esp-idf/components/mbedtls/port/esp_mem.c:42

I (408) boot: Loaded app from partition at offset 0x10000
I (408) boot: Disabling RNG early entropy source...
I (409) cpu_start: Pro cpu up.
I (412) cpu_start: Application information:
I (417) cpu_start: Project name:     smart_config
I (423) cpu_start: App version:      v4.2-dev-1575-ga3520970f633-dir
I (430) cpu_start: Compile time:     May 19 2020 15:59:33
I (436) cpu_start: ELF file SHA256:  d07a49aa70f4ca9e...
I (442) cpu_start: ESP-IDF:          v4.2-dev-1575-ga3520970f633-dir
I (449) cpu_start: Starting app cpu, entry point is 0x400818b8
0x400818b8: call_start_cpu1 at /home/axel/esp/esp-idf/components/esp32/cpu_start.c:286

I (440) cpu_start: App cpu up.
I (459) heap_init: Initializing. RAM available for dynamic allocation:
I (466) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
[... 107 lines cut ...]
I (663) wifi:Init dynamic tx buffer num: 32
I (663) wifi:Init data frame dynamic rx buffer num: 32
I (673) wifi:Init management frame dynamic rx buffer num: 32
I (673) wifi:Init management short buffer num: 32
I (683) wifi:Init static rx buffer size: 1600
I (683) wifi:Init static rx buffer num: 10
I (683) wifi:Init dynamic rx buffer num: 32
I (783) phy: phy_version: 4181, c9a03e2, Mar 10 2020, 12:45:06, 0, 0
I (783) wifi:mode : sta (c8:2b:96:ae:4d:00)
I (843) smartconfig: SC version: V2.9.0

### Expected Behavior
smartconfig should work no matter calling esp_restart() to reboot or press reset button to reboot.

### Actual Behavior
smartconfig does not work if running softAP first then calling esp_restart to reboot.
### Steps to reproduce

Just run the attached diff for smart_config example code.

### Code to reproduce this issue
Attached the diff of my modification to reporduce the issue.
(Mainly combine softAP example with smart_config example)

[smart_config_issue.diff.txt](https://github.com/espressif/esp-idf/files/4654797/smart_config_issue.diff.txt)

Alvin1Zhang, 2020-05-20:
@AxelLin Thanks for the report. We will look into.

zhangyanjiaoesp, 2020-06-03:
@AxelLin 
I have reproduced your problem. And will reply you as soon as we found the root cause, thanks.

closed 2020-06-12 by igrr

chegewara, 2020-07-14:
Is this backported to v3.3?

zhangyanjiaoesp, 2020-07-14:
v3.3 doesn't have this problem, don't need to backport.