espressif/esp-idf #5324: mbedtls error after reading a lot of font files from spiffs

korstiaanS, 2020-05-20:
Hi,

In v3.3 I have the following problem:

I'm using OTA together with reading fonts from a SPIFFS filesystem.
They work fine if I don't read to many files but if I want to read all my font files (= a lot of malloc) my OTA won't work anymore and I get the following error:

`E (6198744) esp-tls: mbedtls_ssl_handshake returned -0x4290`

So if I for example only read these font files:

```
readFontFromSpiffs("/spiffs/fonts/lucidaconsole8", &fnt_lucidaconsole8);            // 7.5
readFontFromSpiffs("/spiffs/fonts/lucidaconsole20bold", &fnt_lucidaconsole20bold);  // 7.5
readFontFromSpiffs("/spiffs/fonts/lucidaconsole24bold", &fnt_lucidaconsole24bold);  // 2.9
readFontFromSpiffs("/spiffs/fonts/lucidaconsole30bold", &fnt_lucidaconsole30bold);  // 7.5
readFontFromSpiffs("/spiffs/fonts/lucidaconsole52bold", &fnt_lucidaconsole52bold);  // 7.5
readFontFromSpiffs("/spiffs/fonts/verdana8", &fnt_verdana8);                        // 7.5
readFontFromSpiffs("/spiffs/fonts/tahoma12bold", &fnt_tahoma12bold);  // 7.5
readFontFromSpiffs("/spiffs/fonts/couriernew7", &fnt_couriernew7);    // 7.5
```

it all works fine but if I read all these the OTA won't work and gives the mbedtls error:

```
readFontFromSpiffs("/spiffs/fonts/arial7", &fnt_arial7);
readFontFromSpiffs("/spiffs/fonts/arial8", &fnt_arial8);
readFontFromSpiffs("/spiffs/fonts/arial10", &fnt_arial10);
readFontFromSpiffs("/spiffs/fonts/arial10bold", &fnt_arial10bold);
readFontFromSpiffs("/spiffs/fonts/arial12", &fnt_arial12);
readFontFromSpiffs("/spiffs/fonts/consolas24bold", &fnt_consolas24bold);
readFontFromSpiffs("/spiffs/fonts/lucidaconsole8", &fnt_lucidaconsole8);            // 7.5
readFontFromSpiffs("/spiffs/fonts/lucidaconsole20bold", &fnt_lucidaconsole20bold);  // 7.5
readFontFromSpiffs("/spiffs/fonts/lucidaconsole22bold", &fnt_lucidaconsole22bold);
readFontFromSpiffs("/spiffs/fonts/lucidaconsole24bold", &fnt_lucidaconsole24bold);  // 2.9
readFontFromSpiffs("/spiffs/fonts/lucidaconsole28bold", &fnt_lucidaconsole28bold);
readFontFromSpiffs("/spiffs/fonts/lucidaconsole30bold", &fnt_lucidaconsole30bold);  // 7.5
readFontFromSpiffs("/spiffs/fonts/lucidaconsole52bold", &fnt_lucidaconsole52bold);  // 7.5
readFontFromSpiffs("/spiffs/fonts/verdana8", &fnt_verdana8);                        // 7.5
readFontFromSpiffs("/spiffs/fonts/verdana8italic", &fnt_verdana8italic);
readFontFromSpiffs("/spiffs/fonts/verdana9", &fnt_verdana9);
readFontFromSpiffs("/spiffs/fonts/verdana10", &fnt_verdana10);
readFontFromSpiffs("/spiffs/fonts/tahoma12bold", &fnt_tahoma12bold);  // 7.5
readFontFromSpiffs("/spiffs/fonts/couriernew7", &fnt_couriernew7);    // 7.5
```

I checked the stack and heap and there is still plenty of it. 
I also use SPI RAM and have still 4194052 bytes free.

Somewhere it looks there is a conflict between these 2? (malloc and mbedtls)

Can somebody help?

Thanks,
Korstiaan

chegewara, 2020-05-20:
https://github.com/espressif/esp-idf/issues/222

korstiaanS, 2020-05-20:
I found 2 ways (workaround?) to solve this:

1. Change Memory allocation strategy from Internal memory to External SPI RAM for mbedTLS
2. My fonts are 1140 bytes or higher so change Maximum malloc() size, in bytes, to always put in internal memory from 16384 to 1024. (SPI RAM config)

Using 1 of these will fix the problem. Solution 2 makes loading the fonts a lot slower.

It looks like the internal RAM gets to full and the automatic overload to external RAM is not good or too late and causes strange issues?
Anyway, it is not normal that this is not handled automatically?

Korstiaan

AdityaHPatwardhan, 2020-07-06:
@korstiaanS I am not able to produce this issue myself, I think it might have been with the memory usage and mbedtls not having enough memory to perform the handshake.Thank you for suggesting possible solutions. If you may share the bin file for the OTA, or some code with which I am able to reproduce this, I will try to reproduce it and see any other options we may able to solve it

korstiaanS, 2020-07-06:
Hi,

Thanks for trying to help but the whole setup is so complex that it is difficult to reproduce for you.

closed 2020-07-06 by korstiaanS