espressif/esp-idf #5325: uart_wait_tx_done exit with buffer not empty

moggiozzi, 2020-05-20:
Reopening https://github.com/espressif/esp-idf/issues/1002

## Environment

- Development Kit:      Custom board
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): v4.1-beta1-266-g56c1646e2
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply:         USB

## Problem Description
We use chip with DE pin to control data transfer direction (receive or transmit).
Before send data DE pin set to transmit and after to receive mode.
uart_wait_tx_done() return immediately and data transfer not completed.

### Expected Behavior
uart_wait_tx_done() return after all data send

### Actual Behavior
uart_wait_tx_done() return immediately

### Steps to reproduce
1. use uart_wait_tx_done() right after uart_write_bytes()

DE pin - red line
TX pin - yellow line
[image]

### Code to reproduce this issue
Sample poject https://github.com/moggiozzi/esp_uart_test
```c
    usartClientSetModeTx();//change DE pin
    uart_write_bytes(UART_NUM_1, (char *)pData, dataLength);
    uart_wait_tx_done(UART_NUM_1, MB_SERIAL_TX_TOUT_TICKS);
    usartClientSetModeRx();//change DE pin
```
## Other
I think the problem is that the function checks the hardware module, but does not check the circular buffer (p_uart_obj[uart_num]->tx_ring_buf).

koobest, 2020-05-20:
Hi, @moggiozzi 
I think is that case, can you  refer to uart_echo_rs485 example ?

Alvin1Zhang, 2020-06-02:
@moggiozzi Thanks for reporting, would you please help share if any updates for the issue? Thanks.

moggiozzi, 2020-08-18:
> 
> 
> @moggiozzi Thanks for reporting, would you please help share if any updates for the issue? Thanks.

I will try it in rs485 mode and inform results.

moggiozzi, 2020-08-18:
In rs485 mode all works fine, but I had to patch SDK:
esp-idf\components\soc\include\hal\uart_hal.h
---#define uart_hal_set_rts(hal, active_level)  uart_ll_set_rts_active_level((hal)->dev, active_level)
+++#define uart_hal_set_rts(hal, active_level)  uart_ll_set_rts_active_level((hal)->dev, (active_level == 0))
because on my board RTS(DE pin) inversed.
Is it possible to configure RTS pin inversion in the uart driver?

moggiozzi, 2020-08-27:
> Is it possible to configure RTS pin inversion in the uart driver?
Yes
GPIO.func_out_sel_cfg[DE_GPIO_NUM].inv_sel = 1;

closed 2020-08-27 by moggiozzi