espressif/esp-idf #5326: esp_a2d_sink_connect incorrect behaviour - very long timeout

l0ud, 2020-05-20:
## Environment

- Module or chip used: ESP32-WROOM-32D
- IDF version 4.0
- Build System:         idf.py
- Compiler version: 8.2.0
- Operating System:     Linux

- Using an IDE?: No
- Power Supply:  external 3.3V

## Problem Description

Hi, I'm trying to make a2dp speaker which connects to last device after powering up.
Problem happens when:
1. Esp32 is connected to phone, code from a2dp_sink example with modification described bellow.
2. User turns off esp32 by disconnecting power.
3. User quickly turns on esp32 again.

After calling esp_a2d_sink_connect() esp32 will try to connect to phone, but phone still thinks that speaker is connected. In a result, esp_a2d_sink_connect will wait a long time (around 20-30 seconds) 

It will not fail every time, it fails pretty constantly when phone is locked (screen off).
I reproduced that issue with android and ios.

### Expected Behavior
I would expect esp_a2d_sink_connect() to timeout after much shorter period of time (so I could retry connect request) or just to be able to connect to the phone. Ability to adjust that timeout would be useful.
### Actual Behavior
esp_a2d_sink_connect will wait for around 20-30 seconds and eventually fail to connect to the phone. 
### Steps to reproduce
1. Turn on modified a2dp_sink example.
2. Connect to phone, turn on music, lock phone when the music is still playing.
2. Just reboot esp32, and check logs.

### Code to reproduce this issue
i2dp_sink example, with esp_a2d_sink_connect() call added in esp_a2d_sink_connect BT_APP_EVT_STACK_UP function.

last_addr is read from nvs earlier, and it's address to the phone.

```cpp
static void esp_a2d_sink_connect(uint16_t event, void *p_param)
{
    ESP_LOGD(BT_AV_TAG, "%s evt %d", __func__, event);
    switch (event) {
    case BT_APP_EVT_STACK_UP: {
    .....
        /* set discoverable and connectable mode, wait to be connected */
        esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
        if (last_addr[0] || last_addr[1] || last_addr[2] || last_addr[3] || last_addr[4] || last_addr[5])
        {
            esp_err_t status = esp_a2d_sink_connect(last_addr);
            ESP_LOGI(BT_AV_TAG, "Last bda scheduled connection status 0x%x", status);
        }
        break;
```
## Debug Logs

```
I (390) BTDM_INIT: BT controller compile version [79dd6f1]
I (400) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (510) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1040) BT_AV: Last bda scheduled connection status 0x0
I (1050) BT_AV: event: 10
I (1060) BT_AV: event: 10
I (1060) BT_AV: event: 10
I (1060) BT_AV: event: 10
E (1060) BT_APPL: reset flags
I (1060) BT_AV: A2DP connection state: Connecting, [xxx]

LONG PAUSE

W (61060) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2
I (61060) BT_AV: A2DP connection state: Disconnected, [xxx]
```

l0ud, 2020-05-20:
Also, which is more important I see that esp_a2d_sink_connect sometimes CAN connect, but it's not passing any sound and disconnecting after a while (mostly after track change).

```c++
I (1070) BT_AV: A2DP connection state: Connecting, [xxx]
E (2040) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (2140) BT_AV: A2DP audio stream configuration, codec type 0
I (2140) I2S: APLL: Req RATE: 44100, real rate: 44099.988, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11289597.000, SCLK: 1411199.625000, diva: 1, divb: 0
I (2150) BT_AV: Configure audio player 21-15-2-35
I (2150) BT_AV: Audio player configured, sample rate=44100
W (2190) BT_APPL: new conn_srvc id:19, app_id:0
I (2190) BT_AV: A2DP connection state: Connected, [xxx]
I (2380) RCCT: AVRC conn_state evt: state 1, [xxx]
I (2390) RCCT: remote rn_cap: count 6, bitmask 0xe26
I (2460) RCCT: AVRC metadata rsp: attribute id 0x1, On the Border - 2001 Remaster
I (2460) RCCT: AVRC metadata rsp: attribute id 0x2, Al Stewart
I (2470) RCCT: AVRC metadata rsp: attribute id 0x4, Year Of The Cat
I (2470) RCCT: AVRC metadata rsp: attribute id 0x20,
I (2480) RCTG: AVRC register event notification: 13, param: 0x0
E (5390) BT_APPL: bta_dm_pm_btm_status hci_status=12
I (6270) BT_LOG: bta_av_link_role_ok hndl:x41 role:0 conn_audio:x1 bits:1 features:x824b

W (6270) BT_APPL: new conn_srvc id:19, app_id:1
I (6270) BT_AV: A2DP audio state: Started
I (6310) RCCT: AVRC event notification: 1
I (6310) BT_AV: Playback status changed: 0x1
I (14140) RCCT: AVRC event notification: 2
I (14160) RCCT: AVRC event notification: 5
I (14160) BT_AV: Play position changed: 2-ms
E (21250) BT_APPL: bta_av_str_stopped:audio_open_cnt=1, p_data 0x0
W (21250) BT_AVCT: avct_lcb_last_ccb
W (21250) BT_AVCT: 0: aloc:1, lcb:0x0/0x3ffca3e8, ccb:0x3ffca450/0x3ffca468
W (21250) BT_AVCT: 1: aloc:1, lcb:0x3ffca3e8/0x3ffca3e8, ccb:0x3ffca468/0x3ffca468
W (21260) BT_AVCT: 2: aloc:0, lcb:0x0/0x3ffca3e8, ccb:0x3ffca480/0x3ffca468
E (21270) BT_APPL: bta_dm_pm_btm_status hci_status=12
I (21280) BT_AV: A2DP audio state: Stopped
I (21280) BT_AV: A2DP connection state: Disconnected, [xxx]
I (31250) RCCT: AVRC conn_state evt: state 0, [00:00:00:00:00:00]
```

boblane1, 2020-10-14:
Hi, @l0ud 
Thanks for your issue.

For the first problem, sorry to be unable to offer you a good solution.  In Bluetooth, higher layer channel establishment protocols should be designed to restrict timeouts to be 30 seconds or longer. Besides, the time cost of reconnection is also affected by the Bluetooth stack in the phone. For example, the new a2dp connection will not be executed until the previous connection has been
completely disconnected.

For the second problem, maybe you can pull the latest master and try to reproduce this issue. If it still exists, we will look into this as soon as possible.

closed 2021-01-21 by Alvin1Zhang