espressif/esp-idf #5329: cancel an ota started with esp_ota_begin

aep, 2020-05-21:
**Is your feature request related to a problem? Please describe.**

esp_ota_end is the only way to deallocate memory used by calling esp_ota_begin.

it is not documented if its ok to call esp_ota_end when actually the ota must be cancled for example due to connection loss, so i'm assuming it is not.

**Describe the solution you'd like**

- either document that esp_ota_end doesnt actually have any side effects other than deallocation
- or add a flag to esp_ota_end that allows indicating if the ota should be canceled
- or add a new function that deallocates the handle without other effects

lucastcox, 2020-05-21:
> it is not documented if its ok to call esp_ota_end when actually the ota must be cancled for example due to connection loss, so i'm assuming it is not

Why assume it is not okay to call esp_ota_end when you can simply look at the code for esp_ota_end and find out? It's neither a long nor an overly-complicated function (well, in my opinion, that is. Also, apologies if this comes across harshly; that is not at all my intent).

According to the function below, if any of the checks validating the partition pointed to by the handle fails, then nothing will happen except an error be returned and memory cleaned up. It may also write the "final" 16 bytes of the ota before the final validation, but that will have no consequence because the next ota attempt will erase that partition. 

However, @Espressif, it might not be a bad idea to have simple esp_ota_abort(esp_ota_handle_t handle) to handle this specific situation. If a caller knows an ota failed, it could skip the esp_image_verify step and free up the memory used

```c
esp_err_t esp_ota_end(esp_ota_handle_t handle)
{
    ota_ops_entry_t *it;
    esp_err_t ret = ESP_OK;

    for (it = LIST_FIRST(&s_ota_ops_entries_head); it != NULL; it = LIST_NEXT(it, entries)) {
        if (it->handle == handle) {
            break;
        }
    }

    if (it == NULL) {
        return ESP_ERR_NOT_FOUND;
    }

    /* 'it' holds the ota_ops_entry_t for 'handle' */

    // esp_ota_end() is only valid if some data was written to this handle
    if ((it->erased_size == 0) || (it->wrote_size == 0)) {
        ret = ESP_ERR_INVALID_ARG;
        goto cleanup;
    }

    if (it->partial_bytes > 0) {
        /* Write out last 16 bytes, if necessary */
        ret = esp_partition_write(it->part, it->wrote_size, it->partial_data, 16);
        if (ret != ESP_OK) {
            ret = ESP_ERR_INVALID_STATE;
            goto cleanup;
        }
        it->wrote_size += 16;
        it->partial_bytes = 0;
    }

    esp_image_metadata_t data;
    const esp_partition_pos_t part_pos = {
      .offset = it->part->address,
      .size = it->part->size,
    };

    if (esp_image_verify(ESP_IMAGE_VERIFY, &part_pos, &data) != ESP_OK) {
        ret = ESP_ERR_OTA_VALIDATE_FAILED;
        goto cleanup;
    }

 cleanup:
    LIST_REMOVE(it, entries);
    free(it);
    return ret;
}
```

aep, 2020-05-21:
> Why assume it is not okay to call esp_ota_end when you can simply look at the code

undocumented behavior can change. in particular if the functions intend is to always have side effects, someone might assume its ok to add more subtle side effects later.

in particular, it might assume that if all internal checks pass, the image is valid, although it isn't because we have additional signature checks and there is currently no way to signal to esp_ota that they failed

lucastcox, 2020-05-21:
Ah, yeah, good point about image validation outside of the built-in validation. 

In the meantime, you could make a PR for "esp_ota_abort," or instead (not a great workaround, I'll admit) write known bad data to the partition if your validation fails an then call esp_ota_end.

This **should** be all you need if you want to implement it.
```C

static ota_ops_entry_t *get_ota_ops_entry(esp_ota_handle_t handle)
{
    ota_ops_entry_t *it;
    for (it = LIST_FIRST(&s_ota_ops_entries_head); it != NULL; it = LIST_NEXT(it, entries)) {
        if (it->handle == handle) {
            break;
        }
    }
   return it;
}

esp_err_t esp_ota_abort(esp_ota_handle_t handle)
{
    ota_ops_entry_t *it = get_ota_ops_entry(handle);
    esp_err_t ret = ESP_OK;

    if (it == NULL) {
        return ESP_ERR_NOT_FOUND;
    }
    LIST_REMOVE(it, entries);
    free(it);
    return ret;
}
```

lucastcox, 2020-05-21:
Nevermind, just PRd this.

aep, 2020-05-21:
Nice, thanks!

shubhamkulkarni97, 2020-07-10:
@aep @lucastcox, `esp_ota_end` can be called to cancel OTA update and it doesn't have any side effects.

`esp_ota_end` has image validation check and it frees up allocated memory.

Also, as this is not a complex function, it does not have performance impact (if used to cancel OTA update).

We will add this to the documentation to avoid confusions.

lucastcox, 2020-07-10:
> @aep @lucastcox, `esp_ota_end` can be called to cancel OTA update and it doesn't have any side effects.
>
> `esp_ota_end` has image validation check and it frees up allocated memory.
>
> Also, as this is [...]

Thanks. It would also be helpful to add the aforementioned ota_abort function to allow a developer to cancel an ota without the performance impact of a potentially unnecessary image validation.

lucastcox, 2020-07-10:
Also, @shubhamkulkarni97 , surely the image validation has a performance impact? It has to loop through the entire image to computer the checksum. Being able to skip that with a separate abort call would be nice.

aep, 2020-07-10:
I guess it would be ok documenting that the function has no side effects.

I do however still think @lucastcox PR is much better, as it adds a clear API contract that can't accidentally be changed to have side effects later.

closed 2020-12-13 by espressif-bot