espressif/esp-idf #5330: uart_param_config() on 3.3.1 and 3.3.2

chewhs00, 2020-05-21:
On esp-idf 3.3, I use uart_param_config() to reconfigure an existing running uart's parameters such as baudrate, parity and data_bits, instead of calling individual uart_set_. 
When I migrate esp-idf to 3.3.2, the above failed to work, and no error returned when reinitialized uart's parameters again using uart_param_config().
From uart.c file, I noticed there is a change on later version where periph_module_reset(periph_module) is called before changing the parameters. If this is intended, what is the proper way to reinitialize multiple uart's parameters in this case?

Alvin1Zhang, 2020-05-22:
@chewhs00 Thanks for reporting, we will look into @koobest

koobest, 2020-05-22:
Hi, @chewhs00 
The `periph_module_reset ` is intent to bring the UART to a know state after software reset. mostly, `uart_param_config` is called on the first initialization. inside this function, `uart_set_` also is called one by one.

chewhs00, 2020-05-22:
@koobest , as my earlier comment stated, when I tried to reconfigure baud rate and parity by calling uart_param_config(), the communication failed, but the same code works on esp-idf 3.3.

chewhs00, 2020-05-22:
I think I accidentally closed this issue. I just want to know if uart_param_config() can be called again to reinitialize uart parameters for esp-idf 3.3.1 or later

koobest, 2020-05-27:
On esp-idf 4.1 or later we can call `uart_param_config` again to reinitialize uart parameters.

Alvin1Zhang, 2020-06-02:
@chewhs00 Thanks for reporting, would you please help share if any updates for the issue? Thanks.

chewhs00, 2020-06-03:
After migrating to esp-idf 3.3.2, instead of calling uart_param_config() to reinitialize uart's parameters, I called each uart_set_() to initialize the parameters. As @koobest commented, the fix will be applied to esp-idf4.1 or later

Alvin1Zhang, 2020-06-30:
Thanks for reporting, will close now, feel free to reopen or file a new ticket for new issue. Thanks.

closed 2020-06-30 by Alvin1Zhang