espressif/esp-idf #5332: Can't access SPIFFS, Core 0 paniced

stdenits, 2020-05-22:
## Environment

- Development Kit: [none]
- Kit version: [none]
- Module or chip used: [ESP32-PICO-D4]
- IDF version: v4
- Build System: [???]
- Compiler version: [???]
- Operating System: [Windows]
- (Windows only) environment type: [???].
- Using an IDE?: [Yes (VSCode + ESP-IDF Extension)]
- Power Supply: [external 3.3V]

## Problem Description
I am trying to combine soft_AP and Spiffs examples.

The SPIFFS image file is created using spiffsgen.py.
Changes added in sdkconfig (to enable custom partition) and CMakeLists.txt (to generate image file).
The * .csv file is got from the spiffsgen example.

Immediately after receiving information about SPIFFS, the ESP32 core panicked when checking the destination file, as i think.

### Expected Behavior
1. Check if file exists;
1.1) If FALSE, create a blank file;
1.2) If TRUE, read the string from the file;

### Actual Behavior

### Steps to reproduce

### Code to reproduce this issue

```cpp
ESP_LOGI(TAG2, "Initializing SPIFFS");
    
    esp_vfs_spiffs_conf_t conf = {
      .base_path = "/spiffs",
      .partition_label = NULL,
      .max_files = 5,
      .format_if_mount_failed = true
    };

    // Use settings defined above to initialize and mount SPIFFS filesystem.
    // Note: esp_vfs_spiffs_register is an all-in-one convenience function.
    ret = esp_vfs_spiffs_register(&conf);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount or format filesystem");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            ESP_LOGE(TAG2, "Failed to find SPIFFS partition");
        } else {
            ESP_LOGE(TAG2, "Failed to initialize SPIFFS (%s)", esp_err_to_name(ret));
        }
        return;
    }
    
    size_t total = 0, used = 0;
    ret = esp_spiffs_info(conf.partition_label, &total, &used);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG2, "Failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
    } else {
        ESP_LOGI(TAG2, "Partition size: total: %d, used: %d", total, used);
    }

    // Check if destination file exists                           **<- Crashed here**
    if (stat("/spiffs/wifi_param.txt", &st) != 0) {
        f = fopen("/spiffs/wifi_param.txt", "a+");
        if (f == NULL) {
            ESP_LOGE(TAG2, "Failed to create new file");
            return;
        }
        fclose(f);
    } else {
        f = fopen("/spiffs/wifi_param.txt", "r");
        if (f == NULL) {
            ESP_LOGE(TAG2, "Failed to open file for reading");
            return;
        }

        ***

        fclose(f);
    }

```

## Debug Logs

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6860
load:0x40078000,len:14076
ho 0 tail 12 room 4
load:0x40080400,len:4304
entry 0x400806e8
I (73) boot: Chip Revision: 1
I (74) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v4.0-dirty 2nd stage bootloader
I (42) boot: compile time 05:20:14
I (42) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (100) boot: End of partition table
I (104) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x189d8 (100824) map
I (157) esp_image: segment 1: paddr=0x00028a00 vaddr=0x3ffb0000 size=0x032c4 ( 12996) load
I (162) esp_image: segment 2: paddr=0x0002bccc vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/User/esp-idf/components/freertos/xtensa_vectors.S:1778

I (164) esp_image: segment 3: paddr=0x0002c0d4 vaddr=0x40080400 size=0x03f3c ( 16188) load
I (179) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x761dc (483804) map
0x400d0018: _stext at ??:?

I (354) esp_image: segment 5: paddr=0x000a61fc vaddr=0x4008433c size=0x117e4 ( 71652) load
0x4008433c: timer_count_reload at C:/Users/User/esp-idf/components/esp32/esp_timer_esp32.c:157
 (inlined by) timer_alarm_isr at C:/Users/User/esp-idf/components/esp32/esp_timer_esp32.c:248

I (384) esp_image: segment 6: paddr=0x000b79e8 vaddr=0x400c0000 size=0x00064 (   100) load
I (385) esp_image: segment 7: paddr=0x000b7a54 vaddr=0x50000000 size=0x00064 (   100) load
I (405) boot: Loaded app from partition at offset 0x10000
I (405) boot: Disabling RNG early entropy source...
I (405) cpu_start: Pro cpu up.
I (409) cpu_start: Application information:
I (414) cpu_start: Project name:     wifi_softAP
I (419) cpu_start: App version:      1
I (423) cpu_start: Compile time:     May 22 2020 05:20:00
I (429) cpu_start: ELF file SHA256:  e9910eb762c7d49f...
I (435) cpu_start: ESP-IDF:          v4.0-dirty
I (441) cpu_start: Starting app cpu, entry point is 0x40081344
0x40081344: call_start_cpu1 at C:/Users/User/esp-idf/components/esp32/cpu_start.c:272

I (426) cpu_start: App cpu up.
I (451) heap_init: Initializing. RAM available for dynamic allocation:
I (458) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (464) heap_init: At 3FFB9298 len 00026D68 (155 KiB): DRAM
I (470) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (477) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (483) heap_init: At 40095B20 len 0000A4E0 (41 KiB): IRAM
I (489) cpu_start: Pro cpu start user code
I (508) spi_flash: detected chip: gd
I (508) spi_flash: flash io: dio
I (508) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (575) ESP32_SPIFFS: Initializing SPIFFS
I (665) ESP32_SPIFFS: Partition size: total: 896321, used: 251
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400013f9  PS      : 0x00060f30  A0      : 0x800d68ae  A1      : 0x3ffbb060
A2      : 0x50000000  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffbb030
A10     : 0x50000000  A11     : 0x3f403878  A12     : 0x3ffb4164  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x3ffb41b0  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff

ELF file SHA256: e9910eb762c7d49f392164be53cd781847145dc3273099c9a2907f1a5b3e7c8a

Backtrace: 0x400013f6:0x3ffbb060 |<-CORRUPTED

Rebooting...
```

## Other items if possible

[partitions_example.zip](https://github.com/espressif/esp-idf/files/4665524/partitions_example.zip)
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/4665525/sdkconfig.zip)

stdenits, 2020-05-22:
Also, not entirely relevant, but may help...
Can someone explain to me how to execute the erase flash command in VSCode?

dobairoland, 2020-05-22:
Hello @stdenits.

I've added your code to the softapp example and it is working fine for me with IDF v4.0.

Here is the project for testing: [proj.zip](https://github.com/espressif/esp-idf/files/4668068/proj.zip)

> Can someone explain to me how to execute the erase flash command in VSCode?

I would run `idf.py erase_flash` from the terminal of VSCode.

stdenits, 2020-05-22:
New log after flash erasing and flshing firmware.
Erase flash procedure look like cmd-window that appears while 1 sec. Not sure how it must looks like.

`
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6864
ho 0 tail 12 room 4
load:0x40078000,len:14076
load:0x40080400,len:4304
entry 0x400806e8
I (77) boot: Chip Revision: 1
I (77) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot: ESP-IDF v4.0.1-dirty 2nd stage bootloader
I (42) boot: compile time 20:39:11
I (42) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 40MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (100) boot: End of partition table
I (105) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1a970 (108912) map
I (160) esp_image: segment 1: paddr=0x0002a998 vaddr=0x3ffb0000 size=0x038a4 ( 14500) load
I (166) esp_image: segment 2: paddr=0x0002e244 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/Users/Nox/esp-idf/components/freertos/xtensa_vectors.S:1778

I (167) esp_image: segment 3: paddr=0x0002e64c vaddr=0x40080400 size=0x019c4 (  6596) load
I (178) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x7e960 (518496) map
0x400d0018: _stext at ??:?

I (370) esp_image: segment 5: paddr=0x000ae980 vaddr=0x40081dc4 size=0x13e10 ( 81424) load
0x40081dc4: esp_stack_ptr_is_sane at C:/Users/Nox/esp-idf/components/soc/include/soc/soc_memory_layout.h:215
 (inlined by) esp_backtrace_get_next_frame at C:/Users/Nox/esp-idf/components/xtensa/debug_helpers.c:32

I (404) esp_image: segment 6: paddr=0x000c2798 vaddr=0x400c0000 size=0x00064 (   100) load
I (404) esp_image: segment 7: paddr=0x000c2804 vaddr=0x50000000 size=0x00064 (   100) load
I (424) boot: Loaded app from partition at offset 0x10000
I (424) boot: Disabling RNG early entropy source...
I (425) cpu_start: Pro cpu up.
I (428) cpu_start: Application information:
I (433) cpu_start: Project name:     wifi_softAP
I (438) cpu_start: App version:      1
I (443) cpu_start: Compile time:     May 22 2020 20:38:50
I (449) cpu_start: ELF file SHA256:  de74d424066db2ea...
I (455) cpu_start: ESP-IDF:          v4.0.1-dirty
I (460) cpu_start: Starting app cpu, entry point is 0x400813a4
0x400813a4: call_start_cpu1 at C:/Users/Nox/esp-idf/components/esp32/cpu_start.c:271

I (446) cpu_start: App cpu up.
I (471) heap_init: Initializing. RAM available for dynamic allocation:
I (478) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (484) heap_init: At 3FFB9AC0 len 00026540 (153 KiB): DRAM
I (490) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (496) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (503) heap_init: At 40095BD4 len 0000A42C (41 KiB): IRAM
I (509) cpu_start: Pro cpu start user code
I (527) spi_flash: detected chip: gd
I (528) spi_flash: flash io: dio
I (528) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (595) ESP32_SPIFFS: Initializing SPIFFS
I (685) ESP32_SPIFFS: Partition size: total: 896321, used: 251
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400013f9  PS      : 0x00060f30  A0      : 0x800d6a06  A1      : 0x3ffbb880
A2      : 0x50000000  A3      : 0x00000000  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffbb850
A10     : 0x50000000  A11     : 0x3f403874  A12     : 0x3ffb474c  A13     : 0x00000001
A14     : 0x00000000  A15     : 0x3ffb4798  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xffffffff

ELF file SHA256: de74d424066db2ea

Backtrace: 0x400013f6:0x3ffbb880 0x400d6a03:0x3ffbb890 0x400d262a:0x3ffbb8e0 0x40092bf1:0x3ffbb900
0x400d6a03: app_main at C:\Users\Nox\esp\irsta\build/../main/main.c:399

0x400d262a: main_task at C:/Users/Nox/esp-idf/components/esp32/cpu_start.c:553

0x40092bf1: vPortTaskWrapper at C:/Users/Nox/esp-idf/components/freertos/port.c:143

Rebooting...
`

dobairoland, 2020-05-22:
@stdenits Do you get this error with my project?

Please check the output of `git status` why is your IDF "dirty", ie. "v4.0.1-dirty".

stdenits, 2020-05-22:
@dobairoland dobairoland
>Do you get this error with my project?
I didn't test it yet.

>Please check the output of git status
Here is what I got from the VSCode terminal:
> fatal: not a git repository (or any of the parent directories): .git

>why is your IDF "dirty", ie. "v4.0.1-dirty"
I don't know.

stdenits, 2020-05-22:
@dobairoland 
Your project works fine. I don't know why my is not.

closed 2020-05-23 by stdenits

dobairoland, 2020-05-23:
You are welcome. I'm happy that the issue is resolved.