espressif/esp-idf #5334: Can't change attenuation for I2S ADC.

physiii, 2020-05-22:
I need to attenuate the signal read by i2s_read.

When shorting the DAC output to the ADC input all I see is noise. I ran a test by gradually moving up the voltage and notice the average voltage repeats every few hundred mV.

I hoped it was as easy as running: `adc1_config_channel_atten(I2S_ADC_CHANNEL,ADC_ATTEN_DB_11)`

But that did not work nor did setting other attenuations .

I do not see anything in the documentation however searching through esp idf files, I found [this](https://github.com/espressif/esp-idf/blob/master/components/driver/test/test_i2s.c#L226):
` // enable adc sampling, ADC_WIDTH_BIT_12, ADC_ATTEN_DB_11 hard-coded in adc_i2s_mode_init`

How do I attenuate the i2s adc channel so I can spread the 12 bits across 0 to 3v3 range?

MonoAnji, 2020-05-23:
Please make that configurable. It's a pain, that there's not even a simple #ifndef DIG_ADC_ATTEN_DEFUALT 
[sic!]

marina936, 2020-06-11:
Hi,

I have a similar problem, I want to read ADC with 0dB attenuation but when I read from I2S , values are attenuated 11dB.
I tried to change adc attenuation (adc1_config_channel_atten) after calling i2s_set_adc_mode, but that doesn't work for me. Maybe you can try.

brainsucker-na, 2021-03-26:
i2s_set_adc_mode actually uses adc dma patterns, and you can change these by copying implementation of adc_ll_digi_set_pattern_table function into your code or just by adjusting ADC register manually, like
extern "C" {
```
    #include "soc/syscon_reg.h"
    #include "soc/syscon_struct.h"
}
// channel 6 at 12bits with 0db attenuation 
WRITE_PERI_REG(SYSCON_SARADC_SAR1_PATT_TAB1_REG, 0x6C000000);
// you can also use more than one pattern to scan sequentially by using more bytes in register
SYSCON.saradc_ctrl.sar1_patt_len = PatternsCount-1;
```

Also in current esp idf used by arduino framework i2s_adc_enable is restoring adc registers with _i2s_adc_mode_recover call, so it is better to adjust pattern registers after i2s_adc_enable, not before.