espressif/esp-idf #5335: vfs select() for queue

aep, 2020-05-22:
it would be very useful if in addition to sockets, there where other ways to wake up a call to select()

for example the queue would fit well.

Alvin1Zhang, 2020-05-25:
@aep Thanks for raising the feature request.

dobairoland, 2020-05-25:
Hi @aep. Your way of formulation of the request makes me wonder whether really you are interested in the support of adding other ways in general instead of just queues. 

Note that if you want to develop a support for your own driver then the support is already in [Virtual filesystem layer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html). The UART driver is also supported.

We will consider the implementation of the support for queue. I'm not sure when that will be. However, you are encouraged to create a PR for this feature if you like.

aep, 2020-05-27:
i'm not sure how a good implementation would look like, since the vfs layer requires file names.
currently i'm just using a singleton and wrapping it in esp_vfs_t. works fine, but not very reusable.

omnitex, 2026-11-02:
Hi @aep,

If I understand correctly, the functionality you requested should be now supported through `eventfd`, added in espressif/esp-idf@f6fd2eb1648ece6139c9995fcb46d25843b57cdd and available since ESP-IDF v4.4. 

After `esp_vfs_eventfd_register()`, `eventfd()` returns a file descriptor that can be used in `select()` together with sockets, UART, and other VFS file descriptors. Any task can wake the `select()` by writing to it, and so can an ISR if the fd is created with `EFD_SUPPORT_ISR`.

To combine this with a FreeRTOS queue: send the item to the queue, then write to the eventfd. The `select()` loop wakes up on the eventfd and reads from the queue.

- Documentation: [VFS - eventfd()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/vfs.html#eventfd)
- Example: [examples/system/eventfd](https://github.com/espressif/esp-idf/tree/master/examples/system/eventfd)

Closing this as implemented. Feel free to reopen if this doesn't cover your use case.

closed 2026-11-02 by omnitex