espressif/esp-idf #5339: esp_wifi_scan_get_ap_record documentation clarity

bbulkow, 2020-05-23:
**Is your feature request related to a problem? Please describe.**

The documentation regarding this function, and the related scan, is a bit peculiar, because it describes that there are dynamic memory allocations associated with the scan, and 'get_ap_record' should be called, but doesn't go into a few key details.

**Describe the solution you'd like**

I would like documented behavior for the following situations:
- If 'esp_wifi_scan_get_ap_record' is called multiple times after a scan, will it work?
- If 'esp_wifi_scan_get_ap_record' is not called after a scan, perhaps on a repeated basis, will there be a memory leak?
- If 'esp_wifi_scan_get_ap_record' thread safe with scan, or is it critical to make sure that get_ap_record is not called while a scan is in progress, and vice versa?

The documentation states that one should refer to a section about scan function behavior, but I can't find any such section in the documentation. The documentation also suggests referring to the example, but the example is a synchronous scan with a single call to get the records, as opposed to the more normal use of continual background scans which are asynchronous, and then using the event handler.

**Describe alternatives you've considered**

I have considered looking at the source code, but it is not distributed, as esp-idf is not open source.

I have considered measuring stack heaps and trying all the above things, but then I might count on undocumented behavior.

I have considered writing the safest possible code, with interlocks and semaphores, but it is complicated.

**Additional context**

Add any other context or screenshots about the feature request here.

ESP-Marius, 2020-05-27:
Hi, thanks for raising this issue. 

Did you look at this section: [scan-done-event-handling-phase](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?highlight=esp_wifi_scan_get_ap_record#scan-done-event-handling-phase)? It seems to at least address some of your questions. We should definitely give a link though, not just a vague reference....

As for the details, you'll have to wait for a reply from someone on our WiFi team.

YouDONG-ESP, 2020-06-01:
Hi bbulkow,
Answer to your questions:
1. You need to call esp_wifi_scan_get_ap_record once after WIFI_EVENT_SCAN_DONE to get complete scan record and free memory in wifi driver. Call it mutiple times wont return error, considering record has already been free during first call, it won't return any information neither. 
2. esp_wifi_scan_start will cancel previous scan and clear scan record, so it won't cause memory leak if you call it multiple times. 
3. You can call esp_wifi_scan_get_ap_record while scan is in progress, but it will only return information that has been found.

closed 2020-09-07 by Alvin1Zhang