espressif/esp-idf #5340: Better support for timezones

DiegoActionItem, 2020-05-24:
I want to display on local logs the current time, adjested to timezone. (boring details: I get to google and fetch the location by geoip, and then call again google to ask the timezone. I get in return an offset in minutes. 

I am using the `localtime()` function, and to adjust the time, I use this code:
```
    auto positive = offset <= 0 ? '+' : '-' ;
    auto minutes = abs(offset % (60*60));
    auto hours = abs(offset / (60*60));

    char tz[100];
    snprintf(tz, 100, "GMT%c%02d:%02d", positive, hours, minutes);
    setenv("TZ", tz, 1);
    tzset();
    V_PRINT("TZ=%s, offset=%d", tz, offset);
```

(basically seetting HH:HH into "TZ" env variable).

I found out that I need to use a negative value, while on Linux I need to use a positive value (this seems wrong to me).

How should I handle timezones in ESP?

holubcsongor, 2020-05-24:
Every timezone has a standardized name, you have to use this name instead of GMT.
You should search for its name in a search engine
For example: EST5EDT,M3.2.0/2,M11.1.0 for Eastern part of U.S or CST-8 to China.

KonstantinKondrashov, 2020-05-25:
Hi @DiegoActionItem!
Could you observe [a sntp example](https://github.com/espressif/esp-idf/blob/7d75213674b4572f90c68162ad6fe9b16dae65ad/examples/protocols/sntp/main/sntp_example_main.c#L89) and [doc about System Time](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html#get-current-time)?
Thanks.

raldone01, 2020-05-25:
Take a look at this: https://github.com/moment/moment-timezone/issues/314

DiegoActionItem, 2020-05-26:
I don't want to use names, as the internal DB for timezones will not get updated. If we do use names, at some time, some user will get an hour off. I want to use "minutes offset", which will I read from a server.

Yes, some timezones are not in full hours, yes, some countries have different time zones, yes, you cannot know the local time of a location 3 years in advance, since countries change the timezone definitions weeks in advance. The full timezone database is 300kb compressed and needs to be upgraded once a month... yes, properly handling timezone is hard. So - move this logic to the server.

Here is the only official DB: 
https://www.iana.org/time-zones

@raldone01  so - basically you said "no thank you, we don't handle this?"

KonstantinKondrashov, 2020-05-26:
Hi @DiegoActionItem!
Please read about that here https://unix.stackexchange.com/questions/104088/why-does-tz-utc-8-produce-dates-that-are-utc8. 

> Why does TZ=UTC-8 produce dates that are UTC+8?
The current time in Los Angeles is 18:05. But when I run TZ=UTC-8 date --iso=ns, I get: 2013-12-07T10:05:37,788173835+0800
The date utility tells me that the time is 10:05, and even says that it's reporting it as UTC+8. Why?
The reason is that TZ=UTC-8 is interpreted as a POSIX time zone. In the POSIX timezone format, the 3 letters are the timezone abbreviation (which is arbitrary) and the number is the number of hours the timezone is behind UTC. So UTC-8 means a timezone abbreviated "UTC" that is −8 hours behind the real UTC, or UTC + 8 hours.

Does it help with your issue?

KonstantinKondrashov, 2020-06-08:
Hi @DiegoActionItem!
Could you confirm it resolved the issue? If so, please close it. Thanks.

closed 2020-06-08 by DiegoActionItem

KonstantinKondrashov, 2020-06-08:
@DiegoActionItem.
The newlib implementation is the same as GNU lib for that case. This works as described in - https://www.gnu.org/software/libc/manual/html_node/TZ-Variable.html. You should use it in a reverse way, please observe an example below.

> I found out that I need to use a negative value,

- Exactly, you should use a negative value [setenv("TZ", "UTC-x")] for UTC+x if it is  east of the Prime Meridian
- and should use a positive value [setenv("TZ", "UTC+x")] for UTC-x if it is west of the Prime Meridian.

Example:
Area | Timezone | setenv()  |
------------ | -------------|----------------|
| Asia/ Shanghai | Universal Time Coordinated GMT/UTC [UTC+8] | setenv("TZ", "CST-8") |
| USA/ New York | Eastern Time Zone [UTC-5] | setenv("TZ", "EST+5")  |

[The sntp example](https://github.com/espressif/esp-idf/blob/7d75213674b4572f90c68162ad6fe9b16dae65ad/examples/protocols/sntp/main/sntp_example_main.c#L89) demonstrates that. It works as expected it is no an issue.
Thanks.