espressif/esp-idf #5344: Build failure in current master tree: ImportError: bad magic number in 'pyparsing'

AxelLin, 2020-05-25:
## Environment

- Module or chip used: ESP32-WROOM-32
- IDF version: v4.2-dev-1660-g7d75213674b4
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: NO
- Power Supply: USB

## Problem Description

Just did ./install.sh after pull latest code, then I'm not able to build any project now.
Any chance to fix it ASAP?

[95/100] Linking C static library esp-idf/bootloader_support/libbootloader_support.a
[96/100] Linking C static library esp-idf/efuse/libefuse.a 
[97/100] Linking C static library esp-idf/spi_flash/libspi_flash.a
[98/100] Linking C static library esp-idf/main/libmain.a
[99/100] Linking C executable bootloader.elf
[100/100] Generating binary image from built executable
esptool.py v3.0-dev
Generated /home/axel/esp/esp-idf/examples/wifi/getting_started/softAP/build/bootloader/bootloader.bin
[941/944] Generating ld/esp32.project.ld
FAILED: esp-idf/esp32/ld/esp32.project.ld
cd /home/axel/esp/esp-idf/examples/wifi/getting_started/softAP/build/esp-idf/esp32 && /home/axel/.espressif/python_env/idf4.2_py3.8_env/bin/python /home/axel/esp/esp-idf/tools/ldgen/ldgen.py --config /home/axel/esp/esp-idf/examples/wifi/getting_started/softAP/sdkconfig --fragments /home/axel/esp/esp-idf/components/xtensa/linker.lf    /home/axel/esp/esp-idf/components/spi_flash/linker.lf   /home/axel/esp/esp-idf/components/esp_system/linker.lf  /home/axel/esp/esp-idf/components/soc/linker.lf /home/axel/esp/esp-idf/components/esp_event/linker.lf   /home/axel/esp/esp-idf/components/esp_wifi/linker.lf    /home/axel/esp/esp-idf/components/lwip/linker.lf        /home/axel/esp/esp-idf/components/log/linker.lf /home/axel/esp/esp-idf/components/heap/linker.lf        /home/axel/esp/esp-idf/components/esp_ringbuf/linker.lf /home/axel/esp/esp-idf/components/espcoredump/linker.lf /home/axel/esp/esp-idf/components/esp32/linker.lf       /home/axel/esp/esp-idf/components/esp32/ld/esp32_fragments.lf   /home/axel/esp/esp-idf/components/esp_common/linker.lf  /home/axel/esp/esp-idf/components/freertos/linker.lf    /home/axel/esp/esp-idf/components/newlib/newlib.lf      /home/axel/esp/esp-idf/components/app_trace/linker.lf   /home/axel/esp/esp-idf/components/esp_gdbstub/linker.lf --input /home/axel/esp/esp-idf/components/esp32/ld/esp32.project.ld.in --output /home/axel/esp/esp-idf/examples/wifi/getting_started/softAP/build/esp-idf/esp32/ld/esp32.project.ld --kconfig /home/axel/esp/esp-idf/Kconfig --env-file /home/axel/esp/esp-idf/examples/wifi/getting_started/softAP/build/config.env --libraries-file /home/axel/esp/esp-idf/examples/wifi/getting_started/softAP/build/ldgen_libraries --objdump /home/axel/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-objdump
Traceback (most recent call last):
  File "/home/axel/esp/esp-idf/tools/ldgen/ldgen.py", line 26, in <module>
    from fragments import FragmentFile
  File "/home/axel/esp/esp-idf/tools/ldgen/fragments.py", line 19, in <module>
    from sdkconfig import SDKConfig
  File "/home/axel/esp/esp-idf/tools/ldgen/sdkconfig.py", line 18, in <module>
    from pyparsing import Word, alphanums, printables, Combine, Literal, hexnums, quotedString, Optional, nums, removeQuotes, oneOf, Group, infixNotation, opAssoc
ImportError: bad magic number in 'pyparsing': b'\x03\xf3\r\n'
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

dobairoland, 2020-05-25:
@AxelLin Please try to remove `*.pyc` files from the IDF directory.

AxelLin, 2020-05-25:
> @AxelLin Please try to remove `*.pyc` files from the IDF directory.

Hi @dobairoland 
Thanks for prompt response, it works after removing *.pyc files.

closed 2020-05-25 by AxelLin

dobairoland, 2020-05-25:
Great! We will provide a permanent solution for this.