espressif/esp-idf #5346: ESP-IDF Install Error, at step 3 of ESP-IDF Programming Guide (Linux)

starla-eng, 2020-05-25:
----------------------------- Delete below -----------------------------

Help required to install ESP-IDF on Raspberry Pi4
INSTRUCTIONS
============

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      ESP32-DevKitC
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0.1
- Build System: 
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

While at step of ESP-IDF Programming Guide; while running ./install.sh, ERROR  is generated.

### Expected Behavior
install correctly

### Actual Behavior

./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Installing Python environment and packages
Installing Python packages from /home/pi/esp/esp-idf/requirements.txt
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: setuptools in /home/pi/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages (from -r /home/pi/esp/esp-idf/requirements.txt (line 4)) (44.1.0)
Collecting click>=5.0
  Using cached click-7.1.2-py2.py3-none-any.whl (82 kB)
Collecting pyserial>=3.0
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Collecting future>=0.15.2
  Using cached future-0.18.2.tar.gz (829 kB)
Collecting cryptography>=2.1.4
  Using cached cryptography-2.9.2.tar.gz (517 kB)
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.espressif/python_env/idf4.0_py2.7_env/bin/python /home/pi/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-NP7Neb/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; platform_python_implementation != '"'"'PyPy'"'"''
       cwd: None
  Complete output (142 lines):
  DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=40.6.0
    Using cached setuptools-44.1.0-py2.py3-none-any.whl (583 kB)
  Collecting wheel
    Using cached wheel-0.34.2-py2.py3-none-any.whl (26 kB)
  Collecting cffi!=1.11.3,>=1.8
    Using cached cffi-1.14.0.tar.gz (463 kB)
  Collecting pycparser
    Using cached pycparser-2.20-py2.py3-none-any.whl (112 kB)
  Building wheels for collected packages: cffi
    Building wheel for cffi (setup.py): started
    Building wheel for cffi (setup.py): finished with status 'error'
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.espressif/python_env/idf4.0_py2.7_env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-9yJ8Ky/cffi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-9yJ8Ky/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-IWxmtO
         cwd: /tmp/pip-install-9yJ8Ky/cffi/
    Complete output (56 lines):of ESP-IDF Programming Guide
    Package libffi was not found in the pkg-config search path.
    Perhaps you should add the directory containing `libffi.pc'
    to the PKG_CONFIG_PATH environment variable
    No package 'libffi' found
[... 108 lines cut ...]
      copying cffi/_cffi_errors.h -> build/lib.linux-armv7l-2.7/cffi
      running build_ext
      building '_cffi_backend' extension
      creating build/temp.linux-armv7l-2.7
      creating build/temp.linux-armv7l-2.7/c
      arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fdebug-prefix-map=/build/python2.7-InigCj/python2.7-2.7.16=. -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python2.7 -c c/_cffi_backend.c -o build/temp.linux-armv7l-2.7/c/_cffi_backend.o
      c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
       #include <ffi.h>
                ^~~~~~~
      compilation terminated.
      error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
      ----------------------------------------
  ERROR: Command errored out with exit status 1: /home/pi/.espressif/python_env/idf4.0_py2.7_env/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-9yJ8Ky/cffi/setup.py'"'"'; __file__='"'"'/tmp/pip-install-9yJ8Ky/cffi/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-DIRN0w/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-NP7Neb/overlay --compile --install-headers /home/pi/.espressif/python_env/idf4.0_py2.7_env/include/site/python2.7/cffi Check the logs for full command output.
  ----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/.espressif/python_env/idf4.0_py2.7_env/bin/python /home/pi/.espressif/python_env/idf4.0_py2.7_env/lib/python2.7/site-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-NP7Neb/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; platform_python_implementation != '"'"'PyPy'"'"'' Check the logs for full command output.
Traceback (most recent call last):
  File "/home/pi/esp/esp-idf/tools/idf_tools.py", line 1387, in <module>
    main(sys.argv[1:])
  File "/home/pi/esp/esp-idf/tools/idf_tools.py", line 1383, in main
    action_func(args)
  File "/home/pi/esp/esp-idf/tools/idf_tools.py", line 1196, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/pi/.espressif/python_env/idf4.0_py2.7_env/bin/python', '-m', 'pip', 'install', '--no-warn-script-location', '-r', '/home/pi/esp/esp-idf/requirements.txt']' returned non-zero exit status 1

igrr, 2020-05-25:
@starla-eng It looks like cffi package doesn't have wheels for ARM architecture (https://pypi.org/project/cffi/1.14.0/#files), so you will likely need to install `libffi-dev` package using apt-get. Could you please try that?

igrr, 2020-05-25:
Alternatively, you may switch your `python` to Python 3. This will help because there are Python3 wheels for cffi at https://www.piwheels.org/project/cffi/, and you have piwheels.org in your `--extra-index-url`s. This is probably not a bad idea anyway, given that Python 2.7 is no longer supported.

You should be able to change the default Python interpreter on your Raspberry PI using `update-alternatives --config python` command, provided that Python 3 is already installed.

starla-eng, 2020-05-25:
> @starla-eng It looks like cffi package doesn't have wheels for ARM architecture (https://pypi.org/project/cffi/1.14.0/#files), so you will likely need to install `libffi-dev` package using apt-get. Could you please try that?

Ivan,
I tried that and have now completed the installation successful.

Regards,
starla-eng

closed 2020-05-25 by starla-eng