espressif/esp-idf #5347: nvs: doesn't work with external flash chips, overwrites main flash instead

jimparis, 2020-05-25:
Using esp-idf v4.0-386-gb0f053d82dc4

If I create a partition with:
```
esp_flash_spi_device_config_t config = { ... };
esp_flash_t *chip;
spi_bus_add_flash_device(&chip, &config);
esp_flash_init(chip);
esp_partition_register_external(chip, 0x100000, 0x110000,
    "ext_nvs", ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_DATA_NVS);
```
then try to use it with NVS:
```
nvs_flash_init_partition("ext_nvs");
nvs_open_from_partition("ext_nvs", "config", NVS_READWRITE, &handle);
```
it overwrites my main flash instead of the external one :frowning_face: 

This is because `nvs_flash_init_partition` pulls out the partition's address and size, but ignores the chip, and uses the `spi_flash` API on the main chip instead.

0xjakob, 2020-05-26:
Hi @jimparis. I'm sorry it doesn't work yet! We're close to adding support for external flash chips in nvs. But right now, as you noticed, it doesn't work due to the limitation of the `spi_flash` API.
The support of the NVS refactoring which can handle external flash chips is planned for IDV v4.3 so far. We plan to have in in the master branch in few weeks, however.

sicanins, 2020-07-24:
Hi @0xjakob,
 is this meanwhile in the master branch? Desperately needing this feature...
Thanks!

0xjakob, 2020-07-27:
@sicanins Unfortunately not yet. The introduction got delayed a bit. But we're definitely on this.

0xjakob, 2021-03-01:
@sicanins @jimparis We merged a feature to support custom `esp_partition` in NVS setup a few months ago. I'll close this issue. If you still have problems using external partitions, please re-open the issue and let us know what does not work.

closed 2021-03-01 by 0xjakob