espressif/esp-idf #5348: esp_err_to_name incomplete, because it's not built with all used components

jimparis, 2020-05-26:
This is v4.0-386-gb0f053d82dc4, using cmake.

The file `components/esp_common/src/esp_err_to_name.c` has compile-time checks for header files like `nvs.h`, but since `esp_common` does not depend on the `nvs_flash` component, it doesn't get built with `components/nvs_flash/include` in the include path, so the NVS errors aren't included.

This means that if you have a main app with something like this in `CMakeLists.txt`:
```cmake
idf_component_register(SRCS ${srcs}
  PRIV_REQUIRES nvs_flash
  )
```
and your main code calls
```c
esp_err_to_name(ESP_ERR_NVS_NOT_INITIALIZED)
```
you'll just get `"ERROR"` back.

I can work around it with something like this in `project_include.cmake`:

```cmake
# Make sure esp_err_to_name sees all the components I've used, and their dependencies
include_directories("esp-idf/components/nvs_flash/include")
include_directories("esp-idf/components/app_update/include")
include_directories("esp-idf/components/spi_flash/include")
include_directories("esp-idf/components/bootloader_support/include")
```

Alvin1Zhang, 2020-06-03:
@jimparis Thanks for reporting, we will look into. @shivani-tipnis

AxelLin, 2020-08-10:
Test with v4.3-dev-771-gc77c4ccf6c43, I hit similar issue.

Below code shows ERROR rather than "ESP_ERR_ESP_TLS_FAILED_CONNECT_TO_HOST".

ESP_LOGW(TAG, "MQTT_EVENT_ERROR TEST: %d %s", ESP_ERR_ESP_TLS_FAILED_CONNECT_TO_HOST, esp_err_to_name(ESP_ERR_ESP_TLS_FAILED_CONNECT_TO_HOST));
W (13703) MQTT_TEST: MQTT_EVENT_ERROR TEST: 32772 ERROR

shivani-tipnis, 2020-08-10:
@AxelLin Thank you for reporting, will look into it.

Alvin1Zhang, 2020-09-04:
Thanks for reporting and sorry for slow turn around. We have a fix internally under reviewing, will update once the fix is available on GitHub. Thanks.

AxelLin, 2020-11-05:
I tested again with v4.3-dev-1720-g494a124d961c, now it shows correct  "ESP_ERR_ESP_TLS_FAILED_CONNECT_TO_HOST" output.
W (12576) MQTT_CLI: MQTT_EVENT_ERROR TEST: 32772 ESP_ERR_ESP_TLS_FAILED_CONNECT_TO_HOST

So I'm wondering if this issue is fixed or not.

Alvin1Zhang, 2021-12-10:
Thanks for reporting, feel free to reopen.

closed 2021-12-10 by Alvin1Zhang

jimparis, 2021-12-13:
It looks like this was originally reported as https://github.com/espressif/esp-idf/issues/3710 and fixed by https://github.com/espressif/esp-idf/commit/38ccf5f28badeecb3a99a20ad745341a222fa1d0.  That fix predates this bug report, but was in the 4.1 branch, not 4.0.  The later example from @AxelLin was fixed by 8963ebf2e22e1295285692716563e02b8f989d00.