espressif/esp-idf #5349: BIGNUM - Memory allocation failed when using mbedtls_pk_sign

galah92, 2020-05-26:
## Environment

- Development Kit: ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.0.1
- Build System: idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.2.0
- Operating System: Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Visual Studio Code
- Power Supply: USB

## Problem Description

I'm developing an application involving extensive use of WiFi, HTTP requests, MQTT and using `mbedtls` for ssl operation. Occasionally, after a significant amount of time, I need to sign keys with `mbedtls_pk_sign`, and I get an error number 17168 with the following:
> RSA - The private key operation failed : BIGNUM - Memory allocation failed.

### Expected Behavior

My belief is that the WiFi/tcpip stack is taking over the heap and leave no room for `mbedtls`. I'd like to preallocate static buffers for `mbedtls` usage ([reference](https://tls.mbed.org/kb/how-to/using-static-memory-instead-of-the-heap)). How can I achieve this with the ESP-IDF version of `mbedtls`? What more can be done?

Thank you

mahavirj, 2020-05-27:
@galah92 You might want to check per task heap utilization in this case, example at https://github.com/espressif/esp-idf/tree/master/examples/system/heap_task_tracking should be helpful for this.

mahavirj, 2020-06-26:
@galah92 Please see if some of the tips from [post](https://medium.com/the-esp-journal/esp32-memory-analysis-case-study-eacc75fe5431) are helpful for saving on internal memory (especially WiFi buffer configuration). I would go ahead and close the issue but please feel free to re-open if any additional help is required.

closed 2020-06-26 by mahavirj