espressif/esp-idf #5350: rmt: 5 seconds tx hang since v4.1-beta1

sicanins, 2020-05-26:
#5148  Environment

- Development Kit:     none
- Module or chip used:  ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-1547-g5b5a8c276
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it) :esp-2020r1
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: Eclipse
- Power Supply:         3.3V

## Problem Description

The rmt driver shows a weird behavior in the master branch. I was using 4.1-beta1 before and had to update to master because of the secure boot V2.

The rmt is used to implement a bidirectional manchester encoded communication. 
A transmission sequence of several packets in both directions takes about 2 minutes.

### Expected Behavior

All packets get exchanged.

### Actual Behavior

At least once during these 2 minutes, the transmitter blocks for exactly 5 seconds.

It hangs in rmt_write_items() line 1026:

  ```
   // wait tx end
            xSemaphoreTake(p_rmt->tx_sem, portMAX_DELAY);
```

Due to lack of time I reverted the files: rmt.c rmt-hal.c and rmt_ll.h back to the v4.1-beta1.
That fixed the issue for me.

I am willing to test a proposed fix here, but dont have time to create a project that reproduces the error right now.

My project uses 3 rmt channels. 

`
static void _tx_init()
{
  rmt_config_t rmt_tx;
  rmt_tx.channel = RMT_TX_CHANNEL;
  rmt_tx.gpio_num = RMT_TX_GPIO_NUM;
  rmt_tx.mem_block_num = 1;
  rmt_tx.clk_div = 80;
  rmt_tx.tx_config.loop_en = false;
  rmt_tx.tx_config.carrier_duty_percent = 50;
  rmt_tx.tx_config.carrier_freq_hz = 8000000;
  rmt_tx.tx_config.carrier_level = 1;
  rmt_tx.tx_config.carrier_en = 1;
  rmt_tx.tx_config.idle_level = 0;
  rmt_tx.tx_config.idle_output_en = true;
  rmt_tx.rmt_mode = 0;
  rmt_config( &rmt_tx );
  rmt_driver_install( rmt_tx.channel, 0, 0 );
}
`

`
static void _rx_init()
{
  rmt_config_t rmt_rx;
  rmt_rx.channel = RMT_RX_CHANNEL;
  rmt_rx.gpio_num = RMT_RX_GPIO_NUM;
  rmt_rx.clk_div = RMT_CLK_DIV;
  rmt_rx.mem_block_num = 4;
  rmt_rx.rmt_mode = RMT_MODE_RX;
  rmt_rx.rx_config.filter_en = true;
  rmt_rx.rx_config.filter_ticks_thresh = 20;
  rmt_rx.rx_config.idle_threshold = REMOTE_OLED_RX_TIMEOUT;
  rmt_config( &rmt_rx );
  rmt_driver_install( rmt_rx.channel, 2000, 0 );
}
`

And unrelated to the manchester comm an IR receiver, pretty much from the nec example:
`
static void nec_rx_init()
{
  rmt_config_t rmt_rx;
  rmt_rx.channel = RMT_RX_CHANNEL;
  rmt_rx.gpio_num = RMT_RX_GPIO_NUM;
  rmt_rx.clk_div = RMT_CLK_DIV;
  rmt_rx.mem_block_num = 1;
  rmt_rx.rmt_mode = RMT_MODE_RX;
  rmt_rx.rx_config.filter_en = true;
  rmt_rx.rx_config.filter_ticks_thresh = 100;
  rmt_rx.rx_config.idle_threshold = rmt_item32_tIMEOUT_US / 10 * (RMT_TICK_10_US);
  rmt_config( &rmt_rx );
  rmt_driver_install( rmt_rx.channel, 1000, 0 );
}
`

Thanks, 
Simon

Alvin1Zhang, 2020-05-27:
@sicanins Thanks for the report, we will look into.

koobest, 2020-05-27:
@sicanins 
Please check you configuration,e.g.:
```
rmt_config_t rmt_tx = RMT_DEFAULT_CONFIG_TX( RMT_TX_GPIO_NUM, RMT_TX_CHANNEL);
rmt_tx.tx_config.carrier_duty_percent = 50;
rmt_tx.tx_config.carrier_freq_hz = 8000000;
rmt_tx.tx_config.carrier_en = true;
```
 the same for RX channels.

thanks !!

sicanins, 2020-05-27:
@koobest 
Thanks, that fixed it. I think the .flags was not initialized.

Changing the tx init to:

static void _tx_init()
{
  rmt_config_t rmt_tx = RMT_DEFAULT_CONFIG_TX(RMT_TX_GPIO_NUM, RMT_TX_CHANNEL);
  rmt_tx.mem_block_num = 1;
  rmt_tx.clk_div = 80;
  rmt_tx.tx_config.loop_en = false;
  rmt_tx.tx_config.carrier_duty_percent = 50;
  rmt_tx.tx_config.carrier_freq_hz = 8000000;
  rmt_tx.tx_config.carrier_level = RMT_CARRIER_LEVEL_HIGH;
  rmt_tx.tx_config.carrier_duty_percent = 50;
  rmt_tx.tx_config.carrier_en = 1;
  rmt_tx.tx_config.idle_level = RMT_IDLE_LEVEL_LOW;
  rmt_tx.tx_config.idle_output_en = true;

  rmt_tx.rmt_mode = 0;
  rmt_config( &rmt_tx );
  rmt_driver_install( rmt_tx.channel, 0, 0 );
}

already solved it.(rx are also adjusted, but did not make the difference).

Thanks, appreciate your fast response!

closed 2020-05-27 by sicanins