espressif/esp-idf #5351: Random WiFi TCP/IP disconnect

lexxvir, 2020-05-26:
## Environment

- Custom Hardware
- Module or chip used: ESP32-WROVER-B 
- IDF version 3.3.2
- Build System: Make
- Compiler version 1.22.0-80-g6c4433a
- Operating System: Linux

## Problem Description

We use ESP32 in our custom device, it is EMV contactless (ISO14443) cards reader. The device receives TCP/IP connection from the PC and holds it. Then PC continuously polls the device by the commands under this TCP/IP connection. TCP/IP connection is single, there are no other connections. PC sends each command at approximately one per second.

Also the device enables TCP/IP KeepAlive feature on the established connection.
lwIP TCP/IP KeepAlive parameters:
```c
    { SOL_SOCKET,  SO_KEEPALIVE,  1 },    // enable TCP keepalive feature
    { IPPROTO_TCP, TCP_KEEPALIVE, 1000 }, // interval between keepalive probe packets in milliseconds
    { IPPROTO_TCP, TCP_KEEPINTVL, 5 },    // how much time (in seconds) keepalive packets may be unanswered before counter incremented
    { IPPROTO_TCP, TCP_KEEPCNT,   5 },    // max counter value, if counter more than that, server drops connection
``` 

Out customer notices that some of devices have reliable TCP/IP connection, without disconnections.
But other device randomly closes connection at their end. After that, PC tries to reconnect to the device. New connection may be established immediately, but may takes significant time (up to few minutes). All devices placed on the same distance from the PC.

Our customer notices that while TCP/IP connection have dropped, WiFi connection still established according to the Access Point status.

After notice the issue, we have made following settings to increase connection reliability, but without success:
- set `WIFI_PROTOCOL_11B` only (i.e. disable G & N modes)
- set Power Save mode to `WIFI_PS_NONE`

## Notes
We have the Ethernet version of the same hardware. It widely used in the field and have reliable connection. The software is the same for both WiFi and Ethernet version of the device.

- [sdkconfig.zip](https://github.com/espressif/esp-idf/files/4682684/sdkconfig.zip)

liuzfesp, 2020-05-27:
Hi @lexxvir could you help to capture the WiFi packets with wireshark?

lexxvir, 2020-05-27:
Hi @liuzfesp 

Unfortunately no, our customer has no appropriate hardware & experience to capture WiFi traffic.
Anyway, what do you advise to use to capture WiFi traffic on Windows or Linux PC ?
I'll try to check whether it is possible to arrangement this.

esp-HarveyRong, 2020-05-27:
@lexxvir, The recommended Macbook's built-in packet capture tool does not require a packet capture card. It can be used to capture 802.11a/b/g/n/ac packets: 
https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace-mac-os-x-wireless-diagnostics/
Or you can use the tutorial on the official Espressif website to capture packets using Wireshark
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

lexxvir, 2020-05-27:
Hi @esp-HarveyRong 
Thank you for information!

@liuzfesp 
it seems I reproduce same behavior on the my device. I've attached *.pcap file dumped by `airodump-ng`.

I used the following topology:
- The Access Point with IP 10.0.0.1 creates WiFi network (legal)
- The tested device (IP 10.0.0.24) connect to this WiFi network
- The PC (IP 10.0.0.146) connects to Access Point by Ethernet
- The PC works with device through the Access Point

The issue looks like PC (after some time) unable to get answer from the device within 10 seconds timeout. PC repeats their request again, without success. Then PC repeats request again and gets answer on the previous request (I see it by the frame ID that I used in our protocol).

The issue appears at the end of the PCAP trace:
[wifi-traffic-03.cap.zip](https://github.com/espressif/esp-idf/files/4688373/wifi-traffic-03.cap.zip)

esp-HarveyRong, 2020-05-28:
Hi @lexxvir, 
According to the data packet you provided, it was found that the PC did not ACK the ESP32 seq = 34076 packet within 10s, so that ESP32 actively disconnected. Within 10s, ESP32 retransmitted seq = 34076 packets multiple times, but the PC did not respond.
So I think this is the TCP disconnection caused by the PC not responding to ESP32 in time.
[image]

gadget-man, 2020-05-31:
Custom Hardware
Module or chip used: ESP32-WROVER-IB
IDF version v3.3-r4
Build System: Mongoose-os
Compiler version 1.22.0-80-g6c4433a
Operating System: MacOS / Mongoose-os

I think I'm having a similar problem. We have 150 devices deployed and a small number are seeing frequent disconnects, some around every 10 minutes.  The issue causes an MQTT disconnect, however UDP stays up as I have logging continue to run over UDP. 

Once disconnected, the device is unable to re-connect to TCP, query a DNS or reconnect to MQTT until after a reboot. Turning off the Wi-Fi stack then re-enabling it also doesn't reconnect.

Unfortunately I can't provide a packet capture, but do have log details showing the issue from the ESP device (once the TCP times out, after 30 seconds I call wifi-disconnect, then re-connect after 10 seconds but the reconnection fails.
https://pastebin.com/fxfLMNmi

gadget-man, 2020-05-31:
... and here a log showing the original disconnection.  Note this log is collected from a remote device over UDP.
https://pastebin.com/AYcauebu

lexxvir, 2020-06-02:
Hi @esp-HarveyRong 

Thank you for analyze previous log!

Now I've got logs from the real client's device:
[ZOC2006.zip](https://github.com/espressif/esp-idf/files/4715749/ZOC2006.zip)

Both the TTY-log and PCAP files were recorded in the same time on the one device session.

In the `ZOC2006_com101.log` I noticed the following string:
`I (1967986) wifi:bcn_timout,ap_probe_send_start`
What does it mean?
It seems it was the cause of the connection interruption.

In the PCAP log I see a lot of re-transmissions. Which itself leads to the disconnection from the PC side. I'd like to remind that there are few (3-4) devices connected to the PC, but disconnect issues occur only on the one of them.

gadget-man, 2020-06-02:
I've captured a further log, showing the detail of the initial TCP disconnect, followed an number of failed attempts to reconnect and re-establish the MQTT connection. This fails until a re-boot, upon which the device immediately reconnects
https://pastebin.com/y4k0yjhy

esp-HarveyRong, 2020-06-02:
Hi @lexxvir, 
`I (1967986) wifi:bcn_timout,ap_probe_send_start` means that the device has not received the Beacon frame of the AP within the specified time (ESP32 is 6s, that is, 60 Beacon Interval).
 There are many reasons, which may be  memory insufficient of the device or AP power down or  AP RSSI low (signal bad), etc.
The log you provided shows that the AP's Rssi value is  low, which may be the main reason for this phenomenon, or it may be that frequent printing causes insufficient memory.

Hi @gadget-man, Can you use Wireshark to help capture WiFi packets? From the LOG, it is due to the disconnection caused by the device receiving Deauth or Disassoc frames. 
For how to obtain WiFi packets, please refer to https://osxdaily.com/2015/04/23/sniff-packet-capture-packet-trace- mac-os-x-wireless-diagnostics/
Or https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

gadget-man, 2020-06-02:
Hello @esp-HarveyRong,

Thanks for your response. Unfortunately my failing devices are in production at remote locations with domestic clients, so I'm unable to get WiFi packet capture. Is there any way to get further information on the Deauth or Dissasoc frames from additional logging on the ESP32?

esp-HarveyRong, 2020-06-03:
Hi @gadget-man, 
Can you provide more detailed information, such as: whether to connect the encrypted route, encryption type, reproduce frequency, IDF commit ID, can you provide a complete log, including (Info level WiFi log)

lexxvir, 2020-06-03:
Hi @esp-HarveyRong,

Thank you for comments. I don't suspect memory issues because other devices work normally. Also, the issue there was also when I don't printing RSSI to console.

We try to connect external antenna and return to you with feedback.

Any way, thanks for help!

gadget-man, 2020-06-05:
Hi @esp-HarveyRong,

We are using mongoose-os, connecting to AWS IoT Core MQTT with TLS encryption via their ATS endpoint for signed certificates, ESP-IDF 3.3-r4.
The issue is intermittent, occurring anything between 2-3 times/day to every 1-12 minutes on affected devices. Note this only affects around 2% of our devices, all running the same firmware.
Unfortunately I'm only able to provide logs for what I can capture over UDP. I have attached a further log below which may provide additional information. I am working with the guys at mos to see if we can also log the WiFi info log to the UDP stream.
https://pastebin.com/CcZqQgRt

gadget-man, 2020-06-09:
Is there a way to utilise the `esp_log_set_vprintf` to elevate the WiFi log to transmit over UDP?  Using mos, I already get the majority of my logging over UDP, but not the esp level wifi logs which only appear on serial (and as my device is remote I can't access these).

liuzfesp, 2020-07-13:
HI @gadget-man, the MQTT (TCP) connection issue may related to:

- The communication between ESP device and AP
- The communication between AP and the MQTT endpoint

Generally speaking, we need following info to effectively debugging the issue:

- Capture packets between ESP device and AP, namely the WiFi packets
- Capture packets between AP and MATT endpoint, namely the wired packets

Without the packet, it's very difficult to address the issue. 

Another way is to provide a demo code that we can replicate the issue ourselves, then we can debug it locally.

gadget-man, 2020-07-21:
Apologies for the delay. I've now managed to capture some local logging information for a device:
https://pastebin.com/rREqVr3q

You'll see it operates fine for a while, but then starts dropping the connection - it then gets stuck in a disconnect / reconnect loop until a watchdog timer eventually forces a reboot.

Please note that throughout the period of the logging, the device is continuing to successfully send UDP log messages.

gadget-man, 2020-07-23:
I've also now managed to capture some logs at the same time as Wireshark packet data. You can see from the logs that at 2818096 there is a `bcn_timout,ap_probe_send_start`, and after that `WiFi STA: Disconnected, reason: 200` goes in a loop, where it reconnect, gets an IP, but as soon as MQTT0 attempts to reconnect the wifi drops out again with another  `bcn_timout,ap_probe_send_start`. 
[LOG00001.TXT](https://github.com/espressif/esp-idf/files/4964938/LOG00001.TXT)
[MacBook Pro_ch7_2020-07-23_09.31.01.593.pcap.zip](https://github.com/espressif/esp-idf/files/4964940/MacBook.Pro_ch7_2020-07-23_09.31.01.593.pcap.zip)

esp-HarveyRong, 2020-08-25:
@gadget-man,

According to the packet capture and Log analysis, `bcn_timout,ap_probe_send_start` occurs because the AP's Rssi value is low. That is -90dBm.

`I (2753846) wifi: security type: 3, phy: bgn, rssi: -95
I (2753856) wifi: pm start, type: 2
mgos_wifi.c:137         WiFi STA: Connected, BSSID c4:41:1e:67:69:24 ch 7 RSSI -94`

When the Rssi value is low, the device cannot receive the `beacon` frame in complex environment, and `bcn_timout,ap_probe_send_start` will appear.

gadget-man, 2020-08-25:
Thanks. On a couple of other devices, I don't get a full WiFi disconnect. ICMP Ping fails with a timeout and my MQTT keep_alive ping fails, but UDP logging still continues. I will try and capture remote logs of this including the IDF Wi-Fi logging, but can you think of any reason why both TCP and ICMP would fail whilst UDP still continues?

esp-HarveyRong, 2020-08-25:
> On a couple of other devices, I don't get a full WiFi disconnect. ICMP Ping fails with a timeout and my MQTT keep_alive ping fails, but UDP logging still continues.

Can you attach a completed Log?

gadget-man, 2020-08-25:
Here's an example but without the wifi: verbose logging as I can't currently get that over the UDP debugging using Mongoose-OS:
```
[Aug 23 04:15:43.906] iParcelBox-30AEA4C5A118 1337 1598152544.179 2|pn_ping.c:35            PING 64 bytes from 10.0.0.1 icmp_seq=83 ttl=64 time=3 ms
[Aug 23 04:15:53.986] iParcelBox-30AEA4C5A118 1338 1598152554.181 2|pn_ping.c:35            PING 64 bytes from 10.0.0.1 icmp_seq=84 ttl=64 time=5 ms
[Aug 23 04:16:04.044] iParcelBox-30AEA4C5A118 1339 1598152564.183 2|pn_ping.c:35            PING 64 bytes from 10.0.0.1 icmp_seq=85 ttl=64 time=7 ms
[Aug 23 04:16:13.955] iParcelBox-30AEA4C5A118 1340 1598152574.180 2|pn_ping.c:35            PING 64 bytes from 10.0.0.1 icmp_seq=86 ttl=64 time=5 ms
[Aug 23 04:16:24.898] iParcelBox-30AEA4C5A118 1341 1598152585.173 2|pn_ping.c:44            PING From 10.0.0.1 icmp_seq=87 timeout
[Aug 23 04:16:24.901] iParcelBox-30AEA4C5A118 1342 1598152585.177 2|
[Aug 23 04:16:26.965] iParcelBox-30AEA4C5A118 1343 1598152587.182 2|mg_mqtt.c:193           Send PINGREQ
[Aug 23 04:16:26.966] iParcelBox-30AEA4C5A118 1344 1598152587.191 2|mg_lwip_net_if.c:518    0x3f801048 tcp_write 31 = 0, 31 0
[Aug 23 04:16:34.946] iParcelBox-30AEA4C5A118 1345 1598152595.173 2|pn_ping.c:44            PING From 10.0.0.1 icmp_seq=88 timeout
[Aug 23 04:16:34.946] iParcelBox-30AEA4C5A118 1346 1598152595.179 2|
[Aug 23 04:16:44.982] iParcelBox-30AEA4C5A118 1347 1598152605.173 2|pn_ping.c:44            PING From 10.0.0.1 icmp_seq=89 timeout
[Aug 23 04:16:44.982] iParcelBox-30AEA4C5A118 1348 1598152605.177 2|
[Aug 23 04:16:54.903] iParcelBox-30AEA4C5A118 1349 1598152615.173 2|pn_ping.c:44            PING From 10.0.0.1 icmp_seq=90 timeout
[Aug 23 04:16:54.903] iParcelBox-30AEA4C5A118 1350 1598152615.177 2|
[Aug 23 04:17:04.950] iParcelBox-30AEA4C5A118 1351 1598152625.173 2|pn_ping.c:44            PING From 10.0.0.1 icmp_seq=91 timeout
[Aug 23 04:17:04.950] iParcelBox-30AEA4C5A118 1352 1598152625.177 2|
[Aug 23 04:17:14.898] iParcelBox-30AEA4C5A118 1353 1598152635.173 2|pn_ping.c:44            PING From 10.0.0.1 icmp_seq=92 timeout
[Aug 23 04:17:14.901] iParcelBox-30AEA4C5A118 1354 1598152635.177 2|
[Aug 23 04:17:24.896] iParcelBox-30AEA4C5A118 1355 1598152645.173 2|pn_ping.c:44            PING From 10.0.0.1 icmp_seq=93 timeout
[Aug 23 04:17:24.899] iParcelBox-30AEA4C5A118 1356 1598152645.177 2|
[Aug 23 04:17:26.247] iParcelBox-30AEA4C5A118 1357 1598152646.513 2|mg_lwip_net_if.c:142    0x3ffe8864 conn error -13
[Aug 23 04:17:26.249] iParcelBox-30AEA4C5A118 1358 1598152646.524 2|mg_ssl_if_mbedtls.c:31  0x3ffe8864 mbedtls_ssl_flush_output() returned -78 (-0x004e)
[Aug 23 04:17:26.259] iParcelBox-30AEA4C5A118 1359 1598152646.534 2|mg_ssl_if_mbedtls.c:31  0x3ffe8864 mbedtls_ssl_write_record() returned -78 (-0x004e)
[Aug 23 04:17:26.268] iParcelBox-30AEA4C5A118 1360 1598152646.544 2|mg_ssl_if_mbedtls.c:31  0x3ffe8864 mbedtls_ssl_send_alert_message() returned -78 (-0x004e)
[Aug 23 04:17:26.276] iParcelBox-30AEA4C5A118 1361 1598152646.549 2|mgos_mqtt_conn.c:253    MQTT0 Disconnect
[Aug 23 04:17:26.279] iParcelBox-30AEA4C5A118 1362 1598152646.556 2|mgos_event.c:135        ev MOS5 triggered 1 handlers
[Aug 23 04:17:26.287] iParcelBox-30AEA4C5A118 1363 1598152646.563 2|mgos_event.c:135        ev MOS5 triggered 1 handlers
[Aug 23 04:17:26.303] iParcelBox-30AEA4C5A118 1364 1598152646.580 2|init.js:474             MQTT Disconnect
[Aug 23 04:17:26.325] iParcelBox-30AEA4C5A118 1365 1598152646.599 2|init.js:482             Starting MQTT Disconnect Timer: 1073643452
[Aug 23 04:17:26.330] iParcelBox-30AEA4C5A118 1366 1598152646.607 2|mgos_mqtt_conn.c:546    MQTT0 connecting after 1978 ms
[Aug 23 04:17:28.397] iParcelBox-30AEA4C5A118 1367 1598152648.595 2|mgos_mqtt_conn.c:468    MQTT0 connecting to a24buxh69lukw9-ats.iot.eu-west-1.amazonaws.com:8883
[Aug 23 04:17:28.397] iParcelBox-30AEA4C5A118 1368 1598152648.603 2|mgos_event.c:135        ev MOS6 triggered 1 handlers
```

gadget-man, 2020-08-27:
Not sure if it's relevant, but on most of the devices I have displaying this error, the disconnect is VERY consistent in terms of the timing. Here's a screenshot from my dashboard showing one such device, usually happening every 15 mins 16 seconds. Average RSSI > 60%....
[image]

HarveyRong-Esp, 2020-10-22:
`Not sure if it's relevant, but on most of the devices I have displaying this error, the disconnect is VERY consistent in terms of the timing. Here's a screenshot from my dashboard showing one such device, usually happening every 15 mins 16 seconds. Average RSSI > 60%....`

Hi @gadget-man, for this kind of issue, I still hope you can provide data package. Without a data package, it is difficult to solve this issue. Or we can provide demo code to reproduce it ourselves and debug it locally.

HarveyRong-Esp, 2020-11-06:
Hi @gadget-man,
Look at the Log. Use Light sleep. Can you turn off sleep for testing? It is suspected that this may be related to tbtt.
If closing sleep can solve the issue, please help provide the wifi firmware version.

HarveyRong-Esp, 2020-12-08:
HI @gadget-man，Did you turn off the sleep test? Can you provide the WiFi firmware version you are using?

gadget-man, 2020-12-08:
For the particular situation when I provided the logs, turning off sleep mode did solve the problem. However I still sometimes get the repeating error on other devices. Recently one customer found that removing their LG TV from the same wifi network solved the issue - we have no idea what might have been causing a conflict between the two devices but any ideas welcome! Unfortunately I was unable to get a packet capture for this situation.

HarveyRong-Esp, 2020-12-09:
@gadget-man, after the LG TV is connected to the router, will all the devices be disconnected or only the ESP32? If only ESP32 is disconnected, can you provide the WiFi Log of ESP32? What is the model of the router?

gadget-man, 2020-12-09:
I believe the issue was limited to the ESP32 device. Unfortunately I don't have any further details but will try and find out from our customer.

gadget-man, 2020-12-15:
We have had another customer with the same issue - after 8-10 minutes the ESP32 loses the TCP connection to our AWS MQTT server, and an ICMP Ping to the local router IP address times out, however UDP stays online and so we are able to receive remote logging events.  The model of the TV is LG 55UJ701V. As soon as the customer either disconnects the TV or moves it to a different network, the problem goes away. No other devices on their network appear to be affected. Unfortunately I'm not able to get a network packet capture.

HarveyRong-Esp, 2020-12-21:
Hi @gadget-man, is it just TCP disconnected? Has the WiFi disconnect? Unfortunately, we do not have an environment to reproduce issue, so we can only rely on you to provide us with information. This kind of issue can generally only be analyzed by LOG + ESP32 and AP wireless package + AP to AWS wired package. I hope you can Provide such information.

gadget-man, 2020-12-21:
Yes it's only TCP that gets disconnected - WiFi stays live and I continue to receive logs remotely over UDP. Note the issue is not only the AWS connection - ICMP ping to the local router also fails.

We will try and find a way to do some packet capture, and share it here if we are able to do so.

HarveyRong-Esp, 2020-12-23:
@gadget-man， 
Why does UDP communicate normally when Ping fails?
Are you only using 11b? If only use 11b, there is a possibility that LGTV will use up all the bandwidth after connecting to the router. As a result, esp32 will not have the opportunity to preempt, so it may not be able to send out packets. 
Can you change it to 11n and test it?

gadget-man, 2020-12-27:
We'd love to know why UDP is fine but TCP and Ping both fail!!!  We are using bgn as the default network, so that can't be what's causing it.

HarveyRong-Esp, 2020-12-29:
@gadget-man, Usually ping failure indicates that there is a problem with the IP.  Since TCP and UDP are both transport layer protocols and both need to communicate based on the network layer IP protocol, Normally, both TCP and UDP will fail.  Can you tell us in detail? How to ping it? Where is ping?

HarveyRong-Esp, 2021-01-21:
Hi @gadget-man , Are there any updates? Can it be reproduced?

HarveyRong-Esp, 2021-01-23:
Hi @gadget-man 
Are there any updates? If not, I will close issue. If there is issue, please open it at any time.

gadget-man, 2021-01-23:
It still occurs every day on a couple of devices, however they are deployed production devices and I'm unable to get Wireshark packet capture for them at the moment.

Those devices that fail, I'm using the ESP-IDF ICMP Ping script to ping directly from the ESP32 to the local router device - so even a basic ping between say 192.168.1.18 and 192.168.1.1 will fail.

HarveyRong-Esp, 2021-01-23:
You mean the PING  esp32  connected router will fail? Is it possible to check whether the router configuration prohibits PING function.

gadget-man, 2021-01-23:
Yes that's correct - ping will start fine, we get correct ping response for 8-10 minutes, then ping starts to fail, AWS drops, MQTT fails, but UDP stays online, sending mos debug logs remotely to our monitoring server.

HarveyRong-Esp, 2021-01-23:
This can only be done by capturing data packets to check the ping and MQTT failure reasons, or you can provide a minimal example of reproducing, I try to reproduce it locally.

gadget-man, 2021-01-23:
Unfortunately it only occurs on a small number of production devices, I'm unable to replicate locally. As soon as I get get data packet capture I'll share it here.

jack0c, 2022-05-20:
@gadget-man Are you still working on this issue?
Let me repeat the issue: the esp32 devices with connect to the same router with LGTV, lose TCP connection and fail to ping the router, but success to send UDP packet to the sever. Through I don't know if it will recover automatically,  I will try to analyze the issue.
I think the cause may be:
1. fail to receive data from router because of sleep, but should recover very soon
2. can receive data, but discard by MAC layer because of buffer overflow
3. can receive data, but discard by MAC layer because of sequence number

I am not sure why this happen with LGTV, but I think reason 3 maybe the main reason.
If you are still working on this, please provide the commit, we can provide a patch to test.

jack0c, 2023-04-12:
@gadget-man will close this since long time no update, you can reopen it if you want

closed 2023-04-12 by jack0c