espressif/esp-idf #5352: esp_event: Include `event_data_size` in event callback

nebkat, 2020-05-26:
**Is your feature request related to a problem? Please describe.**

In my application I use the event handler library to distribute data received on a UART to other parts of the application. The event handler signature does not include the size of the `event_data`.

    typedef void (*esp_event_handler_t)(void* event_handler_arg, 
            esp_event_base_t event_base, 
            int32_t event_id, 
            void* event_data); /**< function called when an event is posted to the queue */

Because of this, I am forced to package the message in a struct containing the length, which requires an extra `memcpy` (which is then, in my case unnecessarily, repeated in `esp_event_post_to()`).

    struct uart_message {
        size_t length;
        uint8_t message[LENGTH]
    }

    struct uart_message msg;
    uart_message.length = len;
    memcpy(uart_message.message, buf, len);

    esp_event_post(UART_EVENT, UART_EVENT_READ,  msg, sizeof(size_t) + len, portMAX_DELAY);

**Describe the solution you'd like**

I propose that `event_data_size` is added as a argument to `esp_event_handler_t`, so that raw data can be passed using the event handler system. `event_data` is currently the last argument in `esp_event_handler_t`, so it would not cause any major complications, but of course existing handler functions would have to add the parameter.

**Describe alternatives you've considered**

- Above mentioned struct containing length and the data - extra memcpy
- Send two events, one with the length and one with the data - handler has to store length
- Use `event_id` as the data length - multiple event bases required to replace actual event IDs
- 'Raw' event handler registration, allow handlers to be registered with/without the size argument

**Additional context**

I may be taking the wrong approach using the event handler system to pass around data but I think most elements of it would be reused of a custom system was implemented. Eventually my goal is to have different subsystems (e.g. UART 1/2, TCP client, MQTT server) all posting events when they receive data, with internal 'connections' being made between those subsystems at runtime (e.g. UART 1 => TCP client, UART 2 => MQTT server).

igrr, 2020-05-26:
Hi @nebkat, what is the approximate amount of data you are sending (per second) through the esp_event events? If the amount is relatively small, then the additional overhead of `memcpy`-ing the data will likely be much smaller than the overhead added by esp_event itself (i.e. pushing the event into a queue, waking up the esp_event task, walking the lists of event handlers). If the amount is large, then possibly esp_event is not the best mechanism for this purpose, and you would get better performance if you connect the data source and sink with a [ringbuffer](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_additions.html#ring-buffers) (provided that one source needs to be connected to one sink).

> It may be the case that there would be too many places where the event handler signature would have to change to make this worthwhile

We would also need to maintain compatibility with the previous 4.x IDF versions. That seems to be possible, but we would need to create a new callback type, a new set of functions to register the callbacks (and come up with good names for them!), and handle this extra complexity in the esp_event itself. Overall I would say that the solution with a structure of size and data looks like a better approach here.

nebkat, 2020-05-27:
Hi @igrr, thanks for the response! In my case the amount would relatively large (limited only by UART speed), but it is one source to multiple sinks (depending on configuration), which is why esp_event seemed like a good solution. To achieve something similar would probably replicate a lot of what is implemented in esp_event.

It would be slightly disappointing having a second set of functions for the sake of one parameter, but it's also a shame to have to use a second allocation or rewrite parts of the events system.

Another alternative I forgot to mention was allowing an event to be posted directly without the extra `memcpy`, so that the copy to the structure is only done once and the handler signature doesn't have to change?

igrr, 2020-05-27:
> Another alternative I forgot to mention was allowing an event to be posted directly without the extra memcpy, so that the copy to the structure is only done once and the handler signature doesn't have to change?

How would the lifetime of the structure be managed in this case? Since you have multiple subscribers, the subscriber cannot be responsible for freeing the dynamically allocated structure, as there may be other subscribers which have not handled the event yet. This is the reason why esp_event will in most cases allocate a buffer, copy the event data into the buffer, and free the buffer when all the subscribers have handled the event.

If you know the exact number of subscribers, then you could use reference counting to manage the lifetime of these structures. For instance, the structure becomes:
```c++
struct Message {
  int refcnt;
  int size;
  char data[0];
};
```

Initially, you allocate this structure, set `refcnt` to the number of subscribers, fill `size` and `data`, and pass _the pointer_ to the structure to esp_event. The pointer is only 4 bytes, and esp_event will send it along as a value, without copying the content of the structure. Each subscriber (event handler) will need to process the message, decrement `refcnt`, and `free` the structure if refcnt reaches zero.

However, if the set of subscribers is not fixed, then this approach won't work.

> In my case the amount would relatively large (limited only by UART speed)

If each event passes along a relatively large amount of data (so that you have not too many events per second), then it is probably okay to use esp_event in this scenario, even with an additional allocation/memcpy overhead.

nebkat, 2020-05-27:
> How would the lifetime of the structure be managed in this case?

I was thinking it could be freed the same way as normal events inside esp_event but I see how this could be misleading or cause issues so probably best avoided. 

> If each event passes along a relatively large amount of data (so that you have not too many events per second), then it is probably okay to use esp_event in this scenario, even with an additional allocation/memcpy overhead.

I guess I'll stick to this approach then, I was probably overestimating the impact of the additional memcpy. Thanks!

closed 2020-05-27 by nebkat

Aietes, 2020-07-07:
This issue has been closed, but I wonder why this would not be considered in future releases? Having the data length in the event handler would be very useful, and frankly more straightforward than the current implementation.