espressif/esp-idf #5354: Crash when erasing flash with CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY set

devanlai, 2020-05-26:
## Environment

- Development Kit:      Custom PCBA
- Module or chip used:  ESP32-WROVER-E (ECO v3)
- IDF version: v4.2-dev-1660-g7d7521367
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         External 5V

## Problem Description

When using external PSRAM with the option `CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY` enabled and running WiFi, I'm encountering crashes as soon as it attempts to erase internal flash. I've prepared a simplified test case using one of the HTTP server examples with no code changes - only adjusting the `sdkconfig`.

In the simplified test case using the [protocols/http_server/file_serving example](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving), it connects to WiFi and then attempts to format the SPIFFS partition and crashes. The panic handler seems to have issues and instead silently resets most of the time.

### Expected Behavior

Mutual exclusion between flash erase/write access and PSRAM/flash cache access should be managed automatically and correctly when using official ESP IDF APIs. In the sample code, it should not crash when `CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY` is set, WiFi is enabled, and we attempt to format the SPIFFS partition.

### Actual Behavior

It crashes when attempting to erase flash.

### Steps to reproduce

1. Erase the flash on the device so that the SPIFFS partition is invalid
2. Flash the file_serving example using the attached sdkconfig (with WiFi credentials modified for your network)
3. Wait for it to connect to WiFi
4. Observe that the SPIFFS format operation causes the device to crash.

Additional note: running the same application source code on the same hardware but rebuilt without CONFIG_SPIRAM_ALLOW_BS_SEG_EXTERNAL_MEMORY works fine.

### Code to reproduce this issue

This issue can be reproduced using the [file_serving](https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving) example with the attached sdkconfig.

Be sure to update the example WiFi credentials for your test network.

## Debug Logs

```
I (1679) example_connect: Connecting to <redacted>...
W (1679) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (1869) phy: phy_version: 4181, c9a03e2, Mar 10 2020, 12:45:06, 0, 2
I (1879) wifi:mode : sta (24:0a:c4:e1:d0:94)
I (1889) example_connect: Waiting for IP(s)
I (2489) wifi:new:<5,0>, old:<1,0>, ap:<255,255>, sta:<5,0>, prof:1
I (3349) wifi:state: init -> auth (b0)
I (3349) wifi:state: auth -> assoc (0)
I (3359) wifi:state: assoc -> run (10)
I (3379) wifi:connected with <redacted>, aid = 11, channel 5, BW20, bssid = 1c:b7:2c:c6:f8:c0
I (3379) wifi:security type: 3, phy: bgn, rssi: -44
I (3379) wifi:pm start, type: 1

I (3379) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (4059) esp_netif_handlers: example_connect: sta ip: 192.168.1.235, mask: 255.255.255.0, gw: 192.168.1.1
I (4059) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.1.235
I (4559) example_connect: Got IPv6 event: Interface "example_connect: sta" address: fe80:0000:0000:0000:260a:c4ff:fee1:d094, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4559) example_connect: Connected to example_connect: sta
I (4569) example_connect: - IPv4 address: 192.168.1.235
I (4579) example_connect: - IPv6 address: fe80:0000:0000:0000:260a:c4ff:fee1:d094, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4589) example: Initializing SPIFFS
W (4589) SPIFFS: mount failed, -10025. formatting...
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7132
load:0x40078000,len:14128
load:0x40080400,len:4060
0x40080400: _init at ??:?
entry 0x40080688
```

## Other items if possible
- [sdkconfig.txt](https://github.com/espressif/esp-idf/files/4685264/sdkconfig.txt)
[log.file_server.20200526151338.txt](https://github.com/espressif/esp-idf/files/4685278/log.file_server.20200526151338.txt)

Alvin1Zhang, 2020-05-27:
@devanlai Thanks for the report, we will look into.

mahavirj, 2020-05-27:
@devanlai Can confirm this issue. If you would like quick fix then please try below patch:

```
diff --git components/esp32/ld/esp32.extram.bss.ld components/esp32/ld/esp32.extram.bss.ld
index 582f6eb6ea..dc31f16ebe 100644
--- components/esp32/ld/esp32.extram.bss.ld
+++ components/esp32/ld/esp32.extram.bss.ld
@@ -8,7 +8,7 @@ SECTIONS
   {
     _ext_ram_bss_start = ABSOLUTE(.);
     *(.ext_ram.bss*)
-    *libnet80211.a:(.dynsbss .sbss .sbss.* .gnu.linkonce.sb.* .scommon .sbss2.* .gnu.linkonce.sb2.* .dynbss .bss .bss.* .share.mem .gnu.linkonce.b.* COMMON)
+    /**libnet80211.a:(.dynsbss .sbss .sbss.* .gnu.linkonce.sb.* .scommon .sbss2.* .gnu.linkonce.sb2.* .dynbss .bss .bss.* .share.mem .gnu.linkonce.b.* COMMON)*/
     *libpp.a:(.dynsbss .sbss .sbss.* .gnu.linkonce.sb.* .scommon .sbss2.* .gnu.linkonce.sb2.* .dynbss .bss .bss.* .share.mem .gnu.linkonce.b.* COMMON)
     *liblwip.a:(.dynsbss .sbss .sbss.* .gnu.linkonce.sb.* .scommon .sbss2.* .gnu.linkonce.sb2.* .dynbss .bss .bss.* .share.mem .gnu.linkonce.b.* COMMON)
     *libbt.a:(EXCLUDE_FILE (libbtdm_app.a) .dynsbss .sbss .sbss.* .gnu.linkonce.sb.* .scommon .sbss2.* .gnu.linkonce.sb2.* .dynbss .bss .bss.* .share.mem .gnu.linkonce.b.* COMMON)
```
(Note: It seems like WiFi interrupt handler tries to access some symbols from SPIRAM whilst cache is disabled, above patch moves all those symbols back to DRAM)

Our WiFi team will get back with proper fix soon.

devanlai, 2020-05-27:
Hi @mahavirj, I can confirm that the quick fix works for now. It's a good middle ground between crashing with lots of free internal memory vs working but with very little memory left.

Please keep me updated when the proper fix is available.

Thanks!

TimXia, 2020-05-28:
@devanlai @mahavirj  This is a known issue. We are trying to fix it.

TimXia, 2020-05-29:
@mahavirj How did you know the issue is caued by accessing some symbols from SPIRAM in WiFi interrupt handler?

closed 2020-06-12 by igrr