espressif/esp-idf #5356: ESP - NETIF crashes randomly

Adamantan, 2020-05-27:
### Environment
Development Kit: ESP32-HUZZAH32
IDF version v4.0.1
Development Env: CMake
Operating System: Windows
Power Supply: USB
### Problem Description
I am working on a complex gateway solution between UART and a cloud service provider. The messages received on UART are forwarded to the cloud service through an HTTP connection.
The gateway code was working reliably, but since I have changed to the NETIF component due to the deprication of the old TCP/IP components, the program keeps intermittently crashing in the middle of a transfer. It keeps doing that even tough several successful transfers have been executed before that.
I have managed to trace it back, that the crash always happens at this line of the "transport.c" in the tcp_transport component:

 return t->_connect(t, host, port, timeout_ms);

None of the arguments are invalid, either for the successful or the crashing transfers.
E (77840) TRANSPORT: CC6.1.3 - t: 0x3ffd8f6c
E (77850) TRANSPORT: CC6.1.3 - port: 443
E (77850) TRANSPORT: CC6.1.3 - timeout: 5000

### Expected behavior
The transfer is successful every time

I am stuck on this since days, please if anybody have seen this before, any info is appreciated.

david-cermak, 2020-05-27:
Hi @Adamantan 

Could you please share the original log output when the crash happens including the backtrace (decoded to file/line numbers)?

Also, please correct the version number, esp-netif was introduced in `v4.1`, so the *IDF version v4.0.1* looks like a mistake.

Adamantan, 2020-05-29:
Hi @david-cermak 

I have corrected the version number, you were right about the oversight.
The version i use is v4.2-dev-994-gc9f29e0b5.

As requested, here is the original log with the backtrace:
```

I (245332) esp_http_client_perform: 1
E (245332) HTTP_CLIENT: CC1
E (245332) HTTP_CLIENT: CC2
E (245342) HTTP_CLIENT: CC3
E (245342) HTTP_CLIENT: Begin connect to: https://api.allthingstalk.io:443
E (245352) HTTP_CLIENT: CC3.1 - Transport Type: 0x3ffdfe3c
E (245352) HTTP_CLIENT: CC4
E (245362) HTTP_CLIENT: CC6
E (245362) HTTP_CLIENT: CC6 - Client: 0x3ffdfb9c
E (245372) HTTP_CLIENT: CC6 - Transport Type: 0x3ffdfe3c
E (245372) HTTP_CLIENT: CC6 - Conninfo host : api.allthingstalk.io
E (245382) HTTP_CLIENT: CC6 - Conn port     : 443
E (245382) HTTP_CLIENT: CC6 - Conn timeout  : 10000
E (245392) HTTP_CLIENT: CC6.1
E (245392) TRANSPORT: CC6.1.1
E (245402) TRANSPORT: CC6.1.2
E (245402) TRANSPORT: CC6.1.3
E (245402) TRANSPORT: CC6.1.3 - t: 0x3ffdfe3c
E (245412) TRANSPORT: CC6.1.3 - host: api.allthingstalk.io
E (245422) TRANSPORT: CC6.1.3 - port: 443
E (245422) TRANSPORT: CC6.1.3 - timeout: 10000
E (245432) TRANSPORT: All is cool
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40117078: 22891289 0000f01d 81004136
0x40117078: mbedtls_mpi_free at C:/msys32/home/adama/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:115

Core  0 register dump:
PC      : 0x4011707c  PS      : 0x00060330  A0      : 0x00040020  A1      : 0x3ffbe520
0x4011707c: mbedtls_mpi_free at C:/msys32/home/adama/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:117

A2      : 0x3ffbe540  A3      : 0x3ffbe588  A4      : 0x00000008  A5      : 0x00000000
A6      : 0x00000200  A7      : 0x0000021f  A8      : 0x00000000  A9      : 0x3ffbe500
A10     : 0x3ffdd004  A11     : 0x00000010  A12     : 0x00000020  A13     : 0x3ffddb90  
A14     : 0x62c1d8d0  A15     : 0xfb555368  SAR     : 0x0000001f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace:0x40117079:0x3ffbe520 0x0004001d:0x3ffbe540 |<-CORRUPTED
0x40117079: mbedtls_mpi_free at C:/msys32/home/adama/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:115
ELF file SHA256: 275bac15f4703954

```

david-cermak, 2020-06-01:
@Adamantan Thanks for posting the log!
It is a bit hard to understand the issue when the backtrace is corrupted. Does it crash always this way -- i.e. IllegalInstruction after the `mbedtls_mpi_free()` or differently every time?
At this point I would suggest checking for stack overflow, enabling [this watchpoint](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-freertos-watchpoint-end-of-stack) in the project configuration menu is helpful,
as well as checking for [high watermark](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv427uxTaskGetStackHighWaterMark12TaskHandle_t) within your tasks.

You've also mentioned earlier that the crash initiates somewhere in the tcp-transport, what indicates this was the case?

Let me also ask which APIs from ESP-NETIF your project uses? (converting from the tcpip-adapter API should be more-less transparent, not sure what could be causing this trouble - other than stack issues)

Could you please also [enable core dump](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/core_dump.html#print-core-dump-to-uart), decode it and post here?

JohnL1926, 2020-07-10:
Was there any resolution to this issue? I seem to be getting the same thing. The exception is below, with the same corrupted backtrace.

```
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4010602c: 22891289 0000f01d 81004136
0x4010602c: mbedtls_mpi_free at /home/****/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:115

Core  1 register dump:
PC      : 0x40106030  PS      : 0x00060830  A0      : 0x00040020  A1      : 0x3ffc8480  
0x40106030: mbedtls_mpi_free at /home/****/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:117

A2      : 0x3ffc84a0  A3      : 0x00000000  A4      : 0x3ffd1400  A5      : 0x00000000  
A6      : 0x3ffd0474  A7      : 0x00060420  A8      : 0x00000000  A9      : 0x3ffc8460  
A10     : 0x3ffd047c  A11     : 0x00000008  A12     : 0x947add4f  A13     : 0x3ffd1400  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace:0x4010602d:0x3ffc8480 0x0004001d:0x3ffc84a0 |<-CORRUPTED
0x4010602d: mbedtls_mpi_free at /home/****/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:115
```

I've enabled strong stack smashing protection, comprehensive heap memory debugging, and core dump to flash. I wish I could give the decoded core dump, but I'm running into another issue there as well. There's another exception while dumping:
```
I (14503) esp_core_dump_flash: Save core dump to flash...
I (14509) esp_core_dump_elf: Found tasks: 11
D (14514) esp_core_dump_elf: ================= Calc data size ===============
D (14522) esp_core_dump_port: Stack len = 3036 (3ffcaf40 3ffcbb1c)
D (14528) esp_core_dump_elf: Task #0, (TCB:3ffc9600) is current crashed task.
D (14535) esp_core_dump_port: Stack len = 544 (3ffc3510 3ffc3730)
D (14541) esp_core_dump_port: Stack len = 544 (3ffc0f70 3ffc1190)
D (14547) esp_core_dump_port: Stack len = 2764 (3ffcb050 3ffcbb1c)
D (14554) esp_core_dump_port: Current task (TCB:3ffc9600), EXIT/PC/PS/A0/SP deadbeef 40107db8 60630 40020 3ffcb110
D (14564) esp_core_dump_port: Add regs for task 0x3ffc9600
D (14570) esp_core_dump_port: Bad TCB addr=0!
D (14574) esp_core_dump_elf: The current crashed task has broken (TCB:0)!
D (14581) esp_core_dump_port: Add regs for task 0x0
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d3f1d  PS      : 0x00060333  A0      : 0x800d4288  A1      : 0x3ffcad90  
0x400d3f1d: esp_core_dump_get_regs_from_stack at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_port.c:359

A2      : 0x00000001  A3      : 0x3ffbe8af  A4      : 0x3ffb5e70  A5      : 0xffffffff  
A6      : 0x00000000  A7      : 0xfffffffe  A8      : 0x0000006f  A9      : 0x3ffcad00  
A10     : 0x00000000  A11     : 0x3ffcad21  A12     : 0x3ffcaccf  A13     : 0x00000030  
A14     : 0x00000000  A15     : 0x3ffcacd6  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace:0x400d3f1a:0x3ffcad90 0x400d4285:0x3ffcadb0 0x400d4b6e:0x3ffcade0 0x400d4c3e:0x3ffcae10 0x400d4d7c:0x3ffcae40 0x400d4f1e:0x3ffcae80 0x400d5136:0x3ffcaec0 0x400d521d:0x3ffcaee0 0x400d453a:0x3ffcaf30 0x400d3ce5:0x3ffcaf50 0x40088232:0x3ffcaf70 0x40087b8e:0x3ffcafe0 0x40087f69:0x3ffcb030 0x4008283e:0x3ffcb050 0x40107db5:0x3ffcb110 0x0004001d:0x3ffcb130 |<-CORRUPTED
0x400d3f1a: esp_core_dump_get_regs_from_stack at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_port.c:355 (discriminator 1)

0x400d4285: esp_core_dump_get_task_regs_dump at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_port.c:426

0x400d4b6e: elf_add_regs at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_elf.c:261

0x400d4c3e: elf_process_task_regdump at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_elf.c:348

0x400d4d7c: elf_process_tasks_regs at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_elf.c:447

0x400d4f1e: elf_write_tasks_data at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_elf.c:491 (discriminator 2)

0x400d5136: esp_core_dump_do_write_elf_pass at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_elf.c:562 (discriminator 2)

0x400d521d: esp_core_dump_write_elf at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_elf.c:595 (discriminator 2)

0x400d453a: esp_core_dump_write at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_common.c:248 (discriminator 3)

0x400d3ce5: esp_core_dump_to_flash at /home/****/esp/esp-idf/components/espcoredump/src/core_dump_flash.c:257

0x40088232: esp_panic_handler at /home/****/esp/esp-idf/components/esp_system/panic.c:286

0x40087b8e: panic_handler at /home/****/esp/esp-idf/components/esp_system/port/panic_handler.c:522

0x40087f69: xt_unhandled_exception at /home/****/esp/esp-idf/components/esp_system/port/panic_handler.c:536

0x4008283e: _xt_user_exc at /home/****/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:697

0x40107db5: mbedtls_mpi_free at /home/****/esp/esp-idf/components/mbedtls/mbedtls/library/bignum.c:115

ELF file SHA256: d6d045958d7c5f02

Re-entered core dump! Exception happened during core dump!
```

Any suggestions? I'm new to debugging with the ESP32 so any tips are appreciated.

gerekon, 2020-08-04:
> `D (14570) esp_core_dump_port: Bad TCB addr=0!`

@JohnL1926 Looks like FrreeRTOS task list is corrupted. We are preparing the fix for coredump. But I am afraid that in case of huge memory overwriting (e.g. buffer overflow) it will not help to point to the exact cause of the problem.

Alvin1Zhang, 2021-12-27:
@Adamantan @JohnL1926 Thanks for reporting and sharing the details, would you please help share if any updates for the issue? Thanks.

closed 2022-03-17 by Alvin1Zhang