espressif/esp-idf #5357: WIFI MESH 和 蓝牙同时启用 出现错误

fzboffice, 2020-05-28:
W (2570) wifi:[beacon]interval:100ms
I (2570) wifi:mode : sta (84:0d:8e:c6:20:a4) + softAP (84:0d:8e:c6:20:a5)
I (2590) wifi:Total power save buffer number: 8
I (2590) wifi:Init max length of beacon: 752/752
I (2590) wifi:Init max length of beacon: 752/752
I (2600) mesh: <nvs>read layer:0, err:0x1102
I (2600) mesh: <nvs>read assoc:0, err:0x1102
I (2600) mesh: [IO]disable self-organizing<reconnect>
W (2610) wifi:[beacon]interval:100ms
I (2610) mesh: [CONFIG]invalid router settings, ssid_len:0, ssid:, bssid:00:00:00:00:00:00
I (3580) wifi:Set ps type: 0

E (3580) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!

abort() was called at PC 0x40197533 on core 0
0x40197533: pm_set_sleep_type at ??:?

Backtrace:0x4009645e:0x3ffd6980 0x4009610d:0x3ffd69a0 0x400993ee:0x3ffd69c0 0x40197533:0x3ffd6a30 0x401813c0:0x3ffd6a50 0x4017ff36:0x3ffd6a70 0x400909d2:0x3ffd6a90 0x400995b9:0x3ffd6ac0
0x4009645e: panic_abort at /mnt/d/esp/esp-mdf/esp-idf/components/esp_system/panic.c:330

0x4009610d: esp_system_abort at /mnt/d/esp/esp-mdf/esp-idf/components/esp_system/system_api.c:68

0x400993ee: abort at /mnt/d/esp/esp-mdf/esp-idf/components/newlib/abort.c:46

0x40197533: pm_set_sleep_type at ??:?

0x401813c0: wifi_set_ps_process at ??:?

0x4017ff36: ieee80211_ioctl_process at ??:?

0x400909d2: ppTask at ??:?

0x400995b9: vPortTaskWrapper at /mnt/d/esp/esp-mdf/esp-idf/components/freertos/xtensa/port.c:143

ELF file SHA256: dcafe4505242a868

CPU halted.

mesh将wifi的ps设置为WIFI_PS_NONE，引起wifi报错

Alvin1Zhang, 2020-05-28:
@fzboffice Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

fzboffice, 2020-05-28:
> @fzboffice Thanks for reporting. Would you please help provide more details as suggested in the [issue [...]

- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it):ESP-IDF v4.2-dev-1126-gd85d3d969-dirty
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:    Linux

Alvin1Zhang, 2020-05-28:
Thanks for sharing the details, we will look into.

fzboffice, 2020-06-17:
@Alvin1Zhang 您好，有进展吗

ESP-iPENCIL, 2020-06-17:
will fix it and let you know after the merge. @fzboffice @dongdong004

fzboffice, 2020-06-17:
@ESP-iPENCIL 万分感谢!!!

Alvin1Zhang, 2020-07-01:
Thanks for reporting and late reply. We have a fix under internal reviewing, the issue will be updated and closed once the fix is available on GitHub.

closed 2020-07-10 by espressif-bot

poisonousfish, 2020-08-17:
> 感谢您的举报和及时回复。我们正在内部审核中提供一个修复程序，一旦该修复程序在GitHub上可用，此问题将被更新并关闭。
This issue does not appear to have been fixed at 4.1 RC,and I have the same problem.
这个问题似乎在4.1RC版本中仍未修复,我也有一样的问题。

Alvin1Zhang, 2020-08-17:
We have back ported the fix to release/4.1, and the fix is available at https://github.com/espressif/esp-idf/commit/8d54933e10266f4f606cc87e2564a25c96282cc2. Would you please help check if the issue still happens? Thanks.

poisonousfish, 2020-08-17:
> Thanks for reporting and late reply. We have a fix under internal reviewing, the issue will be updated and closed once the fix is available on GitHub.
oh,the issue still happens，I'm a little different from them, I'm just using the wifi

I (1374) wifi:config NVS flash: enabled
I (1374) wifi:config nano formating: disabled
I (1374) wifi:Init dynamic tx buffer num: 32
I (1374) wifi:Init data frame dynamic rx buffer num: 32
I (1384) wifi:Init management frame dynamic rx buffer num: 32
I (1384) wifi:Init management short buffer num: 32
I (1394) wifi:Init static rx buffer size: 1600
I (1394) wifi:Init static rx buffer num: 10
I (1394) wifi:Init dynamic rx buffer num: 32
I (1414) wifi:mode : sta (a4:cf:12:64:5f:68)
I (1414) wifi:Set ps type: 0

E (1414) wifi:Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!

YouDONG-ESP, 2020-11-03:
Hi @poisonousfish, is this issue still happens?  Normally it's only happening in wifi Bluetooth coexist mode

wishinlife, 2022-11-09:
@YouDONG-ESP @Alvin1Zhang I have the same problem on release/v5.0 only wifi mesh and ble is open at the same time. 
Can the problem be fixed? Or is there some kind of usage restriction?

Espressif-liuuuu, 2022-11-09:
hi @wishinlife , could you please set wifi ps type to min mode by calling `esp_wifi_set_ps(1)`? Thats a limitation for coexistence.

Espressif-liuuuu, 2022-11-10:
hi @wishinlife , thanks for your report.
Could you please provide us with an example to let us know the order of calling? I`m wondering if there is any issue.

wishinlife, 2022-11-10:
@Espressif-liuuuu I don't know how to upload files. I just merged the official _internal_communication_ and _bleprph_ examples into one. And start wifi mesh before ble in app_main function.

In addition, I found a bug in NimBle, The macro BLE_HS_ADV_MAX_SZ is define small, the following code will cause the advertisement data to have insufficient buffer length, ble_gap_adv_set_fields function will return value 4.
`struct ble_hs_adv_fields fields;
    const char *name;
    int rc;

    /**
     *  Set the advertisement data included in our advertisements:
     *     o Flags (indicates advertisement type and other general info).
     *     o Advertising tx power.
     *     o Device name.
     *     o 16-bit service UUIDs (alert notifications).
     */

    memset(&fields, 0, sizeof fields);

    /* Advertise two flags:
     *     o Discoverability in forthcoming advertisement (general)
     *     o BLE-only (BR/EDR unsupported).
     */
    fields.flags = BLE_HS_ADV_F_DISC_GEN |
                   BLE_HS_ADV_F_BREDR_UNSUP;

    /* Indicate that the TX power level field should be included; have the
     * stack fill this value automatically.  This is done by assigning the
     * special value BLE_HS_ADV_TX_PWR_LVL_AUTO.
     */
    fields.tx_pwr_lvl_is_present = 1;
    fields.tx_pwr_lvl = BLE_HS_ADV_TX_PWR_LVL_AUTO;

    name = ble_svc_gap_device_name();
    fields.name = (uint8_t *)name;
    fields.name_len = strlen(name);
    fields.name_is_complete = 1;

    fields.uuids128 = (ble_uuid128_t[]) {
        gatt_svr_svc_uuid
    };
    fields.num_uuids128 = 1;
    fields.uuids128_is_complete = 1;

    rc = ble_gap_adv_set_fields(&fields);
    if (rc != 0) {
        ESP_LOGE(TAG, "error setting advertisement data; rc=%d", rc);
        return;
    }`