espressif/esp-idf #5358: BLE passive scan stops working

KaeLL, 2020-05-28:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32D0WDQ6 (revision 1)
- IDF version: v4.2-dev-1660-g7d7521367
- Build System:         Make
- Compiler version: esp-2020r1 8.2.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         external 5V

## Problem Description

My app has a passive scan functionality implemented, in which it scans for beacons, supposedly without stopping ( `esp_ble_gap_start_scanning(0)` ), using Bluedroid (it's a slightly modified version of the ble_ibeacon example).
The problem I'm having is that the scan apparently stops working. By stops working, I mean the callback registered by the `esp_ble_gap_register_callback` function isn't called at all after a certain amount of time (It varied between ~30mins to ~100minutes in my tests).
I'm also using idf's WiFi and MQTT, on the latest snapshot of the master branch (7d75213)

I found this old issue
https://github.com/espressif/esp-idf/issues/911
which seems to be about the same problem, though the manifestation is different, due to there being no error bluetooth logs, and the OP's proposed solution not working (I tested it). I also tried forcing the antenna calibration as suggested by TianHao, but it didn't work also.

Here's the `esp_ble_scan_param_t`
```cpp
static esp_ble_scan_params_t ble_scan_params = {
	.scan_type          = BLE_SCAN_TYPE_PASSIVE,
	.own_addr_type      = BLE_ADDR_TYPE_PUBLIC,
	.scan_filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
	.scan_duplicate     = BLE_SCAN_DUPLICATE_DISABLE,
	.scan_interval      = 0x50,
	.scan_window        = 0x30};
```
Here are the controller menuconfig options
[image]

And the Bluedroid options
[image]

### Expected Behavior

Scan not stop

### Actual Behavior

Scan stopping

### Steps to reproduce

ble_ibeacon example + wifi + mqtt + settings shown above
run app until the callback stops getting called

KaeLL, 2020-05-28:
Of the last couple times I was able to reproduce this issue, it seems to happen right after the board receives MQTT data, and writes it to the NVS partition.
The thing is that this entire operation (receiving of data data over MQTT, parsing and writing of it to the NVS) is done periodically, and after a random run of this operation, the scan stops working.

I was having another issue, that also happened in the same fashion (randomly, but during the nvs writing operation) during the execution of the `nvs_set_blob` function (as I recall from the stack trace), in which the interrupt watchdog would trigger and cause a restart. Unfortunately I didn't save the stack trace to post it here. I had no reason to think these two problems were even remotely related, but I now I'm under the impression I might've been wrong about this.

Alvin1Zhang, 2020-05-29:
@KaeLL Thanks for the detailed report, we will look into.

KaeLL, 2020-05-29:
Ok. It's about ~20 hours of board uptime now, with the board doing the same thing, only that I removed the periodic NVS writes (it only receives data AND performs a NVS write at initialization; further receives of the same data isn't written to the NVS if the data is the same), and the scan is working. Evidence pointing to this being an issue with the constant NVS writes is piling up.  
I don't know exactly how the Bluetooth stack interacts with the NVS, more specifically if it's constantly reading it, but my app only reads from it at initialization (further reads are done from memory, since the data is kept there during board operation), and the rest is just writes, only that now they happen way less frequently.  
I will be keeping this optimization to avoid unnecessary writes, but my badly optimized code from before seems to reveal a problem that still exists and the issue can still happen anytime.

WCCWCC, 2020-06-01:
Hi @KaeLL ，
 * Can you tell me which branch of gitlab or github you are currently using?

  * I will update the wifi and bluetooth libs on the branch you specified for debug scan stop issues.

 * You need to update the application to the corresponding debug branch in order to use the debug version of the library. After reproducing the problem, I can find the root cause through the log.

KaeLL, 2020-06-01:
> * Can you tell me which branch of gitlab or github you are currently using?

> I'm also using idf's WiFi and MQTT, on the latest snapshot of the master branch (7d75213)

KaeLL, 2020-06-02:
Any updates regarding this issue?

WCCWCC, 2020-06-03:
Hi @KaeLL ,

    I will provide you with a debug branch on github before this evening.

WCCWCC, 2020-06-03:
Hi @KaeLL ,
  *  esp32 -> /component/esp_wifi_lib_esp32
  *   libbtdm_app.a ->  /component/bt/controller/lib
  *   hciblecmds.c  ->  /component/bt/host/bluedroid/stack/hcic/hciblecmds.c

[user.zip](https://github.com/espressif/esp-idf/files/4722411/user.zip)

 *  Step1:  After decompressing the file in the attachment, replace it with the corresponding log.
 *  Step2:  extern void IRAM_ATTR print_log_info();  When you detect that scan stops, please call this function to output log information.

KaeLL, 2020-06-03:
I will do that in a moment and get back to you with the results, but earlier I said
> I was having another issue, that also happened during the execution of the nvs_set_blob function, though intermittently, as it didn't happen consistently, in which the interrupt watchdog would trigger and cause a restart. Unfortunately I didn't save the stack trace to post it here.

Well, here it is
```
ASSERT_PARAM(-218959118 0), in arch_main.c at line 327
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x4008931e  PS      : 0x00060634  A0      : 0x8019d9e8  A1      : 0x3ffbf5f0
0x4008931e: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054
A6      : 0x3ffbdbdc  A7      : 0x60008050  A8      : 0x80089319  A9      : 0x3ffbf5d0
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40089251  LEND    : 0x40089258  LCOUNT  : 0x00000000
0x40089251: r_assert_param at ??:?

0x40089258: r_assert_param at ??:?

Core  0 was running in ISR context:
EPC1    : 0x401b6c47  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008931e
0x401b6c47: uart_hal_write_txfifo at /esp-idf/components/soc/src/hal/uart_hal_iram.c:35

0x4008931e: r_assert_param at ??:?

Backtrace:0x4008931b:0x3ffbf5f0 0x4019d9e5:0x3ffbf610 0x40019fb5:0x3ffbf630 0x40046683:0x3ffbf660 0x40047515:0x3ffbf680 0x4008bbf5:0x3ffbf6a0 0x4008b3c5:0x3ffbf6c0 0x4008d37e:0x3ffbf6e0 0x4008e257:0x3ffbf700 0x4008589d:0x3ffbf720 0x4008640b:0x3ffe1760 0x40086d55:0x3ffe1780 0x40086d61:0x3ffe17a0 0x4008e555:0x3ffe17c0 0x40086ac3:0x3ffe17e0 0x40086ca5:0x3ffe1810 0x40115d4d:0x3ffe1830 0x40114aa5:0x3ffe1850 0x4011510a:0x3ffe1870 0x40113974:0x3ffe1900 0x40113ca3:0x3ffe1970 0x401141ed:0x3ffe19e0 0x40112c13:0x3ffe1a00 0x4012e4d3:0x3ffe1a40 0x4012e68d:0x3ffe1a60 0x400fec65:0x3ffe1a80 0x400fda85:0x3ffe1ac0 0x400fdb8f:0x3ffe1af0 0x401772d2:0x3ffe1b20 0x4017770c:0x3ffe1b40 0x401778c6:0x3ffe1b80 0x40177d80:0x3ffe1bb0
0x4008931b: r_assert_param at ??:?

0x4019d9e5: r_platform_reset at ??:?

0x4008bbf5: r_lld_evt_end at ??:?

0x4008b3c5: r_lld_evt_end_isr at ??:?

0x4008d37e: r_rwble_isr at ??:?

0x4008e257: r_rwbtdm_isr_wrapper at intc.c:?

0x4008589d: _xt_lowint1 at /esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x4008640b: spi_flash_disable_interrupts_caches_and_other_cpu at /esp-idf/components/spi_flash/cache_utils.c:145 (discriminator 1)

0x40086d55: cache_disable at /esp-idf/components/spi_flash/spi_flash_os_func_app.c:55

0x40086d61: spi1_start at /esp-idf/components/spi_flash/spi_flash_os_func_app.c:83

0x4008e555: spiflash_start at /esp-idf/components/spi_flash/esp_flash_api.c:77

0x40086ac3: esp_flash_read at /esp-idf/components/spi_flash/esp_flash_api.c:544

0x40086ca5: spi_flash_read at /esp-idf/components/spi_flash/esp_flash_api.c:760

0x40115d4d: nvs::nvs_flash_read(unsigned int, void*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_ops.cpp:76

0x40114aa5: nvs::Page::readEntry(unsigned int, nvs::Item&) const at /esp-idf/components/nvs_flash/src/nvs_page.cpp:791

0x4011510a: nvs::Page::cmpItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int, unsigned char, nvs::VerOffset) at /esp-idf/components/nvs_flash/src/nvs_page.cpp:351

0x40113974: nvs::Storage::cmpMultiPageBlob(unsigned char, char const*, void const*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_storage.cpp:511

0x40113ca3: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_storage.cpp:293

0x401141ed: nvs::NVSHandleSimple::set_blob(char const*, void const*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_handle_simple.cpp:52

0x40112c13: nvs_set_blob at /esp-idf/components/nvs_flash/src/nvs_api.cpp:380

0x4012e4d3: set_blob_nvs at app/components/config_ra/config_ra.c:45 (discriminator 6)

0x4012e68d: cfgSave at app/components/config_ra/config_ra.c:146

0x400fec65: settings_handler at app/components/mqtt_ra/mqtt_ra_cb.c:377 (discriminator 6)

0x400fda85: process_data at app/components/mqtt_ra/mqtt_ra.c:54

0x400fdb8f: mqtt_event_handler at app/components/mqtt_ra/mqtt_ra.c:94 (discriminator 9)

0x401772d2: esp_mqtt_dispatch_event at /esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:812

0x4017770c: deliver_publish at /esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:860 (discriminator 4)

0x401778c6: mqtt_process_receive at /esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1109 (discriminator 6)

0x40177d80: esp_mqtt_task at /esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1273

Core  1 register dump:
PC      : 0x401b251a  PS      : 0x00060f34  A0      : 0x800d1df5  A1      : 0x3ffcfed0
0x401b251a: esp_pm_impl_waiti at /esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x3ffcff9c  A4      : 0x80092224  A5      : 0x3ffbfcd0
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d1dba  A9      : 0x3ffcfea0
A10     : 0x00000000  A11     : 0x3ffcff9c  A12     : 0x80082544  A13     : 0x3ffbfcb0
A14     : 0x3ffc97c8  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x401b2517:0x3ffcfed0 0x400d1df2:0x3ffcfef0 0x400904d9:0x3ffcff10
0x401b2517: esp_pm_impl_waiti at /esp-idf/components/esp32/pm_esp32.c:484

0x400d1df2: esp_vApplicationIdleHook at /esp-idf/components/esp_common/src/freertos_hooks.c:63

0x400904d9: prvIdleTask at /esp-idf/components/freertos/tasks.c:3385 (discriminator 1)
```
I incresed the amount of MQTT updates, which increases the number of NVS writes, and crashes the app after some time.
I'm posting this because I'm not sure if the errors are related. If they're not, I can open another issue, and provide a PoC, though I'll be able to do that only for next week.

KaeLL, 2020-06-03:
> * Step1:  After decompressing the file in the attachment, replace it to the corresponding directory.

Done. But with the caveat that for hciblecmds.c, there is no /hcic/ folder, only /hci/. I created the /hcic/ folder, and copied the file there.
Also, when I build the app, the linker complains about `undefined reference to 'test_func_print'`
```
/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/nvmd/Documents/AR/Documents/VSCode/esp_idf/master/esp-idf/components/bt/controller/lib/libbtdm_app.a(lld_evt.o): in function `r_lld_evt_schedule':
lld_evt.c:(.iram1.text+0x644): undefined reference to `test_func_print'
/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: lld_evt.c:(.iram1.text+0x661): undefined reference to `test_func_print'
/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/nvmd/Documents/AR/Documents/VSCode/esp_idf/master/esp-idf/components/bt/controller/lib/libbtdm_app.a(lld_evt.o): in function `r_lld_evt_init_evt':
lld_evt.c:(.iram1.text+0xb3b): undefined reference to `test_func_print'
/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/nvmd/Documents/AR/Documents/VSCode/esp_idf/master/esp-idf/components/bt/controller/lib/libbtdm_app.a(llm.o): in function `r_llm_init_hack':
llm.c:(.text+0x4e8): undefined reference to `test_func_print'
/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: llm.c:(.text+0x4fe): undefined reference to `test_func_print'
collect2: error: ld returned 1 exit status
```

KaeLL, 2020-06-03:
Will do that, but before that, I need a way to recover from this issue, as it's a critical one for my application. I'm thinking about testing a restart of the bluetooth stack (controller, bluedroid, etc). Any tips, warnings or anything on how to recover from it?  

Also, earlier I said:
> I was having another issue, that also happened during the execution of the nvs_set_blob function, though intermittently, as it didn't happen consistently, in which the interrupt watchdog would trigger and cause a restart. Unfortunately I didn't save the stack trace to post it here.

Here it is
```
ASSERT_PARAM(-218959118 0), in arch_main.c at line 327
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x4008931e  PS      : 0x00060634  A0      : 0x8019d9e8  A1      : 0x3ffbf5f0
0x4008931e: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054
A6      : 0x3ffbdbdc  A7      : 0x60008050  A8      : 0x80089319  A9      : 0x3ffbf5d0
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40089251  LEND    : 0x40089258  LCOUNT  : 0x00000000
0x40089251: r_assert_param at ??:?

0x40089258: r_assert_param at ??:?

Core  0 was running in ISR context:
EPC1    : 0x401b6c47  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008931e
0x401b6c47: uart_hal_write_txfifo at /esp-idf/components/soc/src/hal/uart_hal_iram.c:35

0x4008931e: r_assert_param at ??:?

Backtrace:0x4008931b:0x3ffbf5f0 0x4019d9e5:0x3ffbf610 0x40019fb5:0x3ffbf630 0x40046683:0x3ffbf660 0x40047515:0x3ffbf680 0x4008bbf5:0x3ffbf6a0 0x4008b3c5:0x3ffbf6c0 0x4008d37e:0x3ffbf6e0 0x4008e257:0x3ffbf700 0x4008589d:0x3ffbf720 0x4008640b:0x3ffe1760 0x40086d55:0x3ffe1780 0x40086d61:0x3ffe17a0 0x4008e555:0x3ffe17c0 0x40086ac3:0x3ffe17e0 0x40086ca5:0x3ffe1810 0x40115d4d:0x3ffe1830 0x40114aa5:0x3ffe1850 0x4011510a:0x3ffe1870 0x40113974:0x3ffe1900 0x40113ca3:0x3ffe1970 0x401141ed:0x3ffe19e0 0x40112c13:0x3ffe1a00 0x4012e4d3:0x3ffe1a40 0x4012e68d:0x3ffe1a60 0x400fec65:0x3ffe1a80 0x400fda85:0x3ffe1ac0 0x400fdb8f:0x3ffe1af0 0x401772d2:0x3ffe1b20 0x4017770c:0x3ffe1b40 0x401778c6:0x3ffe1b80 0x40177d80:0x3ffe1bb0
0x4008931b: r_assert_param at ??:?

0x4019d9e5: r_platform_reset at ??:?

0x4008bbf5: r_lld_evt_end at ??:?

0x4008b3c5: r_lld_evt_end_isr at ??:?

0x4008d37e: r_rwble_isr at ??:?

0x4008e257: r_rwbtdm_isr_wrapper at intc.c:?

0x4008589d: _xt_lowint1 at /esp-idf/components/freertos/xtensa/xtensa_vectors.S:1105

0x4008640b: spi_flash_disable_interrupts_caches_and_other_cpu at /esp-idf/components/spi_flash/cache_utils.c:145 (discriminator 1)

0x40086d55: cache_disable at /esp-idf/components/spi_flash/spi_flash_os_func_app.c:55

0x40086d61: spi1_start at /esp-idf/components/spi_flash/spi_flash_os_func_app.c:83

0x4008e555: spiflash_start at /esp-idf/components/spi_flash/esp_flash_api.c:77

0x40086ac3: esp_flash_read at /esp-idf/components/spi_flash/esp_flash_api.c:544

0x40086ca5: spi_flash_read at /esp-idf/components/spi_flash/esp_flash_api.c:760

0x40115d4d: nvs::nvs_flash_read(unsigned int, void*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_ops.cpp:76

0x40114aa5: nvs::Page::readEntry(unsigned int, nvs::Item&) const at /esp-idf/components/nvs_flash/src/nvs_page.cpp:791

0x4011510a: nvs::Page::cmpItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int, unsigned char, nvs::VerOffset) at /esp-idf/components/nvs_flash/src/nvs_page.cpp:351

0x40113974: nvs::Storage::cmpMultiPageBlob(unsigned char, char const*, void const*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_storage.cpp:511

0x40113ca3: nvs::Storage::writeItem(unsigned char, nvs::ItemType, char const*, void const*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_storage.cpp:293

0x401141ed: nvs::NVSHandleSimple::set_blob(char const*, void const*, unsigned int) at /esp-idf/components/nvs_flash/src/nvs_handle_simple.cpp:52

0x40112c13: nvs_set_blob at /esp-idf/components/nvs_flash/src/nvs_api.cpp:380

0x4012e4d3: set_blob_nvs at app/components/config_ra/config_ra.c:45 (discriminator 6)

0x4012e68d: cfgSave at app/components/config_ra/config_ra.c:146

0x400fec65: settings_handler at app/components/mqtt_ra/mqtt_ra_cb.c:377 (discriminator 6)

0x400fda85: process_data at app/components/mqtt_ra/mqtt_ra.c:54

0x400fdb8f: mqtt_event_handler at app/components/mqtt_ra/mqtt_ra.c:94 (discriminator 9)

0x401772d2: esp_mqtt_dispatch_event at /esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:812

0x4017770c: deliver_publish at /esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:860 (discriminator 4)

0x401778c6: mqtt_process_receive at /esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1109 (discriminator 6)

0x40177d80: esp_mqtt_task at /esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1273

Core  1 register dump:
PC      : 0x401b251a  PS      : 0x00060f34  A0      : 0x800d1df5  A1      : 0x3ffcfed0
0x401b251a: esp_pm_impl_waiti at /esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x3ffcff9c  A4      : 0x80092224  A5      : 0x3ffbfcd0
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x800d1dba  A9      : 0x3ffcfea0
A10     : 0x00000000  A11     : 0x3ffcff9c  A12     : 0x80082544  A13     : 0x3ffbfcb0
A14     : 0x3ffc97c8  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x401b2517:0x3ffcfed0 0x400d1df2:0x3ffcfef0 0x400904d9:0x3ffcff10
0x401b2517: esp_pm_impl_waiti at /esp-idf/components/esp32/pm_esp32.c:484

0x400d1df2: esp_vApplicationIdleHook at /esp-idf/components/esp_common/src/freertos_hooks.c:63

0x400904d9: prvIdleTask at /esp-idf/components/freertos/tasks.c:3385 (discriminator 1)
```

KaeLL, 2020-06-03:
So I managed to compile the app with the files you provided, but the thing is that I am getting the issue I mentioned earlier today in the previous post to this one way more frequently. I haven't been able to repro. the scan stopping issue at all, with the interrupt WDT triggering all the time.
I also tried the regular idf files, and it's the same thing.
It really seems the constant NVS writes are causing the 2 problems. At least from the testing POV. Should I consider the possibility of the Flash starting to fail?

WCCWCC, 2020-06-05:
@KaeLL 
   * ASSERT_PARAM(-218959118 0), in arch_main.c at line 327 ,Can you reissue this issue, not the same issue. This error log is related to malloc data. When you use coexistence, it is recommended that you turn on the PSRAM configuration option.  
     * component config->ESP32-specific->Support for externa, SPI-connected RAM.
    *  component config-> Bluetooth->Bluedroid Options-> BT/BLE will first malloc the memory...
    *   component config-> Bluetooth->Bluedroid Options->Use dynamic memory allocation..

  *  Can you reproduce the previous problem after replacing the library now?

KaeLL, 2020-06-05:
> When you use coexistence, it is recommended that you turn on the PSRAM configuration option.

Even though I don't have PSRAM installed? That's weird. Nonetheless, I'll enable it and see what happens.

> * Can you reproduce the previous problem after replacing the library now?

Unfortunately no, for 2 reasons. The first is that was able to reproduce this issue only once in the last 3 days, due to this other issue in which the int. WDT triggers and reboots the ESP32. Which leads me to the second reason, that is the fact that unfortunately, even though that would unlock you to work on this issue, I'm tasked with finding a way around these issues first. I'll be testing a few theories today about how to mitigate the effects of these 2 errors now, and after I'm finished, I'll be able to work solely on reproducing this issue and providing you with the info you need, which should be the case next week.

WCCWCC, 2020-06-08:
Hi @KaeLL ,

Even though I don't have PSRAM installed? That's weird. Anyway, I tried that, and the linker complained about IRAM overflow of over 7.3KiB.

To solve this problem, you can turn off the configuration items: 

component config-> Wi-Fi -> WiFi IRAM speed optimization
component config-> Wi-Fi -> WiFi RX IRAM speed optimization

KaeLL, 2020-06-18:
> Hi @KaeLL ,
>
> Even though I don't have PSRAM installed? That's weird. Anyway, I tried that, and the linker complained about IRAM overflow of over 7.3KiB.
>
> To solve this problem, you can turn off the [...]

Resulted in
```
62) psram: This chip is ESP32-D0WD
E (462) cpu_start: Failed to init external RAM!
Re-enable cpu cache.

abort() was called at PC 0x400817dd on core 0
0x400817dd: call_start_cpu0 at esp-idf/components/esp32/cpu_start.c:187 (discriminator 1)

Backtrace:0x4009417a:0x3ffe3ba0 0x40094859:0x3ffe3bc0 0x40098aa9:0x3ffe3be0 0x400817dd:0x3ffe3c50 0x4007921d:0x3ffe3c90 |<-CORRUPTED
0x4009417a: panic_abort at esp-idf/components/esp_system/panic.c:330

0x40094859: esp_system_abort at esp-idf/components/esp_system/system_api.c:100

0x40098aa9: abort at esp-idf/components/newlib/abort.c:46

0x400817dd: call_start_cpu0 at esp-idf/components/esp32/cpu_start.c:187 (discriminator 1)

ELF file SHA256: 97d9353a4fcccc71
```
I'll resume working on this tomorrow, and I'll get back to you with whatever I get from the tests.

Alvin1Zhang, 2020-06-30:
@KaeLL Thanks for reporting, would you please help share if any updates for this issue? Thanks.

KaeLL, 2020-06-30:
I felt like not responding unless I had something for you, which I don't, but ended up switching to something else and forgetting about letting you know.
I ran the tests a day after I said I would, though it was a preliminar test, in which I did not replace the files with the ones you sent me, to make sure I could still repro the issue under normal conditions. Unfortunately, between getting nothing and getting different flavors of the [error I mentioned before](https://github.com/espressif/esp-idf/issues/5358#issuecomment-638389068), I wasn't able to reproduce the issue right away. The uncertainty, lack of equipment at home, and the fact that reducing the NVS writes to a minimum seem to have helped just about enough, which made pressing on this issue a waste of time for now.
So I won't be able to help. You can close this issue, and if I experience it again, I'll reopen, but that's it.

Alvin1Zhang, 2020-07-01:
Many thanks for reporting and sharing the details which help us in debugging the issue, feel free to reopen with any updates. Thanks.

closed 2020-07-01 by Alvin1Zhang