espressif/esp-idf #5359: ble mesh client

chegewara, 2020-05-28:
Hi,
i am playing with ble mesh and i have small problem with sending push button message after reset.
esp32 example flashed:
- ble mesh node on/off client,
- setup to keep persistent provisioning data,
- everything seems to works fine, except i am getting this error code after reboot esp32:
```
E (10632) BLE_MESH: bt_mesh_app_key_get, Failed to get AppKey
E (10632) BLE_MESH: gen_set_state, Failed to send Generic Set message (err -2)
I (10642) EXAMPLE: example_ble_mesh_generic_client_cb: event is 1, error code is -2, opcode is 0x8203
I (10652) EXAMPLE: ESP_BLE_MESH_GENERIC_CLIENT_SET_STATE_EVT
```
It looks like app key is not bonded persistently. I dont see this problem with node server model.
After re-setting app key it is working fine until next reboot.
commit 7d75213674b4572f90c68162ad6fe9b16dae65ad

Alvin1Zhang, 2020-05-29:
Thanks for reporting, we will look into.

Campou, 2020-05-29:
Hi @chegewara 

Thanks for reporting this.

Please help to invoke nvs related functions to store the following global variables in nvs.
```
static uint16_t node_net_idx = ESP_BLE_MESH_KEY_UNUSED;
static uint16_t node_app_idx = ESP_BLE_MESH_KEY_UNUSED;
```
Otherwise after reboot, these two values will be initialized to "0xFFFF". And when using these values for sending a message, the appkey will be failed to find.

Internally we will fix this in the example, and add some error log in the mesh stack.

Thanks.

closed 2020-06-12 by igrr