espressif/esp-idf #5362: cmake: allow to specify project directory in project macro

dshil, 2020-05-29:
**Is your feature request related to a problem? Please describe.**

I have tons of small projects that reuse common cmake script stored in the separate directory.

In the small project I have the following:

```cmake
cmake_minimum_required(VERSION 3.5)

set(ENV{CORE_PATH} ${CMAKE_CURRENT_LIST_DIR}/core)
set(ENV{FW_PATH} ${CMAKE_CURRENT_LIST_DIR})
set(ENV{FW_ID} foo)

# include commonly used script
include($ENV{CORE_PATH}/tools/cmake/project.cmake)
project(foo)
```

In the commonly used cmake script I have the following:

```cmake
# Some pre-configuration
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(project_name)
# Some post-configuration
```

Also each small project defines its own partition table. I use the following in my sdkconfig.defaults stored in project directory:

```plain
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_OFFSET=0x0000
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="projects/bla/partitions.csv"
```

When I tried to compile it with v4.0.1 it fails because build system can't find partitions.csv file, more precisely I see the following:

```plain
 Partition table CSV file
  /foo/bar/baz/tools/cmake/projects/foo/partitions.csv
  not found.  Change custom partition CSV path in menuconfig.
```

As you might see `projects/foo/partitions.csv` is the suffix specified in sdkconfig.defaults.

It works for v3.3.1 but it does not work for v4.0.1 because in v4.0.1 in project.cmake you have the following in `project` macro:

```cmake
    idf_build_process(${IDF_TARGET}
                    SDKCONFIG_DEFAULTS "${sdkconfig_defaults}"
                    SDKCONFIG ${sdkconfig}
                    BUILD_DIR ${build_dir}
                    PROJECT_NAME ${CMAKE_PROJECT_NAME}
                    PROJECT_DIR ${CMAKE_CURRENT_LIST_DIR}
                    PROJECT_VER "${project_ver}"
                    COMPONENTS "${components};${test_components}")
```

After you set `PROJECT_DIR ${CMAKE_CURRENT_LIST_DIR}` the build will fail because `CMAKE_CURRENT_LIST_DIR` is a path of the commonly used script.

**Describe the solution you'd like**

As each cmake macro has optional arguments we can pass the path of project directory to macro itself and use it in `idf_build_process`.

Probably you can suggest another approach.

Alvin1Zhang, 2020-06-01:
@dshil Thanks for raising the feature request, we will evaluate.

renzbagaporo, 2020-06-01:
Is specifying absolute path in `CONFIG_PARTITION_TABLE_CUSTOM_FILENAME` an option? I believe this should be possible. @dshil

dshil, 2020-06-01:
> Is specifying absolute path in `CONFIG_PARTITION_TABLE_CUSTOM_FILENAME` an option? I believe this should be possible. @dshil

Hi @renzbagaporo, yes, it works only when you are only a single developer and store all your repos under pre-difined location. This approach will fail when you share you functionality between other users. As you might see from the description, we (in company) have a lot of small projects (stored in separate repos) that use core components (stored in another repo). Partition tables are also stored within a repo.

dshil, 2020-06-03:
@0xjakob, could you help me somehow (not sure if you are aware of this part of code)?

renzbagaporo, 2020-06-05:
Hi @dshil, sorry for the late reply. I noticed you have a `project` call in your common CMakeLists.txt.

```
# Some pre-configuration
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(project_name)
# Some post-configuration
```

This is not recommended as for the standard project, we override CMake's `project`. If I understand correctly, in effect, you end up calling `idf_build_process` twice: on the common CMakeLists.txt and the small project.

In fact, the assumption for `project.cmake` is that it is going to be at the top-level CMakeLists.txt. If it is viable, the recommended structure is for it to remain that way. For example in the small project:

```
cmake_minimum_required(VERSION 3.5)

include(my_common_pre_configs.cmake)

# include commonly used script
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(foo)

include(my_common_post_configs.cmake)
```

renzbagaporo, 2020-06-08:
Any update @dshil?

dshil, 2020-06-08:
Hi @renzbagaporo, sorry for delay from my side, I will try tomorrow morning and give you feedback asap.

dshil, 2020-06-09:
@renzbagaporo, you were right, thanks for your support. I did the following:

```cmake
cmake_minimum_required(VERSION 3.5)

set(ENV{CORE_PATH} ${CMAKE_CURRENT_LIST_DIR}/core)
set(ENV{FW_PATH} ${CMAKE_CURRENT_LIST_DIR})
set(ENV{FW_ID} foo)

include($ENV{CORE_PATH}/tools/cmake/project_pre_build.cmake)
project(foo)
include($ENV{CORE_PATH}/tools/cmake/project_post_build.cmake)
```

I think we've done here, I will close issue.

closed 2020-06-09 by dshil