espressif/esp-idf #5363: undefined reference to `_Unwind_Resume' (v4.0.1)

dshil, 2020-05-29:
## Environment

- Module or chip used:  ESP32-WROVER-B
- IDF version: v4.0.1
- Build System: idf.py (cmake - the same)
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE: No
- Power Supply: USB

## Debug Logs

```
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(functexcept.o):(.literal._ZSt19__throw_logic_errorPKc+0x8): undefined reference to `_Unwind_Resume'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(functexcept.o): in function `std::__throw_logic_error(char const*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:66: undefined reference to `_Unwind_Resume'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(functexcept.o): in function `std::__throw_length_error(char const*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:78: undefined reference to `_Unwind_Resume'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x0): undefined reference to `_Unwind_GetTextRelBase'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x4): undefined reference to `_Unwind_GetDataRelBase'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x8): undefined reference to `_Unwind_GetRegionStart'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x8): undefined reference to `_Unwind_GetLanguageSpecificData'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0xc): undefined reference to `_Unwind_GetIPInfo'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x10): undefined reference to `_Unwind_SetGR'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x14): undefined reference to `_Unwind_SetIP'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o): in function `base_of_encoded_value(unsigned char, _Unwind_Context*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:114: undefined reference to `_Unwind_GetTextRelBase'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:116: undefined reference to `_Unwind_GetDataRelBase'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:118: undefined reference to `_Unwind_GetRegionStart'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o): in function `parse_lsda_header(_Unwind_Context*, unsigned char const*, lsda_header_info*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:55: undefined reference to `_Unwind_GetRegionStart'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(eh_personality.o): in function `__gxx_personality_v0':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:438: undefined reference to `_Unwind_GetLanguageSpecificData'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:448: undefined reference to `_Unwind_GetIPInfo'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:712: undefined reference to `_Unwind_SetGR'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:714: undefined reference to `_Unwind_SetGR'
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc:716: undefined reference to `_Unwind_SetIP'
[... cut ...]
/home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/dshil/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libstdc++.a(cow-stdexcept.o): in function `std::logic_error::logic_error(char const*)':
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/cow-stdexcept.cc:74: undefined reference to `_Unwind_Resume'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

dshil, 2020-05-29:
I also tried to build an example from IDF_PATH/examples and the build was successful.

0xjakob, 2020-06-01:
Hi @dshil. Did you try `idf.py clean` and `idf.py fullclean` already? Would you have some more information on when this issue occurs? Did you switch to version 4.0 before this issue occurs like indicated in the other posts or did you move a project from c to c++? Thanks for the information!

dshil, 2020-06-01:
Hi @0xjakob and thanks for coming back to my issue.

Yep, I tried to use `clean` and `fullclean` commands but both did not help.

I have the following workflow. We've been using `v3.3.1` for a long time and now we are in switching phase to latest stable release - `v4.0.1`. Some time ago we moved projects from c to c++. It worked well for `v3.3.1` without any issues. We use STL which uses exceptions under the hood. Also we use `new/delete` which may throw `bad_alloc`. 

I found that [exception mechanism is disabled by default](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-compiler-cxx-exceptions). As a result, I would expect that code for exception handling won't be generated but as you might see my build fails on the linker stage with unresolved symbols related to exception handling. It seems strange.

Also, on my host I have the following toolchain:

```
gcc --version
gcc (GCC) 10.1.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cmake version 3.17.3
CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

0xjakob, 2020-06-01:
Hi @dshil, it looks like this is the gcc from your host. Maybe that's related to your problem: If I remember correctly, the toolchain setup changed at v4.0.
Did you follow these steps: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html#step-3-set-up-the-tools?

If yes, please check that `$ xtensa-esp32-elf-gcc --version` outputs something like this:
```
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

A toolchain mix up would explain the linker errors.

If you use `new`, please use `new (std::nothrow)` instead and check the resulting pointer for `nullptr`. 
They way it works is that if a function throws an exception and it reaches code compiled with `-fno-exception`, then it will call `abort()` instead of propagating the exception into that code. Example here: https://github.com/espressif/esp-idf/blob/release/v4.0/components/cxx/test/test_cxx.cpp#L288.
This is also the reason why there shouldn't be any linker errors since libstdc++ is made for switching exceptions off with this mechanism. Besides, the linker errors also show towards the unwind code of libgcc which should be compiled in completely.

Hope this makes sense to you. Feel free to ask if otherwise.

dshil, 2020-06-01:
Hi @0xjakob,

> If yes, please check that $ xtensa-esp32-elf-gcc --version outputs something like this:

Yep, I installed the toolchain using instruction from the provided link.

```
xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

> If you use new, please use new (std::nothrow) instead and check the resulting pointer for nullptr.

Yes, exactly as I did.

> Besides, the linker errors also show towards the unwind code of libgcc which should be compiled in completely.

Could you please clarify, libgcc provided from the installation script?

dshil, 2020-06-01:
Interesting, I tried the same thing on LTS Ubuntu and got the same linker error. Thus, this problem does not relate to Arch Linux.

dshil, 2020-06-01:
@0xjakob The weird thing is that when I set:

```
CONFIG_COMPILER_CXX_EXCEPTIONS=y
```

the linker won't not fail.

Any ideas why it might fail when exceptions are disabled by default or explicitly by:

```
CONFIG_COMPILER_CXX_EXCEPTIONS=n
```

?

igrr, 2020-06-01:
@dshil Looking at the linker error, I would venture a guess that we are missing definitions of 
`std::__throw_logic_error(char const*)` and `std::__throw_length_error(char const*)` in 
https://github.com/espressif/esp-idf/blob/master/components/cxx/cxx_exception_stubs.cpp.

Could you try adding these two definitions? Both can be aliased to `__cxx_fatal_exception_message`.

0xjakob, 2020-06-02:
@dshil yes, the libgcc you need is part of the toolchain. So if installed, then there should be nothing you can do wrong. Please try @igrr's fix.

dshil, 2020-06-02:
@igrr, I added the following at the end of `cxx_exception_stubs.cpp`:

```cpp
void std::__throw_logic_error(const char*) __attribute__((alias("__cxx_fatal_exception_message")));
void std::__throw_length_error(const char*) __attribute__((alias("__cxx_fatal_exception_message")));
```

and as a result got the multiple definition error:

```plain
/builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:78: multiple definition of `std::__throw_length_error(char const*)'; esp-idf/cxx/libcxx.a(cxx_exception_stubs.cpp.obj):/home/dshil/src/esp-idf-v4.0.1/components/cxx/cxx_exception_stubs.cpp:23: first defined here
```

Then I went to `no-rtti/libstdc++.a` to explore which symbols do we have, the following is `output` from `nm`:

```plain
 functexcept.o:
         U __cxa_allocate_exception
         U __cxa_free_exception
         U __cxa_throw
         U __gxx_personality_v0
         U strlen
         U _Unwind_Resume
         U _ZN9__gnu_cxx15__snprintf_liteEPcjPKc13__va_list_tag
         U _ZNSt10bad_typeidD1Ev
         U _ZNSt11logic_errorC1EPKc
         U _ZNSt11logic_errorD1Ev
         U _ZNSt11range_errorC1EPKc
         U _ZNSt11range_errorD1Ev
         U _ZNSt12domain_errorC1EPKc
         U _ZNSt12domain_errorD1Ev
         U _ZNSt12length_errorC1EPKc
         U _ZNSt12length_errorD1Ev
         U _ZNSt12out_of_rangeC1EPKc
         U _ZNSt12out_of_rangeD1Ev
         U _ZNSt13bad_exceptionD1Ev
         U _ZNSt13runtime_errorC1EPKc
         U _ZNSt13runtime_errorD1Ev
         U _ZNSt14overflow_errorC1EPKc
         U _ZNSt14overflow_errorD1Ev
         U _ZNSt15underflow_errorC1EPKc
         U _ZNSt15underflow_errorD1Ev
         U _ZNSt16invalid_argumentC1EPKc
         U _ZNSt16invalid_argumentD1Ev
         U _ZNSt8bad_castD1Ev
         U _ZNSt9bad_allocD1Ev
00000000 T _ZSt16__throw_bad_castv
00000000 T _ZSt17__throw_bad_allocv
00000000 T _ZSt18__throw_bad_typeidv
00000000 T _ZSt19__throw_logic_errorPKc
00000000 T _ZSt19__throw_range_errorPKc
00000000 T _ZSt20__throw_domain_errorPKc
00000000 T _ZSt20__throw_length_errorPKc
00000000 T _ZSt20__throw_out_of_rangePKc
00000000 T _ZSt21__throw_bad_exceptionv
00000000 T _ZSt21__throw_runtime_errorPKc
00000000 T _ZSt22__throw_overflow_errorPKc
00000000 T _ZSt23__throw_underflow_errorPKc
00000000 T _ZSt24__throw_invalid_argumentPKc
00000000 T _ZSt24__throw_out_of_range_fmtPKcz
```

For each `throw` stub I see that we have `T`, thus stub was already defined.

igrr, 2020-06-02:
The purpose of `cxx_exception_stubs.cpp` in IDF is to prevent the linker from linking the respective functions from libstdc++, thereby preventing C++ exceptions support from being pulled into the app when it is disabled in IDF.

The duplicate definition error means that `functexcept.o` was linked into the app, i.e. it was needed to satisfy some otherwise unresolved symbol. Most likely it means that we are still missing some `__throw_xxx` stub from `cxx_exception_stubs.cpp`.

Could you please try the following to narrow down the problem:

1. enable C++ exceptions in menuconfig, and build the app
2. `grep` the .map file for symbols from `functexcept.o`

Then we can compare the list of symbols which were linked from functexcept.o with the list of stubs available in cxx_exception_stubs.cpp, and find the ones which are missing.

dshil, 2020-06-02:
@igrr, @0xjakob I have finally linked it, lets continue discussion in PR?

P.S. thanks for your support.

SERIDJ, 2020-06-25:
hi,

i have probelem with only MQTT library https://github.com/espressif/esp-idf/tr ... ocols/mqtt, It worked for le before (2-3 month ago) and i change nothing ... that show this error :

FAILED: mqtt_ssl_mutual_auth.elf
cmd.exe /C "cd . && C:\Users\mohammed.seridj\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address -nostdlib @CMakeFiles\mqtt_ssl_mutual_auth.elf.rsp -o mqtt_ssl_mutual_auth.elf && cd ."
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x0): undefined reference to `_Unwind_GetTextRelBase'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x4): undefined reference to `_Unwind_GetDataRelBase'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal._ZL21base_of_encoded_valuehP15_Unwind_Context+0x8): undefined reference to `_Unwind_GetRegionStart'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x8): undefined reference to `_Unwind_GetLanguageSpecificData'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0xc): undefined reference to `_Unwind_GetIPInfo'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x10): undefined reference to `_Unwind_SetGR'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__gxx_personality_v0+0x14): undefined reference to `_Unwind_SetIP'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o):(.literal.__cxa_call_unexpected+0x0): undefined reference to `_Unwind_Resume'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/lib/no-rtti\libstdc++.a(eh_personality.o): in function `base_of_encoded_value(unsigned char, _Unwind_Context*)':
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:114: undefined reference to `_Unwind_GetTextRelBase'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:116: undefined reference to `_Unwind_GetDataRelBase'
c:/users/mohammed.seridj/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/../libgcc/unwind-pe.h:118: undefined reference to `_Unwind_GetRegionStart' [... cut ...]
/builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_globals.cc:83: undefined reference to `_Unwind_DeleteException'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
Build complete (27 errors, 0 warnings): C:\Users\mohammed.seridj\Desktop\wrkspse\ssl_mutual_auth\build

i i tried to update framework with : https://docs.espressif.com/projects/esp ... index.html and still show same error, all other library work well

i i tried : idf.py clean, idf.py fullclean too but both did not help.

thanks for ur help

dshil, 2020-06-26:
Hi @SERIDJ, could you try to apply my patch and check if it will help?

P.S. @0xjakob I will try to build a simple example with MQTT library and fresh cmake (3.17) and check if the linkage will fail without my patch.

closed 2021-04-27 by espressif-bot