espressif/esp-idf #5364: [ble mesh] lighting example

chegewara, 2020-05-29:
Is there any change to have ble mesh lighting example or at least some help with building lighting models?

Thanks

Alvin1Zhang, 2020-06-01:
@chegewara Thanks for reporting, we will check.

chegewara, 2020-06-03:
Is there some info about timeline, because i have to admit it is time sensitive to me. It does not have to be full example, just some info how to create model for bluedroid mesh.
Thanks.

Campou, 2020-06-03:
Hi @chegewara 

For lighting client example, which is simple. You can take a reference from the `onoff_client` example. Just define the lighting client model, add it into the element structure, and enable the corresponding lighting client in the menuconfig.

For lighting server example, I have attached an example which I wrote several months ago. You can use this example as a reference.

[ble_mesh_ctl_hsl_server.zip](https://github.com/espressif/esp-idf/files/4722291/ble_mesh_ctl_hsl_server.zip)

Thanks.

chegewara, 2020-06-03:
Thanks you very much.
I have to control brightness, not just on/off. I will try with your example and get back to you.

chegewara, 2020-06-05:
@Campou Some help please. With too many elements on server node i get this:
```
E (1127) BT_OSI: osi_alarm_new alarm_cbs exhausted

E (1127) BLE_MESH: k_delayed_work_init, Unable to create alarm
E (1127) BT_OSI: osi_alarm_new alarm_cbs exhausted

E (1127) BLE_MESH: k_delayed_work_init, Unable to create alarm
```
Then it crash on provisioning. Thanks

chegewara, 2020-06-08:
@Campou I have some questions about client node:
- as i understand client node can also subscribe to group, unicast or virtual address, my question is how i can receive messages from subscribed address; for server node we have `ESP_BLE_MESH_GENERIC_SERVER_STATE_CHANGE_EVT` is there similar event for client node?
- this is not client node related per se, where i can get from message length for all implemented models?
`ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_cli_pub, 2 + 1, ROLE_NODE);` <--- 2nd parameter, if its bluetooth standard and i can find in in bluetoth mesh docs thats fine, just some tip

Thanks
lighting server model code was very helpful, thank you

chegewara, 2020-06-11:
I have solved most problems, so i think i can close this issue for now.

closed 2020-06-11 by chegewara

undervane, 2021-01-20:
@chegewara I'm experiencing similar issue like yours, how did you solve this error?

```
E (1127) BT_OSI: osi_alarm_new alarm_cbs exhausted

E (1127) BLE_MESH: k_delayed_work_init, Unable to create alarm
E (1127) BT_OSI: osi_alarm_new alarm_cbs exhausted

E (1127) BLE_MESH: k_delayed_work_init, Unable to create alarm
```

Moreover, did you find the answer on how to guess the number here?

> this is not client node related per se, where i can get from message length for all implemented models?
> `ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_cli_pub, 2 + 1, ROLE_NODE);` <--- 2nd parameter, if its bluetooth standard and i can find in in bluetoth mesh docs thats fine, just some tip

Thanks in advance!

chegewara, 2021-01-20:
This is closed project and i dont remember now. There is one thing which may be helpful. I had to to increase MTU in esp32 code to 69. Im not sure if that fixed first problem or not, sorry.

I didnt find out answer to second question, sorry.

undervane, 2021-01-22:
Thank you very much @chegewara, unfortunately increasing the MTU didn't solve the issue, but your help was very welcome! 🙂

dhairyashah1, 2023-04-19:
@chegewara @undervane @Campou do you recall what changes did you make to solve this alarm issue?

undervane, 2023-04-21:
> @chegewara @undervane @Campou do you recall what changes did you make to solve this alarm issue?

Hey Dhairya! I was not able to get around it, though it was not critical in my case since I was just learning

dhairyashah1, 2023-04-21:
```
E (1127) BT_OSI: osi_alarm_new alarm_cbs exhausted
```
Incase if anyone still faces the above alarm issue -
Setting the macro [`ALARM_CBS_NUM`](https://github.com/espressif/esp-idf/blob/master/components/bt/common/osi/include/osi/alarm.h#L36) to *80* instead of 50 worked for me.