espressif/esp-idf #5365: FAILED: gatt_server_service_table_demo.elf , project builds on v3.3 but on v4.0.1 doesn't.

0xjavi, 2020-05-29:
## Environment

- Development Kit: ESP-DEVKITV1
- Kit version (for WroverKit/PicoKit/DevKitC): V1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
   C:\msys32\home\Sick\esp\pgpemu\pgpemu-esp32>git describe --tags
fatal: No names found, cannot describe anything.
- Build System:  idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Windows 10
- (Windows only) environment type: Used MSYS32 for prior version of idf, upgraded esp-idf to v4.0.1 and im using cmd.
- Using an IDE?: No
- Power Supply: USB

## Problem Description

When doing "idf.py build" on the project I get a Failed log error "FAILED: gatt_server_service_table_demo.elf" and stops building. Things to note: This project builds successfully on the v3.3 but since there is a bug with the bluetooth connection with multiple devices I needed to upgrade.

### Expected Behavior

Project should build successfully.

### Actual Behavior

Project when building crashes and exits.

### Steps to reproduce

1. Install ESP-IDF v4.0.1
2. git clone https://github.com/yohanes/pgpemu.git
3. Go to pgp-cert.c & pgpemu.c and replace #include "hwcrypto/aes.h" to #include "mbedtls/aes.h"
3. idf.py build

// If possible, attach a picture of your setup/wiring here.
Board is connected to USB to my computer.

## Debug Logs

```
C:\msys32\home\Sick\esp\pgpemu\pgpemu-esp32>idf.py build
Checking Python dependencies...
Python requirements from C:\msys32\home\Sick\esp\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\msys32\home\sick\esp\pgpemu\pgpemu-esp32\build
Executing "ninja all"...
[1/6] cmd.exe /C "cd /D C:\msys32\home\Sick\esp\pgpemu\pgpemu-esp32\build\esp-idf\partition_table && C:\Users\Sick\.e...s\cmake\3.13.4\bin\cmake.exe -E echo *******************************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# Espressif ESP32 Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/6] Performing build step for 'bootloader'
ninja: no work to do.
[3/4] Linking CXX executable gatt_server_service_table_demo.elf
FAILED: gatt_server_service_table_demo.elf
cmd.exe /C "cd . && C:\Users\Sick\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\gatt_server_service_table_demo.elf.rsp  -o gatt_server_service_table_demo.elf  && cd ."
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(pgpemu.c.obj):(.literal.generate_first_challenge+0xc): undefined reference to `generate_chal_0'
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(pgpemu.c.obj):(.literal.handle_protocol+0x38): undefined reference to `generate_next_chal'
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(pgpemu.c.obj):(.literal.handle_protocol+0x3c): undefined reference to `decrypt_next'
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(pgpemu.c.obj):(.literal.handle_protocol+0x40): undefined reference to `generate_reconnect_response'
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(pgpemu.c.obj):(.literal.app_main+0x30): undefined reference to `MAC'
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(pgpemu.c.obj): in function `generate_first_challenge':
c:\msys32\home\sick\esp\pgpemu\pgpemu-esp32\build/../main/pgpemu.c:372: undefined reference to `generate_chal_0'
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(pgpemu.c.obj): in function `handle_protocol':
c:\msys32\home\sick\esp\pgpemu\pgpemu-esp32\build/../main/pgpemu.c:497: undefined reference to `generate_next_chal'
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:\msys32\home\sick\esp\pgpemu\pgpemu-esp32\build/../main/pgpemu.c:520: undefined reference to `decrypt_next'
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:\msys32\home\sick\esp\pgpemu\pgpemu-esp32\build/../main/pgpemu.c:539: undefined reference to `decrypt_next'
c:/users/sick/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:\msys32\home\sick\esp\pgpemu\pgpemu-esp32\build/../main/pgpemu.c:561: undefined reference to `generate_reconnect_response'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\msys32\home\Sick\esp\pgpemu\pgpemu-esp32>
```

## Other items if possible

· elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)

[project_elf_src.c.zip](https://github.com/espressif/esp-idf/files/4703712/project_elf_src.c.zip)

chegewara, 2020-05-29:
Some functions are missing, please check menuconfig if all options you need are enabled.

0xjavi, 2020-05-29:
Im looking at component config inside menuconfig but I don't know what's missing exactly, to be honest.

QweryZ, 2020-05-30:
I assume that you forgot to register your modules in the CMakeList.txt file. Have a look at the new building system in v4.0: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html

The compiler cannot find your functions because the class files aren't registered 
`idf_component_register(SRCS " _.c files_ "
                       INCLUDE_DIRS "."
                       REQUIRES _components_ )
`

0xjavi, 2020-05-30:
After doing what you told me, I started getting errors like some files were missing for the Bluetooth connection, I took them from the esp-idf folder and pasted them inside the main folder of the project and I have exactly the same errors as before.

CMakeList.txt:
`idf_component_register(SRCS "pgpemu.c" INCLUDE_DIRS "." REQUIRES mbedtls nvs_flash)`

QweryZ, 2020-05-30:
The errors appear because you still didn't register all .c files and components. The compiler needs all like in the example

> idf_component_register(SRCS "foo.c" "bar.c" ...
>                        INCLUDE_DIRS "include"...
>                        REQUIRES mbedtls ...)
Also consider adding the bluetooth component to "REQUIRES"

0xjavi, 2020-05-30:
Current CMakeList.txt:
`idf_component_register(SRCS "pgpemu.c" "pgpemu.h" "pgp-cert.h" "pgp-cert.c" "secrets.c" "secrets.h" INCLUDE_DIRS "." REQUIRES mbedtls nvs_flash freertos driver)`

Error log:
`NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/pgp-cert.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\pgp-cert.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/pgp-cert.c.obj   -c ../main/pgp-cert.c
../main/pgp-cert.c: In function 'aes_encrypt':
../main/pgp-cert.c:15:2: error: implicit declaration of function 'esp_aes_crypt_ecb'; did you mean 'mbedtls_aes_crypt_ecb'? [-Werror=implicit-function-declaration]
  esp_aes_crypt_ecb(ctx, ESP_AES_ENCRYPT, inp, out);
  ^~~~~~~~~~~~~~~~~
  mbedtls_aes_crypt_ecb
../main/pgp-cert.c:15:25: error: 'ESP_AES_ENCRYPT' undeclared (first use in this function); did you mean 'MBEDTLS_AES_ENCRYPT'?
  esp_aes_crypt_ecb(ctx, ESP_AES_ENCRYPT, inp, out);
                         ^~~~~~~~~~~~~~~
                         MBEDTLS_AES_ENCRYPT
../main/pgp-cert.c:15:25: note: each undeclared identifier is reported only once for each function it appears in
../main/pgp-cert.c: In function 'aes_setkey':
../main/pgp-cert.c:20:2: error: implicit declaration of function 'esp_aes_init'; did you mean 'esp_restart'? [-Werror=implicit-function-declaration]
  esp_aes_init( ctx );
  ^~~~~~~~~~~~
  esp_restart
../main/pgp-cert.c:21:2: error: implicit declaration of function 'esp_aes_setkey'; did you mean 'aes_setkey'? [-Werror=implicit-function-declaration]
  esp_aes_setkey(ctx, key, 128);
  ^~~~~~~~~~~~~~
  aes_setkey
cc1.exe: some warnings being treated as errors`

When trying to solve these errors I replaced the lines with the ones "did you mean..." and is giving more errors. Maybe I continue missing files and components on the CMakeLists file? Shall I make a pull request and tell the guy to update its code to be compatible for v4.0.1 ?

Thanks for your help, I really appreaciate it.

projectgus, 2020-06-02:
@0xjavi The esp_aes_encrypt/esp_aes_decrypt functions were removed in ESP-IDF V4.0, as this was a major release there were some unavoidable breaking changes.

You can rename all the functions and types with the `esp_aes_xxx` prefix to `mbedtls_aes_xxx`  and that code should compile and work on both ESP-IDF v3.x and ESP-IDF v4.x (with no loss of performance or increase in resource use).

> Shall I make a pull request and tell the guy to update its code to be compatible for v4.0.1 ?

It sounds like it, yes. Major version updates are expected to have some possible code changes needed, there's a [list of breaking changes in the release notes](https://github.com/espressif/esp-idf/releases/v4.0). However it looks like you've nearly found all the necessary changes for this version to work with ESP-IDF v4.0 support, now. Nice work!

I'm going to close this issue as this does not appear to be a bug in ESP-IDF.

closed 2020-06-02 by projectgus