espressif/esp-idf #5366: export.sh assumes bash and it's not POSIX compliant

ferrarimarco, 2020-05-29:
## Environment

- Development Kit:      [none]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): latest master (7d75213674b4572f90c68162ad6fe9b16dae65ad)
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): N/A
- Operating System:     Linux and macOS
- (Windows only) environment type: N/A
- Using an IDE?: No
- Power Supply: USB

## Problem Description

[export.sh](https://github.com/espressif/esp-idf/blob/master/export.sh) has multiple issues:

1. Is not POSIX-compliant.
1. Doesn't declare a hashbang/shebang.
1. Likely assumes bash.

### Expected Behavior

`export.sh` to be POSIX-compliant, so you can source it from a `sh` shell.

### Actual Behavior

```shell
initialize-esp-idf.sh: 3: ./export.sh: Syntax error: "(" unexpected
The command "initialize-esp-idf.sh" failed and exited with 2 during .
```

### Steps to reproduce

1. `/bin/sh`
1. `cd esp-idf`
1. `. ./export.sh`

[Example Travis CI build](https://travis-ci.org/github/ferrarimarco/home-lab/jobs/692653213).

### Code to reproduce this issue

```shell
#!/bin/sh
. ./export.sh
```

## Debug Logs

N/A

## Other items if possible

N/A

Alvin1Zhang, 2020-06-01:
@ferrarimarco Thanks for reporting, we will look into.

igrr, 2020-06-08:
Hi @ferrarimarco, thanks again for reporting this. The attached patch should fix the compatibility issues with `dash`, which is what you are using in the linked CI job, and should also make it shellcheck-clean. Will get this merged into master branch soon, and backport to the previous releases.

Since there is no portable method (that i was able to find) to determine the path of the sourced script in dash, you will need to `export` IDF_PATH before sourcing `export.sh`, most likely here: https://github.com/ferrarimarco/home-lab/blob/ac6d38f45efa859f592e236d50a593513f14cc81/scripts/linux/ci/initialize-esp-idf.sh#L11.

Additional note about your CI script: I don't think sourcing `export.sh` in `initialize-esp-idf.sh` will work, since you are running the script in a new subshell (rather than sourcing it) here: https://github.com/ferrarimarco/home-lab/blob/ac6d38f45efa859f592e236d50a593513f14cc81/.travis.yml#L42. Plus there is `sudo` involved, so the environment variables set by `export.sh` will not reach the build stage of your job. To make it work, i think you can: 1) remove `sudo` — it is not necessary for installing IDF, and 2) source `export.sh` into the shell where the build will run.

[5366.zip](https://github.com/espressif/esp-idf/files/4747539/5366.zip)

ferrarimarco, 2020-06-08:
Yep, correct. I can wait for the fix to be merged in master, but thanks!

closed 2020-06-12 by igrr