espressif/esp-idf #5367: idf.py erase_flash flash monitor does not work

AxelLin, 2020-05-30:
## Environment

- Module or chip used: ESP32-WROOM-32
- IDF version: v4.2-dev-1660-g7d75213674b4
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Using an IDE?: NO
- Power Supply: USB

## Problem Description

I'd like to erase flash and then update firmware images.
But the esptool.py --after hard_reset seems does not work.
I have to press RST button again after erase_flash is done.

This causes problem because I don't want to do manually pressing RST button in my production script for download firmware.

$ idf.py erase_flash flash monitor
Executing action: erase_flash
Choosing default port b'/dev/ttyUSB1' (use '-p PORT' option to set a specific serial port)
Running esptool.py in directory /home/axel/esp/esp-idf/examples/wifi/getting_started/softAP/build
Executing "/home/axel/.espressif/python_env/idf4.2_py3.8_env/bin/python /home/axel/esp/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB1 -b 460800 --before default_reset --after hard_reset --chip esp32 erase_flash"...
esptool.py v3.0-dev
Serial port /dev/ttyUSB1
Connecting........_____....._
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:ae:4d:00
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.1s
Hard resetting via RTS pin...
Executing action: flash
Running ninja in directory /home/axel/esp/esp-idf/examples/wifi/getting_started/softAP/build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[1/2] cd /home/axel/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH="/home/axel/esp/esp-...les/wifi/getting_started/softAP/build" -P /home/axel/esp/esp-idf/components/esptool_py/run_esptool.cmake
esptool.py --chip esp32 -p /dev/ttyUSB1 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 wifi_softAP.bin
esptool.py v3.0-dev
Serial port /dev/ttyUSB1
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
CMake Error at run_cmd.cmake:14 (message):
  esptool.py failed
Call Stack (most recent call first):
  run_esptool.cmake:21 (include)

FAILED: CMakeFiles/flash

### Expected Behavior

Add --after hard_reset for esptool.py should do hard_reset.

### Actual Behavior
Add --after hard_reset for esptool.py does not work.

### Steps to reproduce

Just run idf.py erase_flash flash monitor
Or alternatively run esptool.py directly.
e.g.
esptool.py -p /dev/ttyUSB1 -b 460800 --before default_reset --after hard_reset --chip esp32 erase_flash
esptool.py --chip esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash 0x1000 build/bootloader/bootloader.bin ....
It won't flash firmware after erase_flash is done. i.e. --after hard_reset has no effect at all.

Alvin1Zhang, 2020-06-01:
Thanks for reporting, we will look into.

KonstantinKondrashov, 2020-06-01:
Hi @AxelLin! Could you look at the USB-UART chip, which one is used (CP210x)? 
That command work for me, maybe a hw issue?!
Also, please check the connection RTS and DTR with this scheme https://dl.espressif.com/dl/schematics/esp32_devkitc_v4-sch-20180607a.pdf.

AxelLin, 2020-06-01:
Hi @KonstantinKondrashov 
Thanks for the quick response.
Confirmed it's h/w issue. (sorry for the noise, I should check with our h/w guys first.)

closed 2020-06-01 by AxelLin