espressif/esp-idf #5368: Guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0)

Nepenthes, 2020-05-30:
## Environment
Core (if using chip or module): [ESP-WROVER-B]
IDF version (git rev-parse --short HEAD to get the commit id.): v3.3.1-143d26aa4
Development Env: [Make monitor]
Operating System: [Windows]
Power Supply: [external 5.0]
Compiler version: 5.2.0

## Problem Description
硬件看门狗触发，无法定位到业务代码，出现周期8小时到48小时不等，大概一到两天一次，修改加大CONFIG_ESP_INT_WDT_TIMEOUT_MS基本不起作用

## Debug Logs
dump1：
[image]
[image]
dump2：
[image]
[image]
[image]

##project file
sdkconfig文件以及elf文件
[prjFile.zip](https://github.com/espressif/esp-idf/files/4706169/prjFile.zip)

igrr, 2020-05-30:
This should be fixed on release/v3.3 branch in commit https://github.com/espressif/esp-idf/commit/2718fdbd9578b8cf5f4cc9960dd338d3327f1ca6, could you please try to update to v3.3.2?

Nepenthes, 2020-06-02:
7 machines in continuous testing, has been hung up for 27 hours, still continue to monitor

closed 2020-06-03 by Nepenthes