espressif/esp-idf #5370: IDF not linking, undefined reference to `esp_wifi_internal_issue_disconnect'

merix98, 2020-05-31:
## Environment

- Development Kit: ESP32-DevKitC
- IDF version (run ``git describe --tags`` to find it): v4.2-dev-1660-g7d7521367
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.2.0
- Operating System:     Linux, Ubuntu 18
- Using an IDE?: No

## Problem Description
After IDF master pull, my project is not compiling, due to something related with xtensa compiler, not related to my project

//Detailed problem description goes here.
`/home/lorenzo/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/lorenzo/esp/doModem/build/wpa_supplicant/libwpa_supplicant.a(wpa.o):(.literal.wpa_sm_rx_eapol+0x8): undefined reference to `esp_wifi_internal_issue_disconnect'
/home/lorenzo/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/lorenzo/esp/doModem/build/wpa_supplicant/libwpa_supplicant.a(wpa.o): in function `wpa_sm_rx_eapol':
/home/lorenzo/esp/esp-idf/components/wpa_supplicant/src/rsn_supp/wpa.c:1942: undefined reference to `esp_wifi_internal_issue_disconnect'`
### Expected Behavior
Linking
### Actual Behavior
Not linking, cannot find 'esp_wifi_internal_issue_disconnect'
### Steps to reproduce
Pull master, compile a project

## Debug Logs
Furthermore, when I pull esp-idf master with recurse-submodules, I got this errors, I don't know if they are related to the previous problem:
```
From https://github.com/espressif/esp-idf
 * branch                master     -> FETCH_HEAD
Fetching submodule components/asio/asio
Fetching submodule components/bootloader/subproject/components/micro-ecc/micro-ecc
Fetching submodule components/bt/controller/lib
Fetching submodule components/bt/host/nimble/nimble
Fetching submodule components/cbor/tinycbor
Fetching submodule components/coap/libcoap
Fetching submodule components/coap/libcoap/ext/tinydtls
Fetching submodule components/esp_wifi/lib
Fetching submodule components/esptool_py/esptool
Fetching submodule components/expat/expat
Fetching submodule components/json/cJSON
Fetching submodule components/libsodium/libsodium
Fetching submodule components/lwip/lwip
Fetching submodule components/mbedtls/mbedtls
Fetching submodule components/mqtt/esp-mqtt
Fetching submodule components/nghttp/nghttp2
Fetching submodule components/nghttp/nghttp2/third-party/mruby
Fetching submodule components/nghttp/nghttp2/third-party/neverbleed
Fetching submodule components/protobuf-c/protobuf-c
Fetching submodule components/spiffs/spiffs
Fetching submodule components/tinyusb/tinyusb
Fetching submodule components/unity/unity
Fetching submodule examples/build_system/cmake/import_lib/main/lib/tinyxml2
Already up to date.
fatal: repository '../../espressif/esp-cryptoauthlib.git' does not exist
fatal: clone of '../../espressif/esp-cryptoauthlib.git' into submodule path '/home/lorenzo/esp/esp-idf/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' failed
Failed to clone 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib'. Retry scheduled
fatal: repository '../../espressif/esp-cryptoauthlib.git' does not exist
fatal: clone of '../../espressif/esp-cryptoauthlib.git' into submodule path '/home/lorenzo/esp/esp-idf/examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' failed
Failed to clone 'examples/peripherals/secure_element/atecc608_ecdsa/components/esp-cryptoauthlib' a second time, aborting
```

## Other items if possible

Alvin1Zhang, 2020-06-01:
Thanks for reporting, we will look into.

igrr, 2020-06-01:
For the submodule update error, could you please try:

```
git submodule sync --recursive
```

and then

```
git submodule update --init --recursive
```

merix98, 2020-06-02:
Ok this solved my problem! Thanks for the help

closed 2020-06-02 by merix98

praphullorahi, 2022-03-27:
getting error for espressif 3.5.0 but same program compiled successfully for 3.3.0 , kindly help me to resolve it plz....

Linking .pio\build\bluesky\firmware.elf
c:/users/praphull.kumar/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\factory\src\provisioning.o:(.literal.provisioningStart+0x0): undefined reference to wifi_prov_scheme_ble' c:/users/praphull.kumar/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\factory\src\provisioning.o:(.literal.provisioningStart+0x4): undefined reference to wifi_prov_scheme_ble_event_cb_free_btdm'
c:/users/praphull.kumar/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\factory\src\provisioning.o:(.literal.provisioningStart+0x34): undefined reference to wifi_prov_scheme_ble_set_service_uuid' c:/users/praphull.kumar/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\factory\src\provisioning.o: in function provisioningStart':
D:\project_work\testing3\bluesky-firmware/src/provisioning.c:126: undefined reference to `wifi_prov_scheme_ble_set_service_uuid'

collect2.exe: error: ld returned 1 exit status
*** [.pio\build\factory\firmware.elf] Error 1