espressif/esp-idf #5371: GATTC_Discover GATT_BUSY" issue with esp_ble_gattc_register_for_notify()

joehui, 2020-06-01:
## Environment
Development Kit:    ESP32-Wrover-Kit
IDF version:      v4.0
Build System:   idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
Operating System: MacOS  (This shouldn't matter as I have seen the same behavior even if build on a PC)
Power Supply:  USB

## Problem Description

Randomly gets "GATTC_Discover GATT_BUSY" abort message during esp_ble_gattc_register_for_notify().

### Expected Behavior

Firmware runs smoothly without interruption.

### Actual Behavior

Randomly gets "GATTC_Discover GATT_BUSY" abort message during esp_ble_gattc_register_for_notify().

### Steps to reproduce

In our setup, the ESP32 is the client connect to an nRF52 device that is a BLE server.

During the ESP_GATTC_SEARCH_CMPL_EVT, our code calls:
```
	esp_ble_gattc_get_all_char()
	esp_ble_gattc_register_for_notify()
	esp_ble_gattc_read_char()

```
We use LESC pairing, the esp_ble_gattc_read_char() with ESP_GATT_AUTH_REQ_SIGNED_MITM to force the LESC pairing to happen.

During the ESP_GATTC_REG_FOR_NOTIFY_EVT, we call these functions:
```
	esp_ble_gattc_get_attr_count()
	esp_ble_gattc_get_all_descr()
	esp_ble_gattc_write_char_descr()
```

### Code to reproduce this issue

See above. 

## Debug Logs

During development stage, we have a good number of LOG lines everywhere. Everything works fine. The problem is that when we disabled all the LOG lines, we are starting to get these errors like these:

```
I (66674) GATTC_CB: ##>>> gattc_cb() received EVT: ESP_GATTC_REG_FOR_NOTIFY_EVT(38), gattc if 3
E (66706) BT_GATT: GATTC_Discover GATT_BUSY conn_id = 3
E (66706) BT_APPL: discovery on server failed
E (66706) BT_GATT: GATTC_Read GATT_BUSY conn_id = 3
E (66706) BT_GATT: GATTC_Read GATT_BUSY conn_id = 3
E (66706) BT_GATT: GATTC_Read GATT_BUSY conn_id = 3
```

	

## Other items if possible

This leads us to believe that GATTC read or write calls are conflicting with GATTC_Discover process. Could our issue be related to this old issue:
https://github.com/espressif/esp-idf/issues/888

It seems that adding vTaskDelay(); calls with appropriate timing before esp_ble_gattc_register_for_notify() and esp_ble_gattc_read_char() would temporarily hide the issue. It doesn’t seem like a proper fix though.

I am wondering if you are aware of such behaviour and if there is a better way to address this.

chegewara, 2020-06-01:
Just dont call it from gattc callback. When you call it from callback you are blocking bluedroid.

joehui, 2020-06-01:
Thanks for the speedy response.  I am kind of new to this.  What would be the recommended way to fix what I need to do?

joehui, 2020-06-01:
Also, the things I do in the callbacks are very similar to what the examples do.  I am not sure I understand what I did is different.

joehui, 2020-06-03:
chegewara, can you provide further comments?

Is it the call to esp_ble_gattc_read_char() that caused this error?

chegewara, 2020-06-03:
i believe @Campou can help you more about that

joehui, 2020-06-04:
BTW,  in some other instances, I am getting this:
`
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (btuT) 
Core 0 register dump:
PC      : 0x4000c472  PS      : 0x00060136  A0      : 0x8011bcc1  A1      : 0x3ffda4f0  
A2      : 0x3ffda504  A3      : 0x00000000  A4      : 0x00000268  A5      : 0x3ffda574  
A6      : 0x3ffdafa0  A7      : 0x00000026  A8      : 0x80129691  A9      : 0x3ffdad10  
A10     : 0x3ffe4b70  A11     : 0x3ffb9f00  A12     : 0x3ffdad3a  A13     : 0x00000003  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x0000001e  

ELF file SHA256: 6270db1b369b8659fd5461b110d59375bb1797f6f375f8f49a219391d5f523f7

Backtrace: 0x4000c46f:0x3ffda4f0 |<-CORRUPTED

Entering gdb stub now.
$T1f#ebGNU gdb (crosstool-NG esp-2019r2) 8.1.0.20180627-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-host_apple-darwin12 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /Users/jhui/esp32/smartswitchesp32/build/smartswitchesp32.elf...done.
Remote debugging using /dev/cu.SLAB_USBtoUART
0x4000c472 in ?? ()
(gdb) bt
#0  0x4000c472 in ?? ()
#1  0x4011bcc1 in bta_gattc_enqueue (p_clcb=0x3ffe4b70, p_data=0x3ffb9f00) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_utils.c:461
#2  0x40129638 in bta_gattc_write (p_clcb=0x3ffe4b70, p_data=0x3ffb9f00) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1151
#3  0x4011b779 in bta_gattc_sm_execute (p_clcb=0x3ffe4b70, event=<optimized out>, p_data=0x3ffb9f00)
    at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:292
#4  0x40128913 in bta_gattc_pop_command_to_send (p_clcb=0x3ffe4b70) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1524
#5  0x40129691 in bta_gattc_write (p_clcb=0x3ffe4b70, p_data=<optimized out>) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1175
#6  0x4011b779 in bta_gattc_sm_execute (p_clcb=0x3ffe4b70, event=<optimized out>, p_data=0x3ffb9f00)
    at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:292
#7  0x40128913 in bta_gattc_pop_command_to_send (p_clcb=0x3ffe4b70) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1524
#8  0x40129691 in bta_gattc_write (p_clcb=0x3ffe4b70, p_data=<optimized out>) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1175
#9  0x4011b779 in bta_gattc_sm_execute (p_clcb=0x3ffe4b70, event=<optimized out>, p_data=0x3ffb9f00)
    at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:292
#10 0x40128913 in bta_gattc_pop_command_to_send (p_clcb=0x3ffe4b70) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1524
#11 0x40129691 in bta_gattc_write (p_clcb=0x3ffe4b70, p_data=<optimized out>) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1175
#12 0x4011b779 in bta_gattc_sm_execute (p_clcb=0x3ffe4b70, event=<optimized out>, p_data=0x3ffb9f00)
    at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:292
#13 0x40128913 in bta_gattc_pop_command_to_send (p_clcb=0x3ffe4b70) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1524
#14 0x40129691 in bta_gattc_write (p_clcb=0x3ffe4b70, p_data=<optimized out>) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1175
#15 0x4011b779 in bta_gattc_sm_execute (p_clcb=0x3ffe4b70, event=<optimized out>, p_data=0x3ffb9f00)
    at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:292
#16 0x40128913 in bta_gattc_pop_command_to_send (p_clcb=0x3ffe4b70) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1524
#17 0x40129691 in bta_gattc_write (p_clcb=0x3ffe4b70, p_data=<optimized out>) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1175
#18 0x4011b779 in bta_gattc_sm_execute (p_clcb=0x3ffe4b70, event=<optimized out>, p_data=0x3ffb9f00)
    at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:292
#19 0x40128913 in bta_gattc_pop_command_to_send (p_clcb=0x3ffe4b70) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1524
#20 0x40129691 in bta_gattc_write (p_clcb=0x3ffe4b70, p_data=<optimized out>) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_act.c:1175
#21 0x4011b779 in bta_gattc_sm_execute (p_clcb=0x3ffe4b70, event=<optimized out>, p_data=0x3ffb9f00)
    at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:292
[... cut ...]
    at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:292
---Type <return> to continue, or q <return> to quit---
#42 0x4011b884 in bta_gattc_hdl_event (p_msg=0x3ffba1d4) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_main.c:404
#43 0x4011cb6c in bta_sys_event (param=0x3ffba1d4) at /users/jhui/esp32/esp-idf/components/bt/host/bluedroid/bta/sys/bta_sys_main.c:499
#44 0x40118c72 in osi_thread_run (arg=<optimized out>) at /users/jhui/esp32/esp-idf/components/bt/common/osi/thread.c:68`

joehui, 2020-06-05:
Found a temporary fix.

The crash happens during the ESP_GATTC_REG_FOR_NOTIFY_EVT event when it tries to call esp_ble_gattc_write_char_descr().

I suspect that because of the LESC encryption that , the bluedroid stack is somehow not ready to allow writing of char descr.

The temp fix is simply adding a 2 second delay before calling esp_ble_gattc_write_char_descr().  Just added:

     vTaskDelay(2000 / portTICK_PERIOD_MS);

This is not really proper fix.  But it seems stable after the change.  I hope you guys can provide a better fix.

Weijian-Espressif, 2022-07-18:
@joehui
After the API call, wait until the corresponding callback event is triggered before calling the next API

For example, after calling esp_ble_gattc_register_for_notify(), You need to wait for ESP_GATTC_REG_FOR_NOTIFY_EVT to trigger before you can continue calling the next function esp_ble_gattc_write_char_descr ().

You can refer to our examples gattc_client and gatts_server

closed 2025-03-14 by Alvin1Zhang