espressif/esp-idf #5372: nvs_flash_init error

GarMingLi, 2020-06-01:
I have been using the NVS function before and there was no problem during this period. Then I added the OTA function and the use was normal.Later, I burned other routines without OTA function, and then burned them back to the engineering firmware, and the following error occurred:
```
ESP_ERROR_CHECK failed: esp_err_t 0x110d (ESP_ERR_NVS_NO_FREE_PAGES) at 0x4008d938
0x4008d938: _esp_error_check_failed at /home/ljm/esp32/esp-idf/components/esp32/panic.c:739

file: "../main/main.c" line 64
func: app_user_init
expression: nvs_flash_init()

ELF file SHA256: 37df94134d59899ce6ccc5ea61285a9840503240f00fcdcbe6ac16626577db4e

Backtrace: 0x4008d2c0:0x3ffbb580 0x4008d93b:0x3ffbb5a0 0x400d5944:0x3ffbb5c0 0x400d5a39:0x3ffbb5e0 0x400d27d3:0x3ffbb600
0x4008d2c0: invoke_abort at /home/ljm/esp32/esp-idf/components/esp32/panic.c:159

0x4008d93b: _esp_error_check_failed at /home/ljm/esp32/esp-idf/components/esp32/panic.c:740

0x400d5944: app_user_init at /home/ljm/esp32/code/AEGIS_HUB/build/../main/main.c:65 (discriminator 1)

0x400d5a39: app_main at /home/ljm/esp32/code/AEGIS_HUB/build/../main/main.c:104

0x400d27d3: main_task at /home/ljm/esp32/esp-idf/components/esp32/cpu_start.c:564

Rebooting...

```

## Environment

- Module or chip used:  ESP32S
- IDF version:  v4.1-dev-2079-g5dbabae9d
- Build System:      idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Power Supply:    external 3.3V

igrr, 2020-06-01:
OTA- And non-OTA partition tables have different sizes of NVS partitions, so if you switch between the two, you may need to erase NVS contents. The recommended way to handle this in code is shown in the example:
https://github.com/espressif/esp-idf/blob/7d75213674b4572f90c68162ad6fe9b16dae65ad/examples/storage/nvs_rw_blob/main/nvs_blob_example_main.c#L156

GarMingLi, 2020-06-01:
Thank you.The Problem was solved.

closed 2020-06-01 by GarMingLi