espressif/esp-idf #5374: MQTT SSL Muthual auth. cannot connect

Nomuc, 2020-06-01:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (DevKitC): [v3]
- Module or chip used:  [ESP32-WROOM-32D]
- IDF version : v3.3
- Build System: [idf.py]
- Compiler version: v2.50200.80
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [Yes (PlatformIO)]
- Power Supply: [USB]

## Problem

Hello,

I got to create a project using MQTT with SSL Muthual auth., so I used the example project from ESP-IDF, it is going perfectly everytime I try to connect to my broker with ECDSA key type and SHA-256.

Then I try to run it on my project with the same sample code to connect and I got a mbed_ssl handshake problem (-0x7280) everytime :

> esp-tls : mbed_ssl_handshake returned -0x7280
> esp-tls : Certificate verified.
> esp-tls : Failed to open a new connection.
> TRANS_SSL : Failed to open a new connection.
> MQTT_CLIENT : Error transport connect.

I have checked the error by it means nothing to me, it is called CONN_EOF error.

I added debug log level 4 on TLS and I found this :

> (17213) mbedtls: components\mbedtls\mbedtls\library\ssl_tls.c:2779 **ssl->f_send**() returned -80 (-0x0050)
> 
> (17228) mbedtls: components\mbedtls\mbedtls\library\ssl_tls.c:3472 mbedtls_ssl_flush_output() returned -80 (-0x0050)
> 
> (17245) mbedtls: components\mbedtls\mbedtls\library\ssl_tls.c:3315 ssl_write_record() returned -80 (-0x0050)
> 
> (17262) mbedtls: components\mbedtls\mbedtls\library\ssl_cli.c:3379 mbedtls_ssl_write_handshake_msg() returned -80 (-0x0050)
> 
> (17280) mbedtls: components\mbedtls\mbedtls\library\ssl_tls.c:8096 <= handshake
> 
> (17294) esp-tls: mbedtls_ssl_handshake returned -0x50
> (17299) esp-tls: Certificate verified.
> (17308) esp-tls: Failed to open new connection
> (17309) TRANS_SSL: Failed to open a new connection
> (17315) MQTT_CLIENT: Error transport connect

Everytime ssl->f_send is returning -0x0050 or -0x7280 and I do not understand why.

I have copied paste the sdkconfig files into my working project and then it works 2 times out of 10.

Alvin1Zhang, 2020-06-01:
Thanks for reporting, we will look into.  @david-cermak

david-cermak, 2020-06-01:
Hi @Nomuc 

It seems like the server actively closes the connection during handshake (`MBEDTLS_ERR_SSL_CONN_EOF` simply says the connection is closed, so what I think is happening is that the client side tries to write -- continue with the handshake -- but the socket is already closed from the other side, so network write returns `-0x50` and the handshake then terminates `CONN_EOF`)

Are you able to retrieve some logs from the broker? This info should indicate why the connection was closed as apparently from client side the termination of the connection looks unexpected. The other option would be to check the packet capture in wireshark to see which phase of the handshake caused issues (should be visible from mbedtls logs, as well, but would need complete log file).

> I have copied paste the sdkconfig files into my working project and then it works 2 times out of 10.

This is strange, as the mqtt mutual authentication example doesn't define any specific configs, just goes with defaults (maybe if you switch between different IDF versions, there might be some differences, but typically just buffer sizes, etc)... Is it possible to diff the previous sdkconfig just to see where the difference lies?

Nomuc, 2020-06-01:
Hello @david-cermak,

When you say the server is closing before the end of the handshake it is weird because I can connect with the same cert on the mqtt example, I also tried to connect to my broker with mosquitto and it did works.

The logs of debug from the broker are empty, only refreshing ticks...

Find the mbedtls logs below : 

> (6981) MQTT: Before connect event n:7
> (7236) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:8086
> => handshake
> 
> (7240) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_cli.c:3510
> client state: 0
> 
> (7254) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2755
> => flush output
> 
> (7268) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2767
> <= flush output
> 
> (7282) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_cli.c:3510
> client state: 1
> 
> (7297) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2755
> => flush output
> 
> (7311) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2767
> <= flush output
> 
> (7325) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_cli.c:774 => write client hello
> 
> (7345) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:3184
> => write handshake message
> 
> (7357) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:3343
> => write record
> 
> (7374) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2755
> => flush output
> 
> (7383) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2774
> message length: 220, out_left: 220
> 
> (7402) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2779
> ssl->f_send() returned 220 (-0xffffff24)
> 
> (7416) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2807
> <= flush output
> 
> (7430) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:3476
> <= write record
> 
> (7444) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:3320
> <= write handshake message
> 
> (7459) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_cli.c:1106
> <= write client hello
> 
> (7474) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_cli.c:3510
> client state: 2
> 
> (7488) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2755
> => flush output
> 
> (7503) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_tls.c:2767
> <= flush output
> 
> (7517) mbedtls: C:\Users\abu\.platformio\packages\framework-espidf@3.30300.190916\components\mbedtls\mbedtls\library\ssl_cli.c:1499
> => parse server hello
[... 272 lines cut ...]
Define in addition from my project : 
> CONFIG_ESPTOOLPY_BAUD,
> CONFIG_ESPTOOLPY_BAUD_115200B,
> CONFIG_PYTHON,
> CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1,
> CONFIG_ESPTOOLPY_COMPRESSED,
> CONFIG_PARTITION_TABLE_SINGLE_APP,
> CONFIG_ADC_CAL_EFUSE_TP_ENABLE,
> CONFIG_ADC_CAL_EFUSE_VREF_ENABLE,
> CONFIG_ESPTOOLPY_PORT

Define in addition coming from mqtt mutual auth example : 
> CONFIG_IDF_CMAKE,
> CONFIG_IDF_FIRMWARE_CHIP_ID,
> CONFIG_PARTITION_TABLE_CUSTOM,
> CONFIG_BTDM_CTRL_BR_EDR_SCO_DATA_PATH_EFF,
> CONFIG_BTDM_CONTROLLER_MODEM_SLEEP,
> CONFIG_BTDM_MODEM_SLEEP_MODE_ORIG,
> CONFIG_BTDM_LPCLK_SEL_MAIN_XTAL,
> CONFIG_ESP32_TIME_SYSCALL_USE_RTC,
> CONFIG_ESP32_DPORT_DIS_INTERRUPT_LVL, 
> CONFIG_EMAC_L2_TO_L3_RX_BUF_MODE, 
> CONFIG_LWIP_SNTP_UPDATE_DELAY,

PS : I just tried again with new certificates on your test project, I got a -0x50 error.

david-cermak, 2020-06-01:
@Nomuc Thanks for sharing the logs!

> When you say the server is closing before the end 

What I meant was (an assumption) that the server closed the connection, because it didn't like "something" within the handshake, and to find out what that something was, we could check the server log or the packet capture.

From what I can see from your log, it fails in the state nr. 8, which is `MBEDTLS_SSL_CLIENT_KEY_EXCHANGE`, which is just after the `MBEDTLS_SSL_CLIENT_CERTIFICATE` so we could assume that this something is somewhere close to the client certificate.

> The logs of debug from the broker are empty, only refreshing ticks...

What kind of broker you use? It is mosquitto? It should be possible to make the logging more verbose, at least to see the incoming TCP connection.

> but nothing feels related to MQTT,

This issue is not related to MQTT, as the tls handshake fails before it gets to employ the MQTT library at all, but TBH I don't see any related config options to mbedTLS, nor networking. So I think that was just a coincidence that the connection went from none to intermittent after the config update.

Nomuc, 2020-06-01:
Thanks for your help @david-cermak 

As broker we use VerneMQ ( https://docs.vernemq.com/configuration/logging ), even in debug mode there are no interesting logs ... 

I told me the same but that was the only interesting way I had.

Nomuc, 2020-06-01:
We are connecting on mosquitto in this following way : 
`mosquitto_sub -h <dns> -p <port> --cafile CA.crt -t "<topic>" -q 0 -i <clientid> -P <password> -u <username> -d --cert Middle/devices/PRIME256/cert.crt --key Middle/devices/PRIME256/key.pem --insecure`
--insecure is used because the broker cert CN is not the actual broker DNS, so we had to add this code : 
[image]

We instantly return before check CN to avoid problems.

Can this cause unwanted side effects ?

almini, 2020-08-11:
We have the same problem and also use a VerneMQ broker but without a client certificate. 
Could you somehow resolve the issue?

Nomuc, 2020-08-12:
Hi @almini ,

I managed to solve the problem recently, the problem for me was the clock frequency. 

I used to be at 80MHz, I saw that going to 160MHz allowed me to connect to the broker. 
I did some tests (shutting down WiFi) and at 160MHz the product couldn't reconnect to the broker, but at 240MHz it is working everytime at the first try. I think VerneMQ is sending an EOF after 5 seconds of connection try, and ESP32 is too slow under 240MHz ... 

@david-cermak any updates on this ?

david-cermak, 2020-08-17:
@Nomuc Thanks for sharing your update and sorry for not responding. As mentioned in my initial post the server seemed to close the connection due to some server side timeout.

I'm sorry I really wasn't able to see the root cause from the logs nor from the symptoms described here. Moreover I'm not familiar with VerneMQ broker and its config options (I should have asked for the packet capture in the first place to clearly see the connection gets really closed from the server side)

As for the CN check, that should not cause any related issue (and in case of problems we would see the relevant mbedtls logs). Please note that you can achieve the same without modification of mbedtls sources, please see how the `skip_common_name` option is used in `esp-tls` component here https://github.com/espressif/esp-idf/blob/master/components/esp-tls/esp_tls_mbedtls.c#L383 (if you use mbedtls API directly or just set this to `true` if using esp-tls API)

almini, 2020-09-29:
@Nomuc Thanks for the hint. 
In newer versions of VerneMQ it is possible to configure the timeout ([https://github.com/vernemq/vernemq/pull/1527](https://github.com/vernemq/vernemq/pull/1527)). 
Increasing the timeout solved our issues.

Nomuc, 2020-09-29:
Thanks for your answer @almini, I will also try that on my side. 

I will close the issue since the problem is solved for both of us.

closed 2020-09-29 by Nomuc