espressif/esp-idf #5376: Using SPI while cache disabled

ErwinTheRabbit, 2020-06-01:
If you use spi_device_queue_trans in ISR while cache disabled it will panic:
```
Re-enable cpu cache.
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed).
```
although SPI_MASTER_IN_IRAM and SPI_MASTER_ISR_IN_IRAM have been set.

To fix this problem you need to replace `static void spi_bus_intr_enable(void *host)` and `static void spi_bus_intr_disable(void *host)` with `static void SPI_MASTER_ISR_ATTR spi_bus_intr_enable(void *host)` and `static void SPI_MASTER_ISR_ATTR spi_bus_intr_disable(void *host)` in "esp-idf/components/driver/spi_master.c".

I think it's necessary to fix it, because with SPI_MASTER_IN_IRAM it make sense.

oachkatzalschwoaf, 2021-08-02:
struggled a looong time with thie same issue
=> solution: add SPI_MASTER_ISR_ATTR to both function

t-fi, 2024-10-12:
Looks like this is solved in the latest ESP IDF version

closed 2025-03-14 by Alvin1Zhang