espressif/esp-idf #5377: Populating 2D Array Causes cJSON not to be read properly

jonhunterking66, 2020-06-01:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
v4.1-dev-2140-g2976f5dae
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Linux
- Using an IDE?: VS Code 
- Power Supply:         USB

## Problem Description

Really strange issue here. I am trying to parse through a cJSON to populate several 2D arrays. I got my code working for two, 2D arrays but when I try and populate a third with a new object from the cjson my code no longer enters into the loop where it must perform this function. Very strange issue that I cannot figure out. 
### Expected Behavior

New 2D array is populated as expected. 

### Actual Behavior

Trying to assign third 2D array a value causes program to not enter if statement on cJSON_HasObjectItem

### Code to reproduce this issue

This is just the relevant code. 

This code works: 
```cpp
    if (cJSON_HasObjectItem(fields, "thetaPathAdc"))
    {
      ESP_LOGI(TAG, "Recieving zone data");
      //Extracting shouldWater bool
      cJSON *shouldWaterCJSON = cJSON_GetObjectItem(fields, "shouldWater");
      cJSON *booleanValue = cJSON_GetObjectItem(shouldWaterCJSON, "booleanValue");
      //Extracting rPath array
      cJSON *rPath = cJSON_GetObjectItem(fields, "rPathAdc");
      cJSON *arrayValueRadius = cJSON_GetObjectItem(rPath, "arrayValue");
      cJSON *valuesRadius = cJSON_GetObjectItem(arrayValueRadius, "values");
      cJSON *quotesElementRadius;
      //Extracting thetaPath array
      cJSON *thetaPath = cJSON_GetObjectItem(fields, "thetaPathAdc");
      cJSON *arrayValueTheta = cJSON_GetObjectItem(thetaPath, "arrayValue");
      cJSON *valuesTheta = cJSON_GetObjectItem(arrayValueTheta, "values");
      cJSON *quotesElementTheta;
      //Extracting passes array
      cJSON *passes = cJSON_GetObjectItem(fields, "passCount");
      cJSON *arrayValuePasses = cJSON_GetObjectItem(passes, "arrayValue");
      cJSON *valuesPasses = cJSON_GetObjectItem(arrayValuePasses, "values");
      cJSON *quotesPasses;
      if (booleanValue->valueint == 1)
      {
        shouldWaterArray[currZone] = 1;
        int length = 0;
        cJSON_ArrayForEach(quotesElementRadius, valuesRadius) //Gather R
        {
          cJSON *intValueRadius = cJSON_GetObjectItem(quotesElementRadius, "integerValue");
          rArray[currZone][length] = atoi(intValueRadius->valuestring);
          length++;
        }
        pathSizeArray[currZone] = length;
        length = 0;
        cJSON_ArrayForEach(quotesElementTheta, valuesTheta)
        {
          cJSON *intValueTheta = cJSON_GetObjectItem(quotesElementTheta, "integerValue");
          thetaArray[currZone][length] = atoi(intValueTheta->valuestring);
          length++;
        }
        length = 0;
        cJSON_ArrayForEach(quotesPasses, valuesPasses)
        {
          cJSON *intValuePasses = cJSON_GetObjectItem(quotesPasses, "integerValue");
          printf("in passes loop: %d\n", length);
          //passCountArray[currZone][length] = atoi(intValuePasses->valuestring);
          length++;
        }
      }
      else
      {
        shouldWaterArray[currZone] = 0;
      }
      currZone++;
```

Un-commenting passCountArray causes the program to not enter this if statement:
```cpp
    if (cJSON_HasObjectItem(fields, "thetaPathAdc"))
    {
      ESP_LOGI(TAG, "Recieving zone data");
      //Extracting shouldWater bool
      cJSON *shouldWaterCJSON = cJSON_GetObjectItem(fields, "shouldWater");
      cJSON *booleanValue = cJSON_GetObjectItem(shouldWaterCJSON, "booleanValue");
      //Extracting rPath array
      cJSON *rPath = cJSON_GetObjectItem(fields, "rPathAdc");
      cJSON *arrayValueRadius = cJSON_GetObjectItem(rPath, "arrayValue");
      cJSON *valuesRadius = cJSON_GetObjectItem(arrayValueRadius, "values");
      cJSON *quotesElementRadius;
      //Extracting thetaPath array
      cJSON *thetaPath = cJSON_GetObjectItem(fields, "thetaPathAdc");
      cJSON *arrayValueTheta = cJSON_GetObjectItem(thetaPath, "arrayValue");
      cJSON *valuesTheta = cJSON_GetObjectItem(arrayValueTheta, "values");
      cJSON *quotesElementTheta;
      //Extracting passes array
      cJSON *passes = cJSON_GetObjectItem(fields, "passCount");
      cJSON *arrayValuePasses = cJSON_GetObjectItem(passes, "arrayValue");
      cJSON *valuesPasses = cJSON_GetObjectItem(arrayValuePasses, "values");
      cJSON *quotesPasses;
      if (booleanValue->valueint == 1)
      {
        shouldWaterArray[currZone] = 1;
        int length = 0;
        cJSON_ArrayForEach(quotesElementRadius, valuesRadius) //Gather R
        {
          cJSON *intValueRadius = cJSON_GetObjectItem(quotesElementRadius, "integerValue");
          rArray[currZone][length] = atoi(intValueRadius->valuestring);
          length++;
        }
        pathSizeArray[currZone] = length;
        length = 0;
        cJSON_ArrayForEach(quotesElementTheta, valuesTheta)
        {
          cJSON *intValueTheta = cJSON_GetObjectItem(quotesElementTheta, "integerValue");
          thetaArray[currZone][length] = atoi(intValueTheta->valuestring);
          length++;
        }
        length = 0;
        cJSON_ArrayForEach(quotesPasses, valuesPasses)
        {
          cJSON *intValuePasses = cJSON_GetObjectItem(quotesPasses, "integerValue");
          printf("in passes loop: %d\n", length);
          passCountArray[currZone][length] = atoi(intValuePasses->valuestring);
          length++;
        }
      }
      else
      {
        shouldWaterArray[currZone] = 0;
      }
      currZone++;
```

Here is the definition of these arrays:
```cpp
int shouldWaterArray[MAXZONECOUNT];
int passCountArray[MAXZONECOUNT][MAXPATHSIZE] = {0};
int dutyCycleArray[MAXZONECOUNT][MAXPATHSIZE] = {0};
int rArray[MAXZONECOUNT][MAXPATHSIZE] = {0};
int thetaArray[MAXZONECOUNT][MAXPATHSIZE] = {0};
int pathSizeArray[MAXZONECOUNT] = {0};
```

This code also works if I overwrite thetaArray in place of passCountArray.

Alvin1Zhang, 2020-06-02:
@jonhunterking66 Thanks for the detailed report, we will look into.

jonhunterking66, 2020-06-02:
@Alvin1Zhang I think I found the culprit. If I decreased maxzonecount from 10 to 5, or decreased maxpathsize from 200 to 100 I was able to execute as expected. I am guessing this is because I am using too much memory to access such large arrays. Can you confirm this is the case? If I make those variables static would it solve my problem?

mahavirj, 2020-06-03:
@jonhunterking66 Did you check stack utilization of task from which you are doing these operations? API `uxTaskGetStackHighWaterMark()` can help to get high water mark.

jonhunterking66, 2020-06-03:
I will try that. Is my stack to small do you think?

mahavirj, 2020-06-09:
@jonhunterking66 Yes, I had that suspicion. Maybe you can provide entire sample application along with your `sdkconfig`.

jonhunterking66, 2020-06-17:
Solution was to dynamically allocate memory for each 2D array

closed 2020-06-17 by jonhunterking66