espressif/esp-idf #5378: [S2] enable SPIRAM_USE_AHB_DBUS3

chegewara, 2020-06-02:
When i try to enable `SPIRAM_USE_AHB_DBUS3` i have this crash:
```
In function 'esp_spiram_init_cache':
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:100:49: error: 'BUS_AHB_DBUS3_CACHE_SIZE' undeclared (first use in this function); did you mean 'BUS_DRAM1_CACHE_SIZE'?
 #define DBUS3_ONLY_CACHE_SIZE                   BUS_AHB_DBUS3_CACHE_SIZE
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:101:81: note: in expansion of macro 'DBUS3_ONLY_CACHE_SIZE'
 #define DRAM0_DRAM1_DPORT_DBUS3_CACHE_SIZE      (DRAM0_DRAM1_DPORT_CACHE_SIZE + DBUS3_ONLY_CACHE_SIZE)
                                                                                 ^~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:140:28: note: in expansion of macro 'DRAM0_DRAM1_DPORT_DBUS3_CACHE_SIZE'
         if (SPIRAM_SIZE <= DRAM0_DRAM1_DPORT_DBUS3_CACHE_SIZE) {
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:100:49: note: each undeclared identifier is reported only once for each function it appears in
 #define DBUS3_ONLY_CACHE_SIZE                   BUS_AHB_DBUS3_CACHE_SIZE
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:101:81: note: in expansion of macro 'DBUS3_ONLY_CACHE_SIZE'
 #define DRAM0_DRAM1_DPORT_DBUS3_CACHE_SIZE      (DRAM0_DRAM1_DPORT_CACHE_SIZE + DBUS3_ONLY_CACHE_SIZE)
                                                                                 ^~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:140:28: note: in expansion of macro 'DRAM0_DRAM1_DPORT_DBUS3_CACHE_SIZE'
         if (SPIRAM_SIZE <= DRAM0_DRAM1_DPORT_DBUS3_CACHE_SIZE) {
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:110:50: error: 'AHB_DBUS3_ADDRESS_HIGH' undeclared (first use in this function); did you mean 'DPORT_ADDRESS_HIGH'?
 #define SPIRAM_MID_SIZE_MAP_VADDR               (AHB_DBUS3_ADDRESS_HIGH - SPIRAM_SIZE_EXC_DRAM0_DRAM1_DPORT)
                                                  ^~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:142:51: note: in expansion of macro 'SPIRAM_MID_SIZE_MAP_VADDR'
             Cache_Dbus_MMU_Set(MMU_ACCESS_SPIRAM, SPIRAM_MID_SIZE_MAP_VADDR, SPIRAM_MID_SIZE_MAP_PADDR, 64, SPIRAM_MID_SIZE_MAP_SIZE >> 16, 0);
                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:114:49: error: 'AHB_DBUS3_ADDRESS_LOW' undeclared (first use in this function); did you mean 'DROM0_ADDRESS_LOW'?
 #define SPIRAM_BIG_SIZE_MAP_VADDR               AHB_DBUS3_ADDRESS_LOW
                                                 ^~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:145:51: note: in expansion of macro 'SPIRAM_BIG_SIZE_MAP_VADDR'
             Cache_Dbus_MMU_Set(MMU_ACCESS_SPIRAM, SPIRAM_BIG_SIZE_MAP_VADDR, SPIRAM_BIG_SIZE_MAP_PADDR, 64, SPIRAM_BIG_SIZE_MAP_SIZE >> 16, 0);
                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/chegewara/demos/esp32s2/forks/esp-idf/components/soc/soc/esp32s2/include/soc/spi_mem_reg.h:21,
                 from /home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram_psram.h:18,
                 from /home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:27:
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:148:143: error: 'EXTMEM_PRO_DCACHE_MASK_BUS3' undeclared (first use in this function); did you mean 'EXTMEM_PRO_DCACHE_MASK_BUS0'?
         REG_CLR_BIT(EXTMEM_PRO_DCACHE_CTRL1_REG, EXTMEM_PRO_DCACHE_MASK_DRAM1 | EXTMEM_PRO_DCACHE_MASK_DRAM0 | EXTMEM_PRO_DCACHE_MASK_DPORT | EXTMEM_PRO_DCACHE_MASK_BUS3);
                                                                                                                                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/soc/soc/esp32s2/include/soc/soc.h:143:45: note: in definition of macro 'REG_CLR_BIT'
             (*(volatile uint32_t*)(_r) &= ~(_b));                                                                      \
                                             ^~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c: In function 'esp_spiram_add_to_heapalloc':
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:100:49: error: 'BUS_AHB_DBUS3_CACHE_SIZE' undeclared (first use in this function); did you mean 'BUS_DRAM1_CACHE_SIZE'?
 #define DBUS3_ONLY_CACHE_SIZE                   BUS_AHB_DBUS3_CACHE_SIZE
                                                 ^~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:101:81: note: in expansion of macro 'DBUS3_ONLY_CACHE_SIZE'
 #define DRAM0_DRAM1_DPORT_DBUS3_CACHE_SIZE      (DRAM0_DRAM1_DPORT_CACHE_SIZE + DBUS3_ONLY_CACHE_SIZE)
                                                                                 ^~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:255:28: note: in expansion of macro 'DRAM0_DRAM1_DPORT_DBUS3_CACHE_SIZE'
         if (SPIRAM_SIZE <= DRAM0_DRAM1_DPORT_DBUS3_CACHE_SIZE) {
                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:110:50: error: 'AHB_DBUS3_ADDRESS_HIGH' undeclared (first use in this function); did you mean 'DPORT_ADDRESS_HIGH'?
 #define SPIRAM_MID_SIZE_MAP_VADDR               (AHB_DBUS3_ADDRESS_HIGH - SPIRAM_SIZE_EXC_DRAM0_DRAM1_DPORT)
                                                  ^~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:258:64: note: in expansion of macro 'SPIRAM_MID_SIZE_MAP_VADDR'
                 esp_err_t err = heap_caps_add_region((intptr_t)SPIRAM_MID_SIZE_MAP_VADDR + size_for_flash, (intptr_t)SPIRAM_MID_SIZE_MAP_VADDR + SPIRAM_MID_SIZE_MAP_SIZE -1);
                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:114:49: error: 'AHB_DBUS3_ADDRESS_LOW' undeclared (first use in this function); did you mean 'DROM0_ADDRESS_LOW'?
 #define SPIRAM_BIG_SIZE_MAP_VADDR               AHB_DBUS3_ADDRESS_LOW
                                                 ^~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:268:64: note: in expansion of macro 'SPIRAM_BIG_SIZE_MAP_VADDR'
                 esp_err_t err = heap_caps_add_region((intptr_t)SPIRAM_BIG_SIZE_MAP_VADDR, (intptr_t)SPIRAM_BIG_SIZE_MAP_VADDR + SPIRAM_BIG_SIZE_MAP_SIZE -1);
                                                                ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/chegewara/demos/esp32s2/forks/esp-idf/components/esp32s2/spiram.c:293:1: error: control reaches end of non-void function [-Werror=return-type]
 }
```

idf version: unknown, arduino as component with esp32s2 branch of arduino

```
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x540
load:0x40050000,len:0x1620
load:0x40054000,len:0x1a04
entry 0x40050280
I (92) cache: Instruction cache         : size 8KB, 4Ways, cache line size 32Byte
I (92) cache: Data cache                : size 8KB, 4Ways, cache line size 32Byte
I (93) spiram: Found 16MBit SPI RAM device
I (97) spiram: SPI RAM mode: sram 80m
I (100) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (107) cpu_start: Pro cpu up.
I (109) cpu_start: Application information:
I (113) cpu_start: Project name:     hello-world
I (118) cpu_start: App version:      2448837-dirty
I (122) cpu_start: Compile time:     Jun  2 2020 08:29:13
I (127) cpu_start: ELF file SHA256:  d7e5eae3d9d191dc...
I (132) cpu_start: ESP-IDF:          v4.2-dev-1669-g541430f6d-dirty
I (138) cpu_start: Single core mode
I (360) spiram: SPI SRAM memory test OK
I (360) spiram: Instructions copied and mapped to SPIRAM
I (374) spiram: Read only data copied and mapped to SPIRAM
I (381) heap_init: Initializing. RAM available for dynamic allocation:
I (382) heap_init: At 3FFC1888 len 0003A778 (233 KiB): DRAM
I (382) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (387) cpu_start: Pro cpu start user code
I (391) spiram: Adding pool of 1856K of external SPI memory to heap allocator
I (398) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (459) spi_flash: detected chip: generic
I (460) spi_flash: flash io: qio
I (460) cpu_start: Starting scheduler on PRO CPU.
```

chegewara, 2020-06-02:
Ok, i find this:
```
CONFIG_SPIRAM_USE_AHB_DBUS3// TODO Ready to remove this macro esp32s2 no AHB bus access cache
```
https://github.com/espressif/esp-idf/blob/master/components/esp32s2/spiram.c#L139

Alvin1Zhang, 2020-08-04:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/64673b45e5b4f9f25e1133fa33f4779cd683269e, feel free to reopen if the issue still happens.

closed 2020-08-04 by Alvin1Zhang