espressif/esp-idf #5379: ESP32_S2 light sleep enable reboot

waynepeacock, 2020-06-02:
## Environment

- Development Kit:      [ESP32-S2-Saola-1]
- Kit version: [v1.2]
- Module or chip used:  [ESP32-S2-WROOM]
- IDF version (run ``git describe --tags`` to find it):  v4.2-dev-1660-g7d7521367
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

When running the example 'esp-idf/examples/wifi/power_save' the ESP32_S2 will boot as normal, connect to a Wi-Fi AP successfully and then reboot moments after (Less than 2 seconds) with a reset cause rst:0x10 (RTCWDT_RTC_RST). This reboot loop is continuous.

The example 'esp-idf/examples/wifi/getting_started/station' works just fine (connecting and stay connecting) proving the hardware is fine. 

### Expected Behavior

ESP32_S2 to enter into light sleep mode

### Actual Behavior

ESP32_S2 reboots with cause rst:0x10 (RTCWDT_RTC_RST). This reboot loop is continuous.

### Steps to reproduce

1. idf.py set-target esp32s2
2. idf.py menuconfig
3. The only settings changed are: (SSID/Password)
4. idf.py build

I have tried changing SSID/Passwords, WiFi listen Interval (1 and 3), all cause this issue.

### Code to reproduce this issue
The line that causes the issue
```cpp
#if CONFIG_FREERTOS_USE_TICKLESS_IDLE
            .light_sleep_enable = true
#endif
    };
```
Changing the following prevents the ESP32_S2 from rebooting, but will not allow low power mode.
```cpp
 .light_sleep_enable = false
```

## Debug Logs

After reboot
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x180c
load:0x40050000,len:0x14b0
load:0x40054000,len:0x210c
entry 0x400502dc
I (46) boot: ESP-IDF v4.2-dev-1660-g7d7521367 2nd stage bootloader
I (47) boot: compile time 20:03:00
I (47) boot: chip revision: 0
I (50) boot.esp32s2: SPI Speed      : 80MHz
I (55) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 2MB
I (64) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 00100000
I (103) boot: End of partition table
I (107) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x16968 ( 92520) map
I (135) esp_image: segment 1: paddr=0x00026990 vaddr=0x3ffc8b30 size=0x03334 ( 13108) load
I (139) esp_image: segment 2: paddr=0x00029ccc vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at /home/wpeacock/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (142) esp_image: segment 3: paddr=0x0002a0d8 vaddr=0x40024404 size=0x05f40 ( 24384) load
I (157) esp_image: segment 4: paddr=0x00030020 vaddr=0x40080020 size=0x6b4f0 (439536) map
0x40080020: _stext at ??:?

I (251) esp_image: segment 5: paddr=0x0009b518 vaddr=0x4002a344 size=0x0e7e8 ( 59368) load
0x4002a344: vRingbufferReturnItem at /home/wpeacock/esp/esp-idf/components/esp_ringbuf/ringbuf.c:1269 (discriminator 2)

I (278) boot: Loaded app from partition at offset 0x10000
I (278) boot: Disabling RNG early entropy source...
I (279) cache: Instruction cache 	: size 8KB, 4Ways, cache line size 32Byte
I (286) cpu_start: Pro cpu up.
I (290) cpu_start: Application information:
I (295) cpu_start: Project name:     power_save
I (300) cpu_start: App version:      1
I (305) cpu_start: Compile time:     Jun  2 2020 20:02:28
I (311) cpu_start: ELF file SHA256:  3c8a130725baa4e7...
I (317) cpu_start: ESP-IDF:          v4.2-dev-1660-g7d7521367
I (323) cpu_start: Single core mode
I (327) heap_init: Initializing. RAM available for dynamic allocation:
I (334) heap_init: At 3FFD1628 len 0002A9D8 (170 KiB): DRAM
I (341) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (347) cpu_start: Pro cpu start user code
I (407) spi_flash: detected chip: generic
I (408) spi_flash: flash io: dio
W (408) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (418) cpu_start: Starting scheduler on PRO CPU.
I (431) pm_esp32s2: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 10, Light sleep: ENABLED
I (451) wifi:wifi driver task: 3ffda024, prio:23, stack:6656, core=0
I (451) system_api: Base MAC address is not set
I (451) system_api: read default base MAC address from EFUSE
I (461) wifi:wifi firmware version: 517093f
I (461) wifi:wifi certification version: v7.0
I (461) wifi:config NVS flash: enabled
I (471) wifi:config nano formating: disabled
I (471) wifi:Init dynamic tx buffer num: 32
I (471) wifi:Init data frame dynamic rx buffer num: 32
I (481) wifi:Init management frame dynamic rx buffer num: 32
I (481) wifi:Init management short buffer num: 32
I (491) wifi:Init static rx buffer size: 1600
I (491) wifi:Init static rx buffer num: 10
I (501) wifi:Init dynamic rx buffer num: 32
I (671) phy: phy_version: 303, c1e745d, Apr 11 2020, 14:22:04, 0, 2
I (671) wifi:enable tsf
I (671) wifi:mode : sta (7c:df:a1:00:3d:5d)
I (671) power_save: esp_wifi_set_ps().
I (671) wifi:Set ps type: 1

I (791) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (791) wifi:state: init -> auth (b0)
I (811) wifi:state: auth -> assoc (0)
I (831) wifi:state: assoc -> run (10)
I (861) wifi:connected with Torah, aid = 9, channel 6, BW20, bssid = 78:8a:20:5d:7a:e9
I (861) wifi:security type: 3, phy: bgn, rssi: -46
I (861) wifi:pm start, type: 1

I (921) wifi:AP's beacon interval = 102400 us, DTIM period = 1
``` 
device reboots here 
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x10 (RTCWDT_RTC_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe8100,len:0x4
load:0x3ffe8104,len:0x180c
load:0x40050000,len:0x14b0
load:0x40054000,len:0x210c
entry 0x400502dc
I (46) boot: ESP-IDF v4.2-dev-1660-g7d7521367 2nd stage bootloader
I (47) boot: compile time 20:03:00
I (47) boot: chip revision: 0
I (50) boot.esp32s2: SPI Speed      : 80MHz
I (55) boot.esp32s2: SPI Mode       : DIO
I (59) boot.esp32s2: SPI Flash Size : 2MB
I (64) boot: Enabling RNG early entropy source...
I (69) boot: Partition Table:
I (73) boot: ## Label            Usage          Type ST Offset   Length
I (80) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (88) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  2 factory          factory app      00 00 00010000 00100000
I (103) boot: End of partition table
I (107) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x16968 ( 92520) map
I (135) esp_image: segment 1: paddr=0x00026990 vaddr=0x3ffc8b30 size=0x03334 ( 13108) load
I (139) esp_image: segment 2: paddr=0x00029ccc vaddr=0x40024000 size=0x00404 (  1028) load
0x40024000: _WindowOverflow4 at /home/wpeacock/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (142) esp_image: segment 3: paddr=0x0002a0d8 vaddr=0x40024404 size=0x05f40 ( 24384) load
I (157) esp_image: segment 4: paddr=0x00030020 vaddr=0x40080020 size=0x6b4f0 (439536) map
```

chegewara, 2020-06-02:
https://esp32.com/viewtopic.php?f=10&p=56361
`Wi-Fi with automatic light sleep — Not supported (WIFI-2305)`

waynepeacock, 2020-06-02:
Many thanks chegewara, I missed that!

closed 2020-06-02 by waynepeacock

felmue, 2020-08-03:
Good day

I've noticed a while ago that automatic light-sleep for WiFi has been fixed on July 10th,  according to the ESP32-S2 support status document: https://esp32.com/viewtopic.php?f=10&p=56361
`Wi-Fi with automatic light sleep — Supported`

And today I noticed some fresh commits so I made a fresh clone of esp-idf and rebuilt and ran the power_save example, but the ESP32s2 still reboots when light-sleep is set to true.

May I kindly ask if there is a date for when this fix will make it into the master branch?

Thank you very much in advance.

Felix

igrr, 2020-08-03:
Hi @felmue, I ran the power_save example with the latest master branch locally, and didn't meet this issue. Could you please open a new issue and provide the information requested in the issue template, so that we can troubleshoot this?

felmue, 2020-08-03:
Hi @igrr, thank you for replying. I appreciate it. I opened a new issue as you suggested: https://github.com/espressif/esp-idf/issues/5692

Thanks
Felix