espressif/esp-idf #5381: Can Not Read I2S ADC DMA Buffer with Wifi Enabled

physiii, 2020-06-02:
Testing with _ESP-IDF v4.2-dev-1415-ga2263571b_

_esp_wifi_start_ causes _i2s_read_ to read the same data from the DMA buffer for the ADC. I tried changing wifi settings in menuconfig like optimizations, cores, flash. Also tried wrapping i2s_read() in _i2s_adc_enable_ and _i2s_adc_disable_.

See my [gist example](https://gist.github.com/physiii/e8a51897f9542ea5a408e642aa25793d) which is  based on the [i2s_adc_dac](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s_adc_dac) example. I launch wifi station at the bottom of _app_main_

The example plays a test file (_i2s_write_) in one task and reads the DMA buffer (_i2s_read_) from a second task.

Result without starting wifi (this is what I expect):

```
Playing file example: 
======
d2 06 d2 06 ff 76 ff 76 
f0 76 f0 76 f5 76 f5 76 
d0 76 d0 76 ff 76 ff 76 
e0 76 e0 76 ed 76 ed 76 
ff 76 ff 76 f0 76 f0 76 
ff 76 ff 76 ff 76 ff 76 
ff 76 ff 76 1d 77 1d 77 
f0 76 f0 76 ff 76 ff 76 
======
======
a6 75 a6 75 03 76 03 76 
6a 76 6a 76 0f 77 0f 77 
80 77 80 77 00 78 00 78 
51 78 51 78 9f 78 9f 78 
ba 78 ba 78 b0 78 b0 78 
e7 78 e7 78 6b 78 6b 78 
3f 78 3f 78 cb 77 cb 77 
30 77 30 77 c7 76 c7 76 
======
======
5c 78 5c 78 3a 78 3a 78 
a6 77 a6 77 36 77 36 77 
b6 76 b6 76 3f 76 3f 76 
00 76 00 76 72 75 72 75 
31 75 31 75 3f 75 3f 75 
3f 75 3f 75 80 75 80 75 
8c 75 8c 75 00 76 00 76 
7f 76 7f 76 13 77 13 77 
======
======
3f 75 3f 75 73 75 73 75 
00 76 00 76 5b 76 5b 76 
f0 76 f0 76 71 77 71 77 
bb 77 bb 77 0c 78 0c 78 
5a 78 5a 78 7f 78 7f 78 
cd 78 cd 78 7f 78 7f 78 
4f 78 4f 78 de 77 de 77 
67 77 67 77 d2 76 d2 76 
======
======
7f 78 7f 78 38 78 38 78 
be 77 be 77 6c 77 6c 77 
ff 76 ff 76 7f 76 7f 76 
0e 76 0e 76 9a 75 9a 75 
61 75 61 75 3f 75 3f 75 
ff 74 ff 74 3f 75 3f 75 
61 75 61 75 00 76 00 76 
27 76 27 76 c3 76 c3 76 
======
======
ff 74 ff 74 58 75 58 75 
9e 75 9e 75 30 76 30 76 
9d 76 9d 76 1b 77 1b 77 
81 77 81 77 00 78 00 78 
76 78 76 78 7f 78 7f 78 
7e 78 7e 78 7f 78 7f 78 
58 78 58 78 1e 78 1e 78 
b0 77 b0 77 30 77 30 77 
======

```

Then when wifi is started you can see the buffer starts returning the same value like the DMA receive buffer stops being filled.

```
Playing file example: 
I (584) ad/da: ESP_WIFI_MODE_STA
I (604) wifi:wifi driver task: 3ffcf300, prio:23, stack:6656, core=1
I (604) system_api: Base MAC address is not set
I (604) system_api: read default base MAC address from EFUSE
I (604) wifi:wifi firmware version: cdcf32f
I (614) wifi:wifi certification version: v7.0
I (614) wifi:config NVS flash: disabled
I (614) wifi:config nano formating: disabled
I (624) wifi:Init dynamic tx buffer num: 32
I (624) wifi:Init data frame dynamic rx buffer num: 32
I (634) wifi:Init management frame dynamic rx buffer num: 32
I (634) wifi:Init management short buffer num: 32
I (644) wifi:Init static rx buffer size: 1600
I (644) wifi:Init static rx buffer num: 10
I (644) wifi:Init dynamic rx buffer num: 32
I (754) phy: phy_version: 4181, c9a03e2, Mar 10 2020, 12:45:06, 0, 0
I (754) wifi:mode : sta (24:0a:c4:17:e8:c4)
I (754) ad/da: wifi_init_sta finished.
I (1484) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
======
4e 75 4e 75 9f 75 9f 75 
00 76 00 76 70 76 70 76 
f0 76 f0 76 83 77 83 77 
00 78 00 78 63 78 63 78 
58 78 58 78 5a 78 5a 78 
7f 78 7f 78 3c 78 3c 78 
e0 77 e0 77 a7 77 a7 77 
ff 76 ff 76 7f 76 7f 76 
======
I (2254) wifi:state: init -> auth (b0)
I (2264) wifi:state: auth -> assoc (0)
I (2274) wifi:state: assoc -> run (10)
Playing file example: 
I (2294) wifi:connected with Echo_42, aid = 4, channel 6, BW20, bssid = 80:2a:a8:41:64:cd
I (2294) wifi:security type: 4, phy: bgn, rssi: -43
I (2294) wifi:pm start, type: 1

I (2354) wifi:AP's beacon interval = 102400 us, DTIM period = 1
======
ff 74 ff 74 59 75 59 75 
9d 75 9d 75 3e 76 3e 76 
ad 76 ad 76 58 77 58 77 
75 77 75 77 13 78 13 78 
6b 78 6b 78 7f 78 7f 78 
7f 78 7f 78 ae 78 ae 78 
3f 78 3f 78 00 78 00 78 
63 77 63 77 d2 76 d2 76 
======
======
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
======
I (3094) esp_netif_handlers: sta ip: 192.168.1.66, mask: 255.255.255.0, gw: 192.168.1.1
I (3094) ad/da: got ip:192.168.1.66
I (3094) ad/da: connected to ap SSID:Echo_42 password:qscwdvpk
======
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
======
======
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
======
======
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 
fa 74 fa 74 fa 74 fa 74 

```

koobest, 2020-06-03:
Hi,@physiii
Please calling `esp_wifi_set_ps(WIFI_PS_NONE);` to disable the WIFI power save mode.

Alvin1Zhang, 2020-06-03:
@physiii Thanks for reporting and this is duplicate with https://github.com/espressif/esp-idf/issues/3714, will close this ticket and let us follow up in https://github.com/espressif/esp-idf/issues/3714. Thanks.

closed 2020-06-03 by Alvin1Zhang