espressif/esp-idf #5382: Error while configuring environment variables

GarMingLi, 2020-06-03:
When I execute the command:
`. export.sh `
The following error occurred:
```
Adding ESP-IDF tools to PATH...
Traceback (most recent call last):
  File "/home/ljm/esp32/esp-idf/tools/idf_tools.py", line 1387, in <module>
    main(sys.argv[1:])
  File "/home/ljm/esp32/esp-idf/tools/idf_tools.py", line 1383, in main
    action_func(args)
  File "/home/ljm/esp32/esp-idf/tools/idf_tools.py", line 941, in action_export
    tool.find_installed_versions()
  File "/home/ljm/esp32/esp-idf/tools/idf_tools.py", line 511, in find_installed_versions
    ver_str = self.check_version()
  File "/home/ljm/esp32/esp-idf/tools/idf_tools.py", line 464, in check_version
    version_cmd_result = run_cmd_check_output(cmd, None, extra_paths)
  File "/home/ljm/esp32/esp-idf/tools/idf_tools.py", line 187, in run_cmd_check_output
    p = subprocess.Popen(cmd, stdout=subprocess.PIPE, stdin=subprocess.PIPE, stderr=subprocess.PIPE)
  File "/usr/lib/python2.7/subprocess.py", line 394, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1047, in _execute_child
    raise child_exception
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe7 in position 7: ordinal not in range(128)
```
## Environment

- Module or chip used:  ESP32S
- IDF version:  v4.1-dev-2079-g5dbabae9d
- Build System:      idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Power Supply:    external 3.3V

dshil, 2020-06-03:
Hi @GarMingLi, did you try with python3?

Please note, that according to the [release notes from v4.0.1](https://github.com/espressif/esp-idf/releases) it was recommended to use python3:

```
Encourage use of Python 3 (Python 2 is officially no longer supported upstream and ESP-IDF support will be removed in a future version.)
```

GarMingLi, 2020-06-03:
When I switched python to Python 3, it was still wrong：
```
ljm@DESKTOP-APB6ADI:~/esp32/code/hub_test_wifi$ python --version
Python 3.6.9
ljm@DESKTOP-APB6ADI:~/esp32/code/hub_test_wifi$ . /home/ljm/esp32/esp-idf/export.sh 
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
The following Python requirements are not satisfied:
future>=0.15.2
pyparsing>=2.0.3,<2.4.0
pyelftools>=0.22
Please follow the instructions found in the "Set up the tools" section of ESP-IDF Getting Started Guide
```

dshil, 2020-06-03:
I guess that `install.sh` script should install all required python packages. Run `./install.sh` or setup requirements manually as:

```
/usr/bin/python -m pip install --user -r $IDF_PATH/requirements.txt
```

dshil, 2020-06-09:
@GarMingLi any updates?

GarMingLi, 2020-08-17:
When I switch to root, the error is gone

dshil, 2020-08-29:
Hi @GarMingLi , thanks for reply, should we consider this issue as resolved (if yes somebody from ESP-IDF will close the issue)?

cc @Alvin1Zhang

closed 2020-09-05 by GarMingLi