espressif/esp-idf #5383: Installation fails with python environment error

berkutta, 2020-06-03:
I'm trying to install esp-idf according to the latest documentation. But it fails with some strange error during python environment creation. I have Python 3.8.3 installed on an updated Arch Linux installation.

```
➜  esp-idf git:(release/v4.2) ✗ ./tools/idf_tools.py install-python-env
Creating a new Python environment in /home/user/.espressif/python_env/idf4.2_py3.8_env
Installing Python packages from /home/user/esp/esp-idf/requirements.txt
Traceback (most recent call last):
  File "./tools/idf_tools.py", line 1492, in <module>
    main(sys.argv[1:])
  File "./tools/idf_tools.py", line 1488, in main
    action_func(args)
  File "./tools/idf_tools.py", line 1215, in action_install_python_env
    subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
  File "/usr/lib/python3.8/subprocess.py", line 359, in check_call
    retcode = call(*popenargs, **kwargs)
  File "/usr/lib/python3.8/subprocess.py", line 340, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/user/.espressif/python_env/idf4.2_py3.8_env/bin/python'
```

dobairoland, 2020-06-03:
Hi @berkutta. Firstly, you have to run `tools/idf_tools.py install` and only after that you run `tools/idf_tools.py install-python-env`.

Note that it is more convenient to run the `install.sh` script which includes both of these steps.

berkutta, 2020-06-03:
Hi @dobairoland 

Sorry to not mention this, I ran the command explicitly to debug it. The install one runs through nice. I'm getting the same Error with `install.sh`.

```
➜  esp-idf git:(release/v4.2) ✗ ./tools/idf_tools.py install           
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2020r1-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2020r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20200420 (already installed)
```

dobairoland, 2020-06-03:
@berkutta What is your IDF version, i.e. the output of `git describe --tags`?

What shell are you using?

Is the directory `/home/user/.espressif/python_env/idf4.2_py3.8_env/` empty?

berkutta, 2020-06-03:
@dobairoland I used zsh, but now also tried to redo it with bash (with removing the .espressif folder)

```
➜  esp-idf git:(release/v4.2) ✗ git describe --tags 
v4.2-dev-1303-gcd4fa46f5
➜  esp-idf git:(release/v4.2) ✗ ls /home/user/.espressif/python_env/idf4.2_py3.8_env/
ls: cannot access '/home/user/.espressif/python_env/idf4.2_py3.8_env/': No such file or directory
➜  esp-idf git:(release/v4.2) ✗ ls /home/user/.espressif/
dist  tools
```

dobairoland, 2020-06-03:
I'm also using Arch linux and zsh with the same version of `idf_tools.py`.
I have `python-virtualenv 20.0.21-1` installed. I guess you as well because I don't see in the log that it would install it with `pip`.

`idf_tools.py` runs the following command:
`/usr/bin/python -m virtualenv /home/user/.espressif/python_env/idf4.2_py3.8_env`
Could you please try it and see if the environment is created?

You don't have any virtual environment activated in the beginning, ie. `which python` returns `/usr/bin/python`, right?

berkutta, 2020-06-03:
@dobairoland thanks for the debugging steps and clarifying that everything should work fine on Arch Linux. Your direction with maybe something weird with virtualenv was correct. It was an issue with my installation. I think I installed virtualenv and some other things on the system with pip (really onto the system, not into a virtualenv) which caused some issues. After nuking everything python related, reinstalling everything clean with pacman it nearly worked. I still got a weird pip unzip issue when doing the Espressif `install.sh`, but that cleared up after clearing the pip cache. Sorry for reporting my broken system as an issue. Will close this issue.

closed 2020-06-03 by berkutta

dobairoland, 2020-06-04:
@berkutta Great, I'm happy that it got resolved. No reason to apologize!

a-anjos, 2022-03-15:
I had the same error and the problem is the location of bin in the virtualenv folder. Inside:

`~/.espressif/python_env/idf4.0_py3.10_env`
you have:
`lib  local  pyvenv.cfg`
I did:
`ln -s local/bin bin`

Now I have a bin directory (soft link) in the root of the virtualenv directory ;)

fjxgit, 2024-05-18:
ew