espressif/esp-idf #5384: BT Classic GAP PIN pairing

JHedgesBartec, 2020-06-03:
**Is your feature request related to a problem? Please describe.**

I am developing a product where an ESP32 could be used as a replacement to a Bluegiga WT12 module, this includes emulation of iWRAP 3.0.0. iWRAP specifies that pairings for bluetooth classic can me managed and minipulated seperately to connections and the profiles supported. For example I would be able to pair to an android device with no profiles available with the later intention of connecting using SPP.
From what I can see this is not possible for BT Classic using Bluedroid and legacy pin pairing with SSP disabled.

**Describe the solution you'd like**

A clear call to trigger the pairing/bonding process from the ESP32 to a device regardless of the profiles supported provding just a device bda.

The expected behaviour would be the ESP would request a PIN from a third party device which a user would then enter to be handshaked to form the bond.

**Describe alternatives you've considered**

Using SSP but this would not be compatible with existing products that we intend to evolve to use the ESP32 with emulation of iWRAP.

**Additional context**

IDE: Visual Studio and Visual GDB (CMake)
IDF: v4.0.1 Release
Module: ESP-WROOM-32

iWRAP specification: https://www.sparkfun.com/datasheets/Wireless/Bluetooth/iWRAP3_User_Guide.pdf

blueMoods, 2020-06-19:
Hi @JHedgesBartec, 
We can disable SSP in menuconfig. In this case, it will use legacy. And you can set Fixed PIN for ESP32, then customers can input it in remote device if it support variable PIN.

But there is no APIs to trigger the pairing/bonding process from the ESP32 to a device regardless of the profiles, yet. It will be executed when you connect to a profile.

Thanks.

munho-espressif, 2020-09-07:
Hi @JHedgesBartec ,

There are no functions that can be paired and connected separately. The pairing and connection happens automatically when you call esp_spp_connect( ).  If we want to develop this feature, we will let you know.

Thanks.

JHedgesBartec, 2020-09-10:
> 
> 
> Hi @JHedgesBartec ,
> 
> There are no functions that can be paired and connected separately. The pairing and connection happens automatically when you call esp_spp_connect( ). If we want to develop this feature, we will let you know.
> 
> Thanks.

Hi @munho-espressif 

It is dissapointing to hear that this is not being considered as I frequently come accross issues due to devices not having profiles active all of the time, these range from PCs to Android devices.
This limitation is surely causing other bugs where devices do not present bonds in the bond list once connected simply due to not having a clear pairing/bonding process for BT classic. 
https://github.com/espressif/esp-idf/issues/5850

Regards,
Jack.