espressif/esp-idf #5385: Assert when connecting to SPP device with incorrect PIN (BT Classic)

JHedgesBartec, 2020-06-03:
## Environment

- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0.1 Release
- Build System:       CMake
- Compiler version SysGcc 8.2.0
- Operating System:     Windows
- (Windows only) environment type: IDE below for build process
- IDE: VisualStudio and VisualGDB
- Power Supply:         external 3.3V

## Problem Description
Assert thrown on core 0 when attempting to connect to a device using esp_spp_start_discovery() then esp_spp_connect() as master upon a successful discovery completion.

Apologies as I am new to ESP32 development and VisualStudio combined with VisualGDB does not provide a very good debugging environment.

### Expected Behavior
Connection would fail and the appropriate event called.

### Actual Behavior
Application crashes.

### Steps to reproduce
Configure a BT Classic client with SSP disabled (I also have a server running for SPP)
Use a fixed PIN which is different to the device to connect.
Trigger a SPP discovery and then a SPP connect.

## Debug Logs
Taken from the point at which SPP discovery is triggered:
I (123824) Connect Classic: Starting SPP discovery
I (124669) SPP_CB: event 8
I (124670) SPP_CB: Discovery successful, connecting...
I (124672) SPP_CB: event 29
W (125043) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 1 closed: Start failed (res: 12)
W (125045) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 0 closed: Closed (res: 19)
assertion "queue != NULL" failed: file "C:/SysGCC/esp32/esp-idf/v4.0/components/bt/common/osi/fixed_queue.c", line 153, function: fixed_queue_dequeue
abort() was called at PC 0x4016b943 on core 0

## Other items if possible
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/4724386/sdkconfig.zip)

Alvin1Zhang, 2020-06-04:
@JHedgesBartec Thanks for reporting, we will look into.

Wth-Esp, 2020-06-19:
Need protection for Rx and Tx Queue of RFCOMM port when connection not successfully established. There has been an MR to fix this and maybe soon sync to Github.

Wth-Esp, 2020-06-19:
Hi @JHedgesBartec

Would you please give me the commit of your IDF and I will make a log patch for you to reproduce this issue (if possible).

Thanks
Weitianhua

JHedgesBartec, 2020-06-19:
Hi @Wth-Esp 

As stated in the environment we are using VisualGDB paired with Visual Studio and Windows 10, the installation of the IDF doesn't use git as far as I am aware, instead SysProg provide executables to install it with a toolchain. See https://gnutoolchains.com/esp32/ for the available toolchains and IDF versions, we currently use release/v4.0.

VisualGDB has limited support for using master or other commits.

Thanks,
Jack

Wth-Esp, 2020-06-22:
Hi @JHedgesBartec 

Here is a log patch for you to reproduce and please give me the whole log when assert appears.
[log.zip](https://github.com/espressif/esp-idf/files/4811581/log.zip)

Thanks

Weitianhua

Wth-Esp, 2020-07-08:
Any update here?

JHedgesBartec, 2020-07-08:
> 
> 
> Any update here?

Hi @Wth-Esp ,

Please see my previous reply stating the environment and how we are unable to apply patches due to the way VisualGDB provides the IDF framework through installers.
This issue is something I have found 100% reproduction rate when tried, so should be simple for someone else to see also.

I am very pressed on a different commercial project right now so have been unable to commit any time to this.

Regards,
Jack Hedges.

Wth-Esp, 2020-07-08:
Hi @JHedgesBartec 

OK, I have found some place to fix this issue. I will contact with QA to confirm the fix.
I will inform you about it if any update.

Thanks
Weitianhua

Alvin1Zhang, 2020-10-14:
Thanks for reporting and sorry for slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/5812a935ba0b10f8431681c1327cf985bec989ce, feel free to reopen if the issue still happens. Thanks.

closed 2020-10-14 by Alvin1Zhang