espressif/esp-idf #5386: I2S PDM mode problems?

fpgamaster, 2020-06-03:
- Development Kit: none
- Module or chip used:  ESP32-WROOM-32D (Chip is ESP32D0WDQ5 (revision 1))
- IDF version: v4.2-dev-1660-g7d7521367
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r1) 8.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description
1. APLL is not used in I2S PDM mode
2. Strange real rates reported
3. No difference if I2S_CHANNEL_FMT_ONLY_LEFT or I2S_CHANNEL_FMT_ONLY_RIGHT is set

//Detailed problem description goes here.
A MEMS microphone with PDM out is used and configured to send data on rising edge.
I2S is configured as follows:

`
        esp_err_t err;

        i2s_config_t i2s_config;
        i2s_pin_config_t pin_config;

        ESP_LOGI(LOG_TAG, "Configuring I2S...");
        memset(&i2s_config, 0, sizeof(i2s_config));
        memset(&pin_config, 0, sizeof(pin_config));

        i2s_config.mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_PDM);
        i2s_config.sample_rate = 16000;                                 // 16KHz
        i2s_config.bits_per_sample = 16
        i2s_config.channel_format = I2S_CHANNEL_FMT_ONLY_LEFT;
        i2s_config.communication_format = I2S_COMM_FORMAT_PCM;
        i2s_config.intr_alloc_flags = ESP_INTR_FLAG_LEVEL1;             // Interrupt level 1
        i2s_config.use_apll = true;
        i2s_config.dma_buf_count = 2;                                   // number of buffers, 128 max
        i2s_config.dma_buf_len = 128;                                   // size of each buffer

        pin_config.bck_io_num = I2S_PIN_NO_CHANGE;
        pin_config.ws_io_num = MIC_CLK_PIN;                             // Clock
        pin_config.data_in_num = MIC_DATA_PIN;                          // Data input
        pin_config.data_out_num = I2S_PIN_NO_CHANGE;

        // Configuring the I2S driver and pins.
        err = i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
        if (err != ESP_OK) {
                ESP_LOGE(LOG_TAG, "Failed installing driver: %d\n", err);
                return (-1);
        }
        err = i2s_set_pin(I2S_NUM_0, &pin_config);
        if (err != ESP_OK) {
                ESP_LOGE(LOG_TAG, "Failed setting pin: %d\n", err);
                return (-1);
        }
`

1. APLL is not used.
I (2184) I2S: PLL_D2: Req RATE: 16000, real rate: 32258.000, BITS: 16, CLKM: 31, BCK: 5, MCLK: 31.250, SCLK: 1032256.000000, diva: 64, divb: 16
I (2194) I2S: DMA Malloc info, datalen=blocksize=1024, dma_buf_count=2

Data received is incorrect!

2. Reported real rates
I (2184) I2S: PLL_D2: Req RATE: 16000, real rate: 32258.000, BITS: 16
or
I (2212) I2S: PLL_D2: Req RATE: 48000, real rate: 100000.000, BITS: 16
or
I (2214) I2S: PLL_D2: Req RATE: 96000, real rate: 100000.000, BITS: 32

3. Setting i2s_config.channel_format to I2S_CHANNEL_FMT_ONLY_LEFT or I2S_CHANNEL_FMT_ONLY_RIGHT does not change anything.

### Expected Behavior
1. APPL to be used in PDM mode
2. Reported real rates to be close to requested
3. PDM data to be sampled as requested on rising or falling edges

The main question here is whether PDM mode is broken and can be fixed or there are hardware limitations and we need to switch our design to an I2S microphone?

Best Regards

Alvin1Zhang, 2020-06-04:
@fpgamaster Thanks for the detailed report, we will look into.

Alvin1Zhang, 2020-08-18:
Thanks for reporting and sorry for slow turnaround, we have a fix under internal reviewing, the issue will be closed automatically once the fix is available on GitHub. Thanks.

alessandro-montanari, 2020-09-15:
Hi,
we are having exactly the same issues and it seems impossible to use a PDM microphone on an ESP32 PICO.
Is there any update on this?

@fpgamaster did you find a solution?

Thanks!

fpgamaster, 2020-09-15:
HI @alessandro-montanari,

Unfortunately no :(
We have moved to an I2S microphone because of this.

The PDM mode used to work somehow with an older IDF version but can not remember which one.
There were problems too, like changing the DC signal level from time to time most probably because of loosing or wrong byte synchronization ...

closed 2020-10-14 by espressif-bot

RafigRzayev, 2020-10-29:
@Alvin1Zhang From last commit it looks as if it is fixed now? Can you confirm?

I am also trying to use PDM mic, but no progress. 

```cpp

// I2S perhiperhal number
#define I2S_CHANNEL                 I2S_NUM_0 // I2S_NUM_1 doesn't support PDM

// I2S pins
#define I2S_PIN_BIT_CLOCK           I2S_PIN_NO_CHANGE  // not used
#define I2S_PIN_WORD_SELECT         26
#define I2S_PIN_DATA_OUT            I2S_PIN_NO_CHANGE  // not used
#define I2S_PIN_DATA_IN             25

// I2S CONFIG PARAMS
#define I2S_MODE                    (I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_PDM)
#define I2S_SAMPLE_RATE             16000
#define I2S_BITS_PER_SAMPLE         I2S_BITS_PER_SAMPLE_16BIT
#define I2S_CHANNEL_FMT             I2S_CHANNEL_FMT_ONLY_RIGHT
#define I2S_COMM_FMT                I2S_COMM_FORMAT_PCM
#define I2S_INTERRUPT_PRIO          ESP_INTR_FLAG_LEVEL1
#define I2S_DMA_BUF_COUNT           4
#define I2S_DMA_BUF_SIZE            1000
#define I2S_ENABLE_ACCURATE_CLK     true

bool I2S_Init() {
  i2s_config_t i2s_config;
  memset(&i2s_config, 0, sizeof(i2s_config));

  i2s_config.mode = (i2s_mode_t)I2S_MODE;
  i2s_config.sample_rate =  I2S_SAMPLE_RATE;
  i2s_config.bits_per_sample = I2S_BITS_PER_SAMPLE;
  i2s_config.channel_format = I2S_CHANNEL_FMT;
  i2s_config.communication_format = (i2s_comm_format_t)I2S_COMM_FMT;
  i2s_config.intr_alloc_flags = I2S_INTERRUPT_PRIO;
  i2s_config.dma_buf_count = I2S_DMA_BUF_COUNT;
  i2s_config.dma_buf_len = I2S_DMA_BUF_SIZE;
  i2s_config.use_apll = I2S_ENABLE_ACCURATE_CLK;

  i2s_pin_config_t pins = {
    .bck_io_num = I2S_PIN_BIT_CLOCK,
    .ws_io_num =  I2S_PIN_WORD_SELECT,
    .data_out_num = I2S_PIN_DATA_OUT,
    .data_in_num = I2S_PIN_DATA_IN
  };

  if (i2s_driver_install(I2S_CHANNEL, &i2s_config, 0, NULL) != ESP_OK) {
    Serial.println("i2s_driver_install() error");
    return false;
  }

  if (i2s_set_pin(I2S_NUM_0, &pins) != ESP_OK) {
    Serial.println("i2s_set_pin() error");
    return false;
  }
}`

All second bytes in received PCM data should have close values since they are MSB. But I don't have it in my data. Also, there is aperiodic repetition of 0x00 0x80.

![Capture](https://user-images.githubusercontent.com/58329539/97553827-5b9fa780-19ef-11eb-9d26-371e22933488.PNG)