espressif/esp-idf #5387: I can't install esp-idf using python installation from the microsoft store.

SR-Lut3t1um, 2020-06-03:
I'm using windows 10 Build 19635 and I'm getting this installation error:
https://gist.github.com/SR-Lut3t1um/fa418a20d2ee87af93ee325a12d617ee
I ran the installer with right click/Administrative access.

igrr, 2020-06-04:
This might be related to the path length limit issue (https://github.com/pypa/virtualenv/issues/1769#issuecomment-614619504).
Could you please check if `virtualenv` is working correctly in this Python installation? Open the command prompt (cmd.exe) and try the following:

1. Try creating a virtual environment (in `temp_env` directory on the desktop):
    ```
    C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1008.0_x64__qbz5n2kfra8p0\python3.8.exe -m virtualenv C:\Users\tobia\OneDrive\Desktop\temp_env
   ```

2. Try installing IDF packages into this virtual environment:
    ```
    set IDF_PATH=C:\Users\tobia\OneDrive\Desktop\esp-idf
    C:\Users\tobia\OneDrive\Desktop\temp_env\Scripts\python.exe -m pip install -r %IDF_PATH%\requirements.txt
    ```

SR-Lut3t1um, 2020-06-04:
I'm running into this exception, when running the 1.st command:
```
PS C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1008.0_x64__qbz5n2kfra8p0> .\python3.8.exe -m virtualenv C:\Users\tobia\OneDrive\Desktop\temp_env
Exception in thread Thread-6:
Traceback (most recent call last):
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1008.0_x64__qbz5n2kfra8p0\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.8_3.8.1008.0_x64__qbz5n2kfra8p0\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\tobia\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\virtualenv\seed\via_app_data\via_app_data.py", line 52, in _install
    installer.install(creator.interpreter.version_info)
  File "C:\Users\tobia\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\virtualenv\seed\via_app_data\pip_install\base.py", line 51, in install
    for name, module in self._console_scripts.items():
  File "C:\Users\tobia\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\virtualenv\seed\via_app_data\pip_install\base.py", line 118, in _console_scripts
    entry_points = self._dist_info / "entry_points.txt"
  File "C:\Users\tobia\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\local-packages\Python38\site-packages\virtualenv\seed\via_app_data\pip_install\base.py", line 105, in _dist_info
    raise RuntimeError("no dist info")  # pragma: no cover
RuntimeError: no dist info
created virtual environment CPython3.8.3.final.0-64 in 4499ms
  creator Venv(dest=C:\Users\tobia\OneDrive\Desktop\temp_env, clear=False, global=False, describe=CPython3Windows)
  seeder FromAppData(download=False, pip=latest, setuptools=latest, wheel=latest, via=copy, app_data_dir=C:\Users\tobia\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.8_qbz5n2kfra8p0\LocalCache\Local\pypa\virtualenv\seed-app-data\v1.0.1)
  activators BashActivator,BatchActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivato
```

igrr, 2020-06-04:
Okay, so it looks like your Python installation can not create virtual environments. I'm afraid you'll have to install another copy of Python which doesn't have this issue, or use the workaround from the virtualenv issue I've linked above.

Alvin1Zhang, 2021-02-02:
Thanks for reporting and sorry fro slow turnaround, the fix is available at https://github.com/espressif/esp-idf/commit/76acc8ddde0b2fab66f9918792f9f062040ee254, feel free to reopen. Thanks.

closed 2021-02-02 by Alvin1Zhang

georgik, 2021-02-02:
Update: The new installers works alongside Python from Windows Store.
- Online installer (30 MB) : https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
- Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe